betaflight/betaflight #8721: H7 FC - yaw spin / twitching during throttle

TheIsotopes, 2019-08-14:
**Describe the bug**
I have tested an Spracing H7 Nano on different frames with the same fimware build 1805.
On both frames the fc is twitching or spins more than 90 degrees when i gif throttle. I give only slightly throttle (no fast throttle pumps) and already the fc spins or twitching

**Expected behavior**
I suspect a fault in the motor control ... or is the fc may be defective?

**Flight controller configuration**
```
# Betaflight / SPRACINGH7NANO (SP7N) 4.1.0 Aug 13 2019 / 08:33:54 (52173c798) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7NANO

# name: Mark1

# feature
feature -TELEMETRY
feature -RSSI_ADC
feature -OSD
feature -TRANSPONDER

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 9,2::AW:0
led 1 9,3::AW:0
led 2 9,4::AW:0
led 3 9,5::AW:0
led 4 6,5::AW:0
led 5 6,4::AW:0
led 6 6,3::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 0 3
mode_color 6 1 6

# aux
aux 0 0 0 1600 2000 0 0
aux 1 13 1 1000 1200 0 0
aux 2 26 0 1800 2000 0 0
aux 3 31 0 1200 1400 0 0
aux 4 36 1 1800 2000 0 0

# master
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SUMD
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1550
set use_cbat_alerts = ON
set cbat_alert_percent = 20
set vbat_duration_for_warning = 50
set vbat_duration_for_critical = 20
set ibata_scale = 179
set pid_process_denom = 1
set name = Mark1

# profile 0
set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set i_pitch = 85
set i_roll = 80
set d_min_roll = 25
set d_min_pitch = 28
set d_min_boost_gain = 30
set d_min_advance = 50
set ff_interpolate_sp = ON
set ff_max_rate_limit = 100

# rateprofile 0
set roll_rc_rate = 135
set pitch_rc_rate = 135
set yaw_rc_rate = 120
set roll_expo = 15
set pitch_expo = 15
set roll_srate = 75
set pitch_srate = 75

# end the command batch
batch end
```

**Setup / Versions**
SPRACING H7 Nano with Aikon AK32 35A on different Frames

**Additional context**
I added a log to see whats wrong

[BTFL_BLACKBOX_LOG_Mark1_20190814_091724.txt](https://github.com/betaflight/betaflight/files/3500296/BTFL_BLACKBOX_LOG_Mark1_20190814_091724.txt)

etracer65, 2019-08-14:
Looks like a failing gyro. Yaw axis from the gyro is going crazy above a given vibration threshold. Generally the failure mode for ICM gyros.

TheIsotopes, 2019-08-14:
@etracer65 Are you sure that the gyro is broken?
the fc is absolutely new. Did you look at the logfile?

etracer65, 2019-08-14:
Yes, looking at the log file is how I determined it. Sometimes gyros fail - even when new. Yaw gyro is reporting impossible values - changes in DPS from very negative to very positive in extremely short periods of time. Something that's not physically possible. Also appearing on only one axis - agin not physically possible. Typical behavior of a failing ICM gyro.

[image]

TheIsotopes, 2019-08-14:
I did not want to admit it, because the FC is new.

thanks again 👍 ... closed now

hydra, 2019-08-14:
hey @TheIsotopes please contact your retailer/reseller so this can be investigated further if required.  replacement/repair is possible if there's a manufacturing fault, and gyro replacement is possible in crash-damage situations.

TheIsotopes, 2019-08-16:
Now i reopen this issue :-((
That can not be coincidence now. I tested a second new SPracing H7 Nano that has the same problem.

[BTFL_BLACKBOX_LOG_Mark1_20190816_130844.txt](https://github.com/betaflight/betaflight/files/3509302/BTFL_BLACKBOX_LOG_Mark1_20190816_130844.txt)

@hydra  **I'm very mad ... please kick your quality manager out.**

hydra, 2019-08-16:
@TheIsotopes seems very odd that you have the issue on two boards.  I saw your email. via the shop and have replied to it.

I understand you are frustrated, but please keep the conversation here civil and we'll get to the bottom of it.

I've not had any other customers have any issues with their NANO boards so I dont think there's a systemic issue with the hardware.

TheIsotopes, 2019-08-18:
> I've not had any other customers have any issues with their NANO boards so I dont think there's a systemic issue with the hardware.

@hydra I can not imagine why just calculated for me two new nano in a row do not work.
I have already built so many quads where I never had problems with the FC.

What could it be in your opinion that the two do not work?

nerocide, 2019-08-18:
@thelsotopes, this happened to me once, yaw twitching and drifting. I'm using mode 2 on a Qx7s. I finally discovered a falling gimbal on the radio.
But it drove me crazy for some time, I'd try soft mounting, BF yaw dead band, changing FC, motors.
Check your yaw channel behavior on receiver tab while using throttle just to be sure.

TheIsotopes, 2019-08-18:
@nerocide I have several quads that i fly with the same radio.
I would have noticed this on other quads.  imho this can not be due to the stick gimbal.
also all my fc are soft mounted.

bizmar, 2019-08-18:
If you have Blheli_32 ESCs try raising or lowering the PWM frequency. That often fixes yaw twitch problems. I don't know why exactly, other than it presumably reduces spikes that somehow get to the gyro.

hydra, 2019-08-18:
@TheIsotopes can you try it in a different frame?  try changing PWM frequency as @bizmar suggests first though. record logs at different pwm frequency settings and post them so we can follow your progress.

bizmar, 2019-08-18:
Yes, if someone could get some scope data on the PWM "fix" that would be great... I haven't seen any. It fixed this exact issue so many times that I can't say it's a fluke. It's been suggested that it has to do with aliasing, and timer resolution on certain ESCs. [Source]( https://www.youtube.com/watch?v=DDSv5K-Gkeg&feature=youtu.be&list=PLwoDb7WF6c8kXOyPdBog1wtRcxnXMasUb&t=53)

Some suggestions, if you have it at 24k (default) try 48k first, then something in between like 35k, and lastly 16k.

jflyper, 2019-08-18:
I'm very interested if fiddling with motor PWM frequency affect the extreme gyro reading on the specific axis.

TheIsotopes, 2019-08-18:
@bizmar you are my hero 👍 ;-)
you were right with the pwm frequency ... I would never have thought of that
I think now we have all learned something about it

@hydra @jflyper now it's time for the devs to find out why
the question is now it's up to betaflight or blheli32

etracer65, 2019-08-18:
I'm extremely surprised by this. I would think this has to be an electrical noise problem feeding back into the flight controller from the ESCs affecting the gyro power supply.

@TheIsotopes Do you have any additional capacitors added to your build? If not it would be an interesting test to add one.

It's definitely a hardware problem from Betaflight's perspective as the erroneous data IS coming from the gyro. Whether the problem affecting the gyro is electrical (noise in power) or mechanical (vibrations affecting the MEMS gyro) I don't know.

jflyper, 2019-08-18:
> @TheIsotopes Do you have any additional capacitors added to your build? If not it would be an interesting test to add one.

A tiny bypass cap on the gyro Vdd is worth experimenting.

We had an issue with ICM gyros fed by noisy Vdd in the past.
(The issue was induced by noisy sw reg, though.)

TheIsotopes, 2019-08-18:
@etracer65 
> @TheIsotopes Do you have any additional capacitors added to your build? If not it would be an interesting test to add one.

So far I had never soldered extra capacitors on the esc's ... but mostly one 470µF on the bat solder pads from the pdb .... currently i have not soldered a cap on this quad

@jflyper 
> A tiny bypass cap on the gyro Vdd is worth experimenting, too.

I don't want to experiment with that

hydra, 2019-08-18:
the hardware has two bypass caps, same as on H7EXTREME and H7ZERO where there are no issues.  same switching regs and LDOs used. regulator circuits physically far from gyro power supply by design too.  same batch components for all parts.  im thinking the gyros are damaged somehow.  i think I need to test these boards on a scope to see if I can spot any difference between these two boards and a known-good board, would be interesting for sure.

deffo try an extra cap first in case the ESC is very noisy.

@jflyper when you say tiny, what do you mean exactly, size or capacitance?  at what frequency did you have noise that your cap needed to remove and what value did you use in your case?

hydra, 2019-08-18:
oh, @TheIsotopes i missed your post about the results of the PWM frequency change.  very interesting, the gyros are deffo not faulty then.

I'll point people at this thread if anyone else contacts me with the same issue.

The other FCs you tested are physically different size wise so will have different vibration characteristics for sure.  Wondering what else we can all learn from this with an eye towards the future, recommendations anyone?

jflyper, 2019-08-19:
> recommendations anyone?
Two FCs stacked, one under question and another without the issue. The later runs bb only.

jflyper, 2019-08-19:
@TheIsotopes 
Values of bypass caps should be determined by nature of noise present. Do you have an oscope?

mikeller, 2019-08-19:
Could this be physical proximity of ESC power lines to the gyro on the flight controller? Can you post pictures of the build, @TheIsotopes?

TheIsotopes, 2019-08-19:
@jflyper Unfortunately, I do not have an oscope.

@mikeller here the picture ... I hope this is enough

[image]

hydra, 2019-08-19:
looks like the wires are along way from the gyro which is close to the centerline of the PCB on the left hand side.  what you could do is insert a small rf shield below the FC to see if it makes a difference. kinda curious.

mikeller, 2019-08-19:
Potentially a piece of tinfoil, covered with tape on both sides to avoid shorts if it comes loose.

TheIsotopes, 2019-08-19:
@Hydra The closest cables to the gyro are 5V / LED and Buzzer.

hydra, 2019-08-21:
Should we close this issue now that a workaround has been found and there's no software issue to solve?  

We can continue to discuss here as-required once you have time to follow up on suggestions.

TheIsotopes, 2019-08-26:
I've experimented a bit in the last few days.

Now that I've increased the pwm frequency to 32kHz, the twitching / twisting on yaw has been eliminated, but I have noticed increased noise on roll.

So I have the fc installed in another frame with blheli_s esc. twitching / twisting was eliminated without further adjustments on esc's, but also on this frame I can notice the same increased noise on roll. Any ideas, on two different frames the same increased noise on roll?

automatt1c, 2019-08-31:
This is almost definitely an issues with electrical noise. I always add at least a 470µF low ESR Cap on all of my builds. The PWM frequency can help alleviate noise depending on the noise frequency. Extra noisy builds i will add a small 100-220µF cap directly to the flight controller 5v power. As the gyros typically get the power from the 5v and step down to 3v for the gyro.   

its really a gamble with motors and ESC's some are just noisy. Frames can have an effect on electric noise as well. As the frame vibrations can have a cascade effect and cause more electrical noise because lack of gyro filtering. 

@mikeller Copper foil can do wonders on the bottom of the FC or top of ESC's (if its a 4in1) as well.

mikeller, 2019-08-31:
@automatt1c: Copper foil sounds great. Do you have any recommendation on where to get it from?

etracer65, 2019-08-31:
I've used this to shield the GPS from the rest of the electronics in an old DJI Phantom.

https://www.amazon.com/gp/product/B00K5RKD00/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Need to be careful though as the adhesive is supposed to be conductive.

bizmar, 2019-09-01:
Asked Ryan Harrell (QuadMcFly) about it, here's his reply:

> I've seen it on quite a few setups actually, not unique to the H7
It's an electrical noise filtering issue
typically means that there is insufficient care taken on the low voltage power supply side.  Increasing the capacitance in that area can help a good deal. I've also seen it where there is actually some fault in the SMD work. Reflowing with a SMD hot air station has solved it in several cases.
If not adding a tantalum cap (maybe 100uF 3.6v or so) on the 3.3v line will typically clear it up.
Also setting the PWM rate on the ESC reduces the overall noise floor, so this is why it often works as a solution.
48khz is typically the lowest electrical noise.

bizmar, 2019-09-01:
For board designers: I would look at FCs designed to only run 16 / 32k gyro speeds (you know which ones), ignoring everything else, look at the amount of filteirng / capacitance for the gyro. This may be a clue why ICM gyros preforms so badly on BF boards. I'm not a FC designer or an electrical engineer. I may be completely wrong on this, take it with a grain of salt. :)

hydra, 2019-09-01:
All the H7 boards already have a tantalum 100uF cap on the 3v3 line.

TheIsotopes, 2019-09-02:
ok, I'll close here ... so far everything is clear

closed 2019-09-02 by TheIsotopes