betaflight/betaflight #8729: Softserial as one port

voodoo614, 2019-08-17:
It is possible to implement softserial to act as one port. For example to use for GPS, Crossfire or Runcam Split which requires both TX and RX from a single UART.

Resource serial_tx 11 ###
Resource serial_rx 11 ###

Where the above can act as one port. 

It was suggested here by jflyer. 

https://github.com/betaflight/betaflight/pull/5140

mikeller, 2019-08-17:
@voodoo614: Not sure what you are asking for - the current softserial implementation supports all of the modes that hardware UARTs support, and can definitely be used to run GPS / Runcam / Crossfire (not recommended due to reliability requirements).

voodoo614, 2019-08-17:
I was under the impression that softserial has to be either TX or RX. And using both TX and RX for the same softserial wasn't possible. Ok, thank you for the clarification.

closed 2019-08-17 by voodoo614

jflyper, 2019-08-20:
@voodoo614 
There's one limitation of the full-duplex (simultaneous TX and RX) support on software serial.
That is, if TX and RX are serviced by a single timer, then RX activity will clobber active TX if any.
This is caused by how the software serial use timer channels, and applications that require continuous traffic on both TX and RX should avoid a software serial in the particular configuration.

voodoo614, 2019-08-20:
@jflyer. Thank you for the information about the timer. I understand that it is not recommended for anything continuous or important. I am inquiring just for knowledge. But there are situation, where a FC does not even have a full UART.

mikeller, 2019-08-21:
@jflyper:

> There's one limitation of the full-duplex (simultaneous TX and RX) support on software serial.

I wasn't aware of this. Considering that, in the case of softserial TX and RX are using different pins we actually have two timers at our disposal, it should be possible to use them to do full duplex. Are there any road blocks for this, or will this be a lot of extra effort?

jflyper, 2019-08-21:
@mikeller 
Code is ready for true full-duplex operation if two different timers are assigned to each of RX and TX pins. I believe I've tested all cases of simplex-RX, simplex-TX, half-duplex, single-timer-full-duplex and dual-timer-full-duplex case, but things might have broken since then.

https://github.com/betaflight/betaflight/blob/d746b4d3e782dc0ae668c75029940e076b10c55b/src/main/drivers/serial_softserial.c#L321-L323

https://github.com/betaflight/betaflight/blob/d746b4d3e782dc0ae668c75029940e076b10c55b/src/main/drivers/serial_softserial.c#L508-L521