betaflight/betaflight #8735: Issue binding Lumenier Lux F7 with Spektrum SPM4649T

henrytseng, 2019-08-18:
**Describe the bug**
Attempting to set Spektrum SPM4649T SRXL into binding mode using firmware CLI

```
set spektrum_sat_bind = 0 
set spektrum_sat_bind_autoreset = ON 
save
```

does not appear to be working.  Physical binding appears to work correctly.  

**To Reproduce**
Steps to reproduce the behavior:
1. Soldered Lux F7 TX, GND, 4V5 to SPM4649T Serial, GND, 4V.  
2. Plug USB, run Betaflight
3. Configure UART2 for Rx serial
4. Set `spektrum_sat_bind` mode to 9
5. Save and reboot

**Expected behavior**
Expecting receiver to be in binding mode.  Light on module should be blinking fast to indicate binding @ 11ms.  

**Flight controller configuration**
Using Lumenier Lux F7 FC.  Connected receiver through UART2_TX.  Spektrum 2048/SRXL serial receiver provider.  Serial-based receiver mode.  

```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: FREE
A04: ADC_EXT
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_RSSI
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: USB_DETECT
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: SDCARD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Flight controller: Lumenier LUX F7 (bought from GetFPV)
 - Spektrum SPM4649T SRXL - SH808X
 - Betaflight: 3.0.5 (target MATEKF77SE)
 - Receiver on T2, 4V5, GND
 - PDB through port

**Additional context**
Would love to understand how to fix this issue?

jflyper, 2019-08-18:
@henrytseng
Did you set `set spektrum_sat_bind` to `0` or `9`? You have contradicting statements.

I'm also interested in how you set `serialrx_halfduplex`.

In any case, please provide a requested information.

AndersHoglund, 2019-08-18:
Are you sure the Rx is powered from USB? 
Try this:
1. Connect USB.
2. set spektrum_sat_bind = 9 and save in BFC cli.
3. Disconnect USB.
4. Power up from a battery.
5. Press bind in Tx.

henrytseng, 2019-08-18:
> You have contradicting statements.

My mistake, I cut and paste from some doc.

henrytseng, 2019-08-18:
> Are you sure the Rx is powered from USB?

Did try it powered from battery.  Receiver was not in binding mode but I did try binding from my transmitter.

henrytseng, 2019-08-20:
Not sure why this issue was closed?  Can we reopen?  

`set spektrum_sat_bind` is set to `9`

mikeller, 2019-08-20:
@henrytseng: It was closed because you did not supply the information you were asked to supply when opening this issue.

henrytseng, 2019-08-20:
Do you have what you need now?

mikeller, 2019-08-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

henrytseng, 2019-09-19:
I believe I did that above.

closed 2019-10-26 by stale[bot]