betaflight/betaflight #8737: From build #1791 on smartport telemetry doesn't work on NOX target with software serial

joelucid, 2019-08-18:
The issue seems to be caused by resources not correctly being available in all parts of the resource system. Eg if I do a ``resource`` this is the output:

```
# resource
resource BEEPER 1 C13
resource MOTOR 1 A01
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 B01
resource PPM 1 B10
resource LED_STRIP 1 A00
resource SERIAL_TX 1 B06
resource SERIAL_TX 11 A02
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 11 A02
resource INVERTER 2 C14
resource LED 1 A04
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 B10
resource ADC_BATT 1 A05
resource BARO_CS 1 A09
resource FLASH_CS 1 A15
resource OSD_CS 1 A10
resource GYRO_EXTI 1 A08
resource GYRO_CS 1 B12
```

Compare to ``resource show``:

```
# resource show
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: MOTOR 1
A02: SERIAL_TX 11
A03: SERIAL_RX 2
A04: LED 1
A05: ADC_BATT
A06: FREE
A07: MOTOR 2
A08: GYRO_EXTI
A09: PREINIT 2
A10: PREINIT 4
A11: USB
A12: USB
A15: FLASH_CS
B00: FREE
B01: MOTOR 4
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: FREE
B07: FREE
B08: MOTOR 3
B09: FREE
B10: FREE
B12: GYRO_CS 1
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: BEEPER
C14: INVERTER 2
C15: FREE
```

Clearly for example all resources for serial 1 did not make it into the list used for ``resource show``.

This works as workaround:

```
    const timerHardware_t *timerRx = timerGetByTag(tagRx);
//timerAllocate(tagRx, OWNER_SERIAL_RX, RESOURCE_INDEX(portIndex + RESOURCE_SOFT_OFFSET));
    const timerHardware_t *timerTx = timerGetByTag(tagTx);
// timerAllocate(tagTx, OWNER_SERIAL_TX, RESOURCE_INDEX(portIndex + RESOURCE_SOFT_OFFSET));
```

jflyper, 2019-08-18:
Which commit is ci#1791?

etracer65, 2019-08-18:
Looks like https://github.com/betaflight/betaflight/pull/8666

mikeller, 2019-08-19:
@joelucid: Ha, now I see where the problem is - NOX defines both TX and RX for softserial 1 to be on PA1, thus using the same timer. I didn't expect this to be the case in the target configuration, as this obviously makes the softserial unusable for all communication requiring separate RX and TX lines.

As a workaround until a fix is pushed you can set the `SOFTSERIAL_RX 11` resource to any unused pin (e.g. PA0), and it should work again.

Timer sharing is a special case, as pretty much all other cases of resources sharing a timer will break. My change started tracking ownership of timers, and thus blocking resources from using a timer if it is already used, preventing resource conflicts. Not sure what the best way is to make this work, as I am not sure how the timer sharing is implemented for softserial. My hunch would be to change softserial timer allocation to first check if the timer in question is already owned by the 'counterpart' of the softserial in question, and only allocate it if not, and just get a reference to it otherwise. @jflyper, do you have any ideas?

Also, is anybody aware of other instances where timers are shared?

@joelucid: Fyi:
- the difference between `resource` and `resource show` is that `resource` shows all available resource assignments, with some pins potentially claimable by multiple functions (e.g. sharing between `MOTOR` and `LED_STRIP` on some board, where `resource show` only lists the resources that are actually used by a function, i.e. got allocated when the function was initialised because it is configured;
- `serial 1` is the first hardware UART, and I assume it does not show because you have not configured any protocol / driver to use it (having a full config `diff` would help with diagnosing these things.

joelucid, 2019-08-19:
Ah ok - yeah all that makes sense. Great that we’ve got it figured out! And yes Serial 1 is not in use.

mikeller, 2019-08-19:
I went ahead and created #8741 as on second look this seems to be trivial.

@joelucid: Can you confirm that this is working please?

@jflyper: From the looks of it, `IOInit()` seems to be acting lenient, and allowing the allocation of resources even if they are already allocated. I don't know if there is any need for this, do you think we should look into switching this to a 'first come first serve' approach for pins?

jflyper, 2019-08-19:
@mikeller I've been noticing the leniency of the `IOInit` ever since `new IO` had been introduced, and once wrote a code to make sure a pin is not claimed by someone else. We can make it less lenient, but I'm not sure if we have codes that intentionally override previously claimed pins.

mikeller, 2019-08-20:
@jflyper: Agreed re the risk. But I think that the risk of unintended and hard to find bugs in the current implementation is worse - if one owner claims a pin, then does some configuration on it, and then another owner 'steals' the pin, and fails to perform configuration that completely un-does the previous owner's initialisation, we'll end up with a pin that is in an unintended state.
The only way out of this is to start to be strict about pin ownership, and then detect cases of multiple initialisation and change them to have the first owner `IORelease()` the pin before the next owner claims it.
As a mitigation of the risk, my proposal is to not do this now, but wait until early after the 4.1 release, to maximise the time available to deal with the fallout.

mikeller, 2019-09-05:
@joelucid: I think this was fixed in #8741 Can this be closed?

joelucid, 2019-09-05:
Sure thing

closed 2019-09-05 by joelucid