betaflight/betaflight #8739: Spedix GS45A 4in1, Hyperlite F4 OSD and flyaway issues

c0debreaker, 2019-08-19:
I've use 4.0.4 and 3.5.7 and I've encountered 2 flyaways now. Glad I was in a huge open field. My quad suddenly flew away and went up really high that I had to disarm it at 150ft. It broke the arms. Was it possible that the cause of my flyaway is the tight mounting of flight controller? I wasn't aware about `soft mounting` until I was told yesterday.

I'm still confused why suddenly motors would go full throttle. Like yesterday, I was hovering my quad between 5 to 10ft. I moved the throttle really really tiny but OMG, it flew towards the sky! I vanished from my vision in less than 2-3 seconds. My estimate is that it reached 150-200ft in 2-3 seconds only.

Few days before this happened, I was wondering why my quad would fly right away immediately after arming. The throttle was even at 0.

I rebuilt everything tonight. I soft mounted the flight controller. I'll test it tomorrow.

I still would like to hear possible causes of this. Thanks.

jflyper, 2019-08-19:
You are not providing the requested information.

c0debreaker, 2019-08-21:
Oh, I'm so sorry jflyper. I got so busy at work. I think my issue fixed. The problem was that I mounted the flight controller so tight that it caused it to receive all vibrations. I was told that if gyro receives a lot of vibration, it will cause unwanted throttle spike similar to what I experienced. I corrected the mounting of the fc by soft mounting it. I also soft mounted the motors. I flew the other day for 20-25 mins without fly aways.

closed 2019-08-21 by c0debreaker