betaflight/betaflight #8742: GPS Rescue Initial Altitude not respecting gps_rescue_initial_alt value

brunoleaomaia, 2019-08-19:
**Describe the bug**
I've set gps_rescue_initial_alt to 20 and when I change flight mode to GPS Rescue, the drone climbs  to 50m (default value)

**To Reproduce**
- Configure GPS Rescue on a Switch;
- Open CLI and do this commands:
   - set gps_rescue_initial_alt = 20
   - save
- Move Drone to 300m away;
- Change Switch to GPS Rescue Mode;

**Expected behavior**
The drone climb to 20m

**Flight controller configuration**
```
# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.4 Jun 30 2019 / 14:27:40 (f3a95efa3) MSP API: 1.41

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id 

# name
name MAIAFPV

# resources
resource MOTOR 1 B01
resource MOTOR 2 E11
resource MOTOR 3 B00
resource MOTOR 4 E09

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 8192 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 3 32 115200 57600 0 115200
serial 6 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1250 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 46 2 1700 2100 0 0
aux 3 13 1 1700 2100 0 0
aux 4 35 1 1300 1700 0 0

# adjrange
adjrange 0 0 0 900 2100 29 4 0 0

# rxrange

# vtx

# rxfail
rxfail 6 s 1900
rxfail 7 s 1000

# display_name

# master
set acc_calibration = 59,19,61
set mag_hardware = NONE
set mag_calibration = 70,-52,-88
set max_check = 1950
set rssi_channel = 8
set min_throttle = 1040
set motor_pwm_protocol = DSHOT1200
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1000
set gps_rescue_min_sats = 6
set pid_process_denom = 1
set transient_throttle_limit = 10
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_tim1 = 2561
set osd_tim2 = 2562
set osd_vbat_pos = 14722
set osd_rssi_pos = 14670
set osd_tim_1_pos = 14741
set osd_flymode_pos = 6519
set osd_throttle_pos = 6508
set osd_vtx_channel_pos = 2103
set osd_crosshairs_pos = 14542
set osd_current_pos = 6498
set osd_craft_name_pos = 2082
set osd_gps_speed_pos = 4151
set osd_gps_lon_pos = 4161
set osd_gps_lat_pos = 4175
set osd_gps_sats_pos = 4129
set osd_home_dir_pos = 4134
set osd_home_dist_pos = 4143
set osd_altitude_pos = 4135
set osd_debug_pos = 8354
set osd_pidrate_profile_pos = 2100
set osd_warnings_pos = 14602
set osd_avg_cell_voltage_pos = 14732
set osd_esc_rpm_pos = 129
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set debug_mode = D_MIN
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5695

# profile
profile 0

set feedforward_transition = 30
set iterm_relax_type = GYRO
set p_pitch = 30
set i_pitch = 90
set d_pitch = 27
set f_pitch = 120
set p_roll = 28
set i_roll = 84
set d_roll = 25
set f_roll = 120
set p_yaw = 30
set i_yaw = 90
set f_yaw = 30
set d_min_roll = 16
set d_min_pitch = 18
set d_min_advance = 0

# rateprofile
rateprofile 1

set thr_expo = 5
set roll_rc_rate = 80
set pitch_rc_rate = 82
set yaw_rc_rate = 80
set roll_expo = 5
set pitch_expo = 5
set yaw_expo = 5
set roll_srate = 84
set pitch_srate = 77
set yaw_srate = 80
set tpa_rate = 75
set tpa_breakpoint = 1400
set throttle_limit_type = SCALE
set throttle_limit_percent = 85

# end the command batch
batch end

# 
```

**Setup / Versions**
 - Kakute F7;
 - GPS M8N;
 - Barometer On (Working very well);

McGiverGim, 2019-08-19:
According to the wiki, the altitude will be the parameter or the max altitude recorded +15, the highest of both. Did you reach 35 meters while flying before enabling the rescue?

brunoleaomaia, 2019-08-19:
> According to the wiki, the altitude will be the parameter or the max altitude recorded +15, the highest of both. Did you reach 35 meters while flying before enabling the rescue?

No, I was flying under 20 meters

brunoleaomaia, 2019-08-19:
I've tried many times, all time reached 50m before start come back.

etracer65, 2019-08-19:
If you have the barometer enabled, then capture a blackbox log showing the behavior. Also DVR video showing OSD would be useful.

brunoleaomaia, 2019-08-19:
Barometer is working very well, I will make this video and blackbox soon... I need make some settings to record barometer in blackbox?

etracer65, 2019-08-19:
No, if the barometer is enabled and working then the altitude data will be logged.

theSkonk, 2019-08-20:
i have yet to see the barometer on any of my 6 quads who have one be any thing but rubbish.. i found that 70m was the only way for me to get reliable GPS rescue. remember its not about it coming home exactly on spec its about it coming home at all in a bad situation. if you want any kinda exact alt hold in your return to home/rescue then you need flight in your life... this function is literally the last best hope in a poor situation and nothing more

IllusionFpv, 2019-09-03:
In betaflight 4.1 you can set gps_rescue_alt_mode = FIXED_ALT to avoid this https://github.com/betaflight/betaflight/pull/8301

brunoleaomaia, 2019-09-03:
Attached LOG with Two tries of GPS Rescue, in both Altitude was more than gps_rescue_initial_alt (20m)

Video from this flight https://youtu.be/Ky7xYfoIPhw
 
[LOG00002.zip](https://github.com/betaflight/betaflight/files/3569761/LOG00002.zip)

etracer65, 2019-09-03:
Looks like it's working perfectly to me. GPS Rescue determines the initial climb altitude as the greater of the setting (which you have as 20m) or 15m above the highest measured altitude during flight. At 0:36 into the flight you reached 14.3m of altitude so we would expect the altitude setting to be ~30m. With a little overshoot because of the short climb period that's about what it climbed to. Then on your second attempt the previous max altitude was about 36m so we would expect to see around 51m (36 + 15). Again this is what happens.

Everything looks fine to me. In BF4.1 there are some other options to manage the resulting target altitude like setting an absolute limit rather than the max attained + 15 so you'll probably like that.

brunoleaomaia, 2019-09-03:
> Looks like it's working perfectly to me. GPS Rescue determines the initial climb altitude as the greater of the setting (which you have as 20m) or 15m above the highest measured altitude during [...]

Ok, but this is not clear in documentation. Thank you for your explanation. Where I can found the explanation to the new GPS params in 4.1?

etracer65, 2019-09-03:
What's not clear in the documentation? https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode#go-to-the-betaflight-modes-tab-and-add-a-switch-for-gps-rescue-mode-verify-that-the-mode-actually-gets-activated-of-course-no-props

> `set gps_rescue_initial_alt=[number]` (default is 50)
>
> This is the most important parameter. When Rescue Mode is activated, your quad will point home and try to climb to a safe altitude relative to your takeoff point. **This altitude will either be this parameter, or the maximum altitude recorded during flight +15m, whichever is highest.** I personally like to make it 70 or 80 meters.

brunoleaomaia, 2019-09-03:
> What's not clear in the documentation? [...]

It is not clear in which situation it goes to altitude form parameter (gps_rescue_initial_alt) or goes to maximum altitude recorded + 15m.

etracer65, 2019-09-03:
Not sure how it can be clearer.

> This altitude will either be this parameter, or the maximum altitude recorded during flight +15m, whichever is highest.

brunoleaomaia, 2019-09-03:
> Not sure how it can be clearer.
> 
> > This altitude will either be this parameter, or the maximum altitude recorded during flight +15m, whichever is highest.

Ok, thank you! Closing the Issue Now!

closed 2019-09-03 by brunoleaomaia