betaflight/betaflight #8743: Matek F405-STD with FrSky D4R-II receiver PWM RSSI problem

grayver, 2019-08-20:
**Describe the bug**
I have Matek F405-STD flight controller with FrSky D4R-II receiver connected via PPM mode. RSSI signal wire from receiver is connected to Rssi pad on the FC board. PPM channels work well, but RSSI level is always 0% (in Configurator and OSD). Either with turned on or turned off transmitter.

**To Reproduce**
Connect FrSky D4R-II RSSI wire to Matek F405-STD Rssi pad and observe RSSI value in Configurator/OSD.

**Expected behavior**
RSSI value should be 100% with transmitter turned on.

**Flight controller configuration**
```
# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.5.6 Feb 16 2019 / 13:20:25 (c28ca22f9) MSP API: 1.40

board_name MATEKF405
manufacturer_id 

# name
name Alfa Fighter

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource LED_STRIP 1 B08

# mixer

# servo

# servo mix

# feature
feature -RX_SERIAL
feature RX_PPM
feature MOTOR_STOP
feature LED_STRIP

# beeper

# beacon

# map

# serial
serial 1 0 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led
led 0 7,12:S:AW:0
led 1 7,13:S:FW:0
led 2 7,14:S:L:0
led 3 7,15:S:L:0
led 4 8,15:S:L:0
led 5 8,14:S:L:0
led 6 8,13:S:FW:0
led 7 8,12:S:AW:0

# color

# mode_color

# aux
aux 0 0 2 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 28 1 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = 16,19,10
set mag_hardware = NONE
set min_check = 1000
set max_check = 2000
set min_throttle = 1150
set use_unsynced_pwm = OFF
set motor_pwm_protocol = ONESHOT125
set motor_pwm_rate = 480
set deadband = 4
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2103
set osd_rssi_pos = 2137
set osd_tim_2_pos = 2092
set osd_flymode_pos = 2081
set osd_ah_sbar_pos = 271
set osd_ah_pos = 2190
set osd_current_pos = 2497
set osd_mah_drawn_pos = 2505
set osd_altitude_pos = 2517
set osd_warnings_pos = 2410
set osd_nvario_pos = 2328
set vtx_band = 3
set vtx_channel = 5
set vtx_power = 4
set vtx_freq = 5885
set vcd_video_system = PAL

# profile
profile 0

# rateprofile
rateprofile 0
```

**Setup / Versions**
 - Matek F405-STD from Banggood with Betaflight 3.5.6;
 - FrSky D4R-II as RX, Turnigy 9X with er9x firmware and FrSky DJT module as TX;
 - Matek FCHUB-VTX.

**Additional context**
Recently I decided to upgrade my old quad and give it a second chance, as I haven't touched it for ~5 years. My old setup included Naze32 (rev1), MinimOSD and FrSky D4R-II. MinimOSD with Atmel chip hadn't chance to decode high frequency PWM RSSI signal from FrSky receiver, so I used PWM->ADC converter as it was normal practice for this setup and it worked quite acceptable.

In new setup I don't use any PWM->ADC converter, as I suppose it should be supported by Betaflight out of the box according to feature list:
> RSSI via ADC - Uses ADC to read PWM RSSI signals, tested with FrSky D4R-II, X8R, X4R-SB, & XSR

STM F4 is modern high-perfomance chip, so it shouldn't be a problem to decode RSSI PWM signal from receiver. However, it doesn't work :(

I tried to search solution through rcgroups forum threads, but all advices lead to one idea: "Use PWM->ADC converter". Is it really not supported or I'm doing something wrong?

jflyper, 2019-08-20:
> RSSI via ADC - Uses ADC to read PWM RSSI signals, tested with FrSky D4R-II, X8R, X4R-SB, & XSR

This means that PWM encoded RSSI signal is first converted to a voltage level that ADC can then convert to a digital value. You have to use the converter.

There has been an attempt to directly decode PWM encoded RSSI without a timer, but it was abandoned due to severe timing constraint on the PWM signal that some of the receivers imposes.

Ref: #2292

etracer65, 2019-08-20:
You'd be much better off replacing the receiver with a more modern one - like a R-XSR - and use SBUS. Smaller, lighter, and 1/3 the latency (faster response).

grayver, 2019-08-20:
@jflyper Ohh.. it's a pity, that such useful feature was abandoned. I'm a newbie in STM architecture, but is there any way to use hardware accelerated UART port to decode this RSSI PWM signal? Anyway, I will use my PWM->ADC converter again for now..

@etracer65 Thanks, I know that my receiver is little bit outdated. But it still works and I don't want to toss it out. Buying new receiver leads to buying new TX module or even more - new TX. This a little bit expensive for now.

jflyper, 2019-08-21:
@grayver Not with UART, but with a timer, it is a easy task to do. The problem with the timer based approach is that timers are a scarce resource for most flight controller boards. Even with an unused timer, it is not always the case that it is broken out for wiring.

grayver, 2019-08-21:
@jflyper Ok, I see. But why it couldn't be done configurable? As GPS or another peripherial device - if I have free port (and timer?) then just solder to appropriate pad and assign appropriate port to Frsky RSSI PWM input?

mikeller, 2019-08-21:
@grayver: Feel free to open a pull request for these changes.

jflyper, 2019-08-21:
@grayver As I wrote in my previous comment, timer is a scarce resource, and there are many other applications that can only be done with timers. The PWM-encoded RSSI input, on the other hand, can be done with ADC input with a help from a tiny RC filter, so the ADC method is preferred. My previous PR to use general GPIO input was to eliminate the RC filter, but had some trouble providing accurate reading across all platforms. The PR may be resurrected as F3 which I had trouble with is now EOL, but there isn't much needs for the method these days.

grayver, 2019-08-21:
Ok, I got it. Let's then close this issue and I will use my old PWM->ADC converter :)

closed 2019-08-21 by grayver