betaflight/betaflight #8747: Sensor lost - Fault telemetry FC

wtrevine, 2019-08-23:
Hello...
My taranis has constantly alarmed sensor lost. I searched in various forums, several people say that is by the proximity of the radio, when it is in flight, it does not. But I found that this is not the case, the alarm (sensor lost) occurs when the cpu temperature is above 40ºC. I did several tests of time, distance, with fans, and only when the cpu is above 40°C the sensor is lost. All flight controller telemetry is lost.
I do not lose the telemetry of the receiver at the time of the lost sensor, I still have RSSI and receiver voltage.
The microcontrollers do not heat up, but there are nearby voltage regulators and VTX that generate heat and raise the temperature of the flight controller. On a table, without ventilation, it will happen.

Taranis QX7 - OpenTX 2.2.4
XSR - 180322
SKYSTARS F405 AIO - Betaflight 4.0.5

etracer65, 2019-08-23:
You either have a wiring issue or a problem with the receiver. Maybe the receiver is getting too hot. The lost telemetry warning is between your receiver and transmitter and has nothing to do with the flight controller.

Also 40C core temp is completely normal for a F4 and not a problem.

klutvott123, 2019-08-23:
Could be another occurrence of the bytestuff bug. I haven't tested SBUS/Sport on the XSR but the FPORT firmware had the bug. To test this you could try FPORT with this firmware: 
FCC: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-436987665
LBT: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-437283806

FPORT setup guide is in the BF wiki.

jflyper, 2019-08-24:
If the issue is actually linked with core temperature value, I would suspect that there is a problem somewhere in the interpretation/handling of the problematic value, most likely in the receiver software.

klutvott123, 2019-08-24:
> 
> 
> If the issue is actually linked with core temperature value, I would suspect that there is a problem somewhere in the interpretation/handling of the problematic value, most likely in the receiver software.

That's exactly what was happening with the bytestuff bug. The receiver discarded a frame if any value was 125. Frsky fixed it for FPORT but never officially released the firmware for the XSR.

wtrevine, 2019-08-25:
Problem solved!!! Thank you gentlemen!
I upgraded my XSR to the version with Fport as mentioned by klutvott123. Thank you so much!

closed 2019-08-25 by wtrevine