betaflight/betaflight #8753: OSD direction to home wrong direction

siamect, 2019-08-25:
**Describe the bug**
The home direction arrow points in the wrong direction.
If I fly away from the starting point, the arrow points forwards (up).
If I fly towards the starting point the arrow  points backwards (down).
If I fly with the starting position to my left the arrow points left which is correct.
If I fly with the starting position to my right the arrow points right which is correct.

**To Reproduce**

Hmm... fly!

**Expected behavior**
I expect the arrow to point home to the starting point like it does on my two other quads with the BF 4.0.4. they are using the DYS F4 Pro and the Matek F722-SE F7
If I fly away from the starting point, the arrow should point backwards (down).
If I fly towards the starting point the arrow  should point forwards (up).

**Flight controller configuration**
 ```  
# diff

# version
# Betaflight / FURYF4OSD (FY4O) 4.0.5 Aug  1 2019 / 11:57:21 (839661b20) MSP API: 1.41

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 1 1300 2100 0 0
aux 2 46 1 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 7 s 1000

# display_name

# master
set acc_calibration = -65,-45,-101
set rssi_channel = 8
set serialrx_provider = CRSF
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set current_meter = ADC
set vbat_scale = 106
set ibata_scale = 200
set ibata_offset = 50
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 40
set gps_rescue_descent_dist = 100
set gps_rescue_throttle_hover = 1380
set gps_rescue_min_sats = 6
set pid_process_denom = 2
set osd_vbat_pos = 2085
set osd_rssi_pos = 2318
set osd_link_quality_pos = 370
set osd_vtx_channel_pos = 2402
set osd_current_pos = 2091
set osd_mah_drawn_pos = 2156
set osd_gps_speed_pos = 2413
set osd_gps_sats_pos = 2100
set osd_home_dir_pos = 2252
set osd_home_dist_pos = 2254
set osd_altitude_pos = 2421
set vtx_channel = 4
set vtx_power = 2
set vtx_freq = 5800
set vcd_video_system = NTSC

# profile
profile 2

set p_pitch = 62
set p_roll = 60

# rateprofile
rateprofile 0

# end the command batch
batch end
  ``` 

**Setup / Versions**
Diatone Mamba F405 (FURYF4OSD) tested with BF 4.0.4 and 4.0.5

etracer65, 2019-08-25:
If your GPS unit has a compass (magnetometer), then make sure it's oriented correctly and properly calibrated - or disconnect the compass entirely as it's not needed for GPS Rescue.

Otherwise check and fix the settings on your GPS module. These kinds of problems can be caused by using too slow a refresh rate and/or baud rate. Recommend at least 5hz and 57600 baud. Usually you need to use the UBLOX U-center software to change settings in the module.

jflyper, 2019-08-26:
Wasn't this has something to do with OSD glyph set?

@siamect 
Your description is incomplete as it's missing which way you are facing.

McGiverGim, 2019-08-26:
> Wasn't this has something to do with OSD glyph set?

In theory the arrow symbols haven't changed its position with the latest refactors.

siamect, 2019-08-26:
> If your GPS unit has a compass (magnetometer), then make sure it's oriented correctly and properly calibrated - or disconnect the compass entirely as it's not needed for GPS Rescue.
>
> Otherwise check [...]

Thanks for responding. 
So I have magnetometer connected but it is disabled as it is not supported by betaflight in the FURYF4OSD target. (everything worked fine in iNav)
Just to be sure I anyway turned the gps module 180 degrees but the same problem as expected.

I set the baudrate to auto and it gets 115200. Exactly the update Hz I don't know but If I fly for a long time in the same direction I still have the same problems.

siamect, 2019-08-26:
> Wasn't this has something to do with OSD glyph set?
> 
> @siamect
> Your description is incomplete as it's missing which way you are facing.

The glyph set is the same for all my quads. Only the betaflight target differs.
I don't use any head tracker  so the way I'm facing has no significance.

siamect, 2019-08-27:
> significance.
>
> I meant "facing" by "which way the craft is facing" (not the operator).

But that info is there...

jflyper, 2019-08-27:
> I don't use any head tracker so the way I'm facing has no significance.

I meant "facing" by "which way the craft is facing" (not the operator).

jflyper, 2019-08-27:
> If I fly with the starting position to my left the arrow points left which is correct.

This is my interpretation:
If you are flying toward left from the starting point facing left, then arrow should point to down.
So the statement suggests that your craft was not flying to the direction it is facing.

siamect, 2019-08-27:
> @siamect It would help if you provide DVR video of a flight showing the
> problem.

Ok I will do that. The recordings I have at the moment is a bit confusing but I will be flying hopefully in a few days to get clean situations that show how it works.

etracer65, 2019-08-28:
@siamect As part of your test, please also add the "Numerical heading" element to your OSD. Try to keep track of your direction so we can see if the heading is accurate. For example, fly north and the heading should be new 0. This will help to differentiate a problem caused by an incorrect heading vs. a calculation to home point.

Also I would recommend disabling the magnetometer for the tests as if it's improperly calibrated it will mess up the heading. Turn it off on the Configuration tab.

siamect, 2019-08-28:
Thanks for the info. I will try to get it done over the weekend.
/Martin

siamect, 2019-08-31:
I put a video here https://youtu.be/HFyGLvT7nHo to clarify the symptoms better.

etracer65, 2019-08-31:
The main issue I see is that the quad that's having the issue has quite a poor GPS fix (low number of satellites) whereas the other has significantly more satellites and a more stable 3D fix.

siamect, 2019-09-04:
So here is the reason and the solution: I had the wrong font loaded in the FC. Why? 
Well it turns out that just after I load the correct font, the FC will reboot. After that the old (iNav) font is still there! So although I have reloaded the fonts many times, the FC looses it immediately and I didn't notice that until LOLO VPP insisted that the arrows don't look like the bold font in BF.

Solution/workaround:
So I plugged in a battery, connected the cable to the computer and reloaded the font again and voila! The new BF Bold font was loaded and it remained in the FC after power off/on again.

So the immediate problem for me is resolved and I say Thank You to all help I got both here and on YouTube!
Still I think this is a bug...

IvoFPV, 2019-09-04:
@siamect You have to have battery plugged in when uploading a new font. That's always been like that and it is not a bug.

mikeller, 2019-09-04:
@siamect:

> Still I think this is a bug...

Bug or feature, the reason for this is that the OSD chip is not powered from USB on this board, or that the USB power supply isn't strong enough to support flashing a font onto it. There is absolutely nothing that Betaflight can do to change this, take it up with the supplier of your board (who will most likely tell you that this is by design).

siamect, 2019-09-04:
So lets call it a hardware bug...

closed 2019-09-04 by siamect

Secable, 2019-09-07:
Thanks a lot man 😉
Have exactly same problem, and just change the police, all is working fine now.
So, yes, there is an issue with one of the police (don't know which) that put the arrow in wrong way.

Pollymer, 2020-03-03:
Hello everyone, I'm new to the forum, and wonder if anyone can help, I too have the problem of the arrow pointing the wrong way in betaflight gps rescue, my flight controller is matek 405std and I have a tbs m8 gps module, no magnometer or compass, I get the sats fine, the alt and distance reading is good, just the arrow will not point back to home, I wait for at least 8 sats and have tried with even more, Then arm and keep flying straight away from the home point but it still will not point home. the arrow on mine is different though to the one on saimects video, it looks like an actual arrow and not a V, was kind of hoping it would be a font problem also. The only way I have been able to make it point to home is to arm the quad, props off, and walk away with the quad, then if i  turn, yaw the quad say to the left 90 deg, only then will the arrow point home, I even re mounted the flight controller 90 deg left in the quad frame and it then was spot on for one power up, but on the next boot up of the quad it was back to its tricks again . Any help would be much appreciated, thanks.

vikasudasioyo, 2021-06-26:
Hi, Just to help some one here. i was using Mamba F405 Flight Controller with TBS M8.2 GPS. Initially i set baud rate to 9600 as it is  mentioned as default baud rate on official team blackship website. I got that gps symbol enabled on the top of betaflight. Was getting good gps satellites , When i was flying got correct distance back to home, but direction was always incorrect. Altitude measurement we also off. So i did a lot of reading what fixed my issue is setting baud rate to 115200. Learning: Try changing baud rate as well default may not always work correctly.  Thx

vikasudasi, 2021-06-26:
Hi, Just to help some one here. i was using Mamba F405 Flight Controller with TBS M8.2 GPS. Initially i set baud rate to 9600 as it is mentioned as default baud rate on official team blackship website. I got that gps symbol enabled on the top of betaflight. Was getting good gps satellites , When i was flying got correct distance back to home, but direction was always incorrect. Altitude measurement we also off. So i did a lot of reading what fixed my issue is setting baud rate to 115200. Learning: Try changing baud rate as well default may not always work correctly. Thx