betaflight/betaflight #8755: F74x DMA interrupting at half transfer complete without HTIE

jflyper, 2019-08-26:
I want to know if this rings a bell on someone.

I'm working on a DMA GPIO banging DShot motor code atm, and experiencing F74x DMA interrupting at half transfer complete without half transfer complete interrupt enable (HTIE) being set. That is, at DMA interrupt handler, (L/H)ISR only indicates HTIF but HTIE in CR is not set.

It doesn't happen on F405 and F722.

Any info on this? @ledvinap?

ledvinap, 2019-08-26:
@jflyper : Sorry, no idea, I'm not using F7 ... 
As for the bug - check errata, create minimalistic code to trigger it and ask ST

And in code, it should be trivial to check actual flag and ignore spurious interrupt ...

jflyper, 2019-08-26:
@ledvinap Thanks, I'll do ...

jflyper, 2019-08-31:
Hmm... I started to think this is caused by stream mixups.

closed 2019-08-31 by jflyper