betaflight/betaflight #8756: new flight modes for new people

ubanov, 2019-08-26:
I remember when I begin driving drones... a lot of crashes.

Currently when some of my friends want to try a drone, I use a small drone with brushed motors, because it's light and small, and then normally a crash doesn't mean to break anything.

Yesterday I was thinking about it and I have an idea: may be possible to create new flights modes for new babys.

The main mode could be something that will not permit the drone go below some height (let's say that I configure 2 meters... if drone goes below that height in this mode it will correct to 2 meters, or 2,5 meters in order to have a zone, instead of a single height). I think that using this mode with angle mode simultaneally, could be easy to implement, and very good way to a new baby to begin with drones. The Mode could be called: "Don't touch ground" :-D

As the drone will not permit to go below 2 meters, for landing this mode should be deactivated.

After that could be good to have another two modes: "take off mode" (drone go up to the height stablished), and "landing mode" (drone go down slowly until it take ground).

If you have landing mode, could be possible, that if mode "Don't touch ground" is configured, and throthle is at the minimum for some seconds, the drone enters in landing mode automatically. And in the other way, if the drone is in the ground, and "Don't touch ground" mode it's enabled, wait until throthel is touched for taking off

For all the modes the FC need to have barometer (or even a sonar?). I think that this will not work with GPS only.

IvoFPV, 2019-08-26:
@ubanov Problem is that current FCs don't have good enough barometers. Their readings are always wrong on every FC I have. You will often see the height rise or fall when the barometer module gets warmer or colder. This would make this feature unusable. GPS height is not accurate enough also, especially to hold exact 2 meters above the take off point. For 2 meters above the ground we would need to have some kind of a ultrasonic sensors like DJI drones have. Do you know anyone having those on normal 5 inch racing/freestyle builds? I believe you don't...

ubanov, 2019-08-27:
then you think that the only solution will be to use a sonar?

You are right that barometers are not exact enough... a possible solution could be to have a bigger height limit?

A sonar could be easy to install, but have two problems:
- it's a bit big 
- Sonar doesn't work on all the FC (currently I have 2 FC and I can't install the sonar on those FC)

I would install a sonar if it gives me some functionalities... a mode for landing could be a good complement for the Return to home functionality... I would install the sonar only for that.

RustyToad, 2019-08-27:
> Betaflight it's getting better every day. I don't understand why this functionality could not be included in betafligh...

For the same reason as you won't get a towbar on a Ferrari. It's not that a towbar isn't useful, and not that the Ferrari engineers couldn't design one, it's just not the right vehicle for it.

As etracer65 has said, Betaflight does not have any focus on this sort of flight mode. Even the "return to home" mode is kind of approximate, and rough round the edges. It will just sort of fly approximately towards you, and keep going (ie, if by luck it manages to be accurate enough, it will hit you). It is intended as a "get out of jail" card for a bad day, not a regular flight mode.

iNav and ardupilot both have much more extensive, and more refined, use of GPS, barometers, sonar, magnometers etc, and can perform extremely accurate autonomous and bounded flight modes. Betaflight has never intended to do this.

ubanov, 2019-08-27:
The first time I try return to home in Betaflight was 2-3 years ago... it didn't work for me.

Currently Betaflight have a return to home that really works... 

Betaflight it's getting better every day. I don't understand why this functionality could not be included in betafligh...

stawiski, 2019-08-29:
Betaflight is open source.
If you have an idea for a feature, why not write and test it out yourself?

atomgomba, 2019-08-31:
Simulators today are really great and provide good enough means to learn to fly in acro which then the pilot can easily adapt to real life. I think newcomers shall not be "protected" from having to take the effort to learn, and they can do it without breaking too many stuff.

Mateyhv, 2019-09-24:
I agree with all comments above. This feature is out of betaflights scope and way more complex than initialy suggested. Better look at Inav that already has something quite similar to your request.

closed 2019-10-31 by stale[bot]