betaflight/betaflight #8757: Motor remap not working on lux v2 betaflight 4.04

SupaCoop, 2019-08-26:
I've been trying to remap the motors on an old build that I dropped a lux v2 into and it isn't responding to it at all. I started with betaflight 4.04 and went back to 3.5.7 and still no response. It's a qav250 with ancient 12 amp simonk esc's running pwm for the esc protocol but that shouldn't make a difference right?

Here's the default resource dump:

# resource
resource BEEPER 1 B09
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C08
resource MOTOR 4 C09
resource PPM 1 A08
resource LED_STRIP 1 A06
resource SERIAL_TX 1 C04
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 C05
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource LED 1 C15
resource LED 2 C14
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 A08
resource ADC_BATT 1 C00
resource ADC_RSSI 1 C02
resource ADC_CURR 1 C01
resource ADC_EXT 1 C03
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 C13
resource SPI_PREINIT_IPU 1 A04
resource SPI_PREINIT_IPU 2 B12

Here's after the remap:

# resource
resource BEEPER 1 B09
resource MOTOR 1 C09
resource MOTOR 2 C07
resource MOTOR 3 C06
resource MOTOR 4 C08
resource PPM 1 A08
resource LED_STRIP 1 A06
resource SERIAL_TX 1 C04
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 C05
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource LED 1 C15
resource LED 2 C14
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 A08
resource ADC_BATT 1 C00
resource ADC_RSSI 1 C02
resource ADC_CURR 1 C01
resource ADC_EXT 1 C03
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 C13
resource SPI_PREINIT_IPU 1 A04
resource SPI_PREINIT_IPU 2 B12

I don't know how to post actual screen shots of the CLI so I apologize for that, but these are copied and pasted.

Thanks, Cooper

SupaCoop, 2019-08-26:
> @SupaCoop It's not clear what you want to do at all.
>
> If you are trying to use what you call "lux v2" controller, start with a working firmware and config; start with target firmware name it uses.

I’m using a lumenier lux v2 with the LUXV2_RACE target which is the correct one. I’m trying to remap the motor signals to where they’re supposed to be because when i go to the motors tab and spin up motor 1, motor 3 is the one that physically spins up. I went in the cli and remapped the motor signals to the correct pins as seen on the resource lists I copied and pasted but nothing changes when I spin them up again, motor 1 in the motors tab still spins up motor 3.

Thanks, Cooper

jflyper, 2019-08-27:
@SupaCoop It's not clear what you want to do at all.

If you are trying to use what you call "lux v2" controller, start with a working firmware and config; start with target firmware name it uses.

jflyper, 2019-08-27:
Can you refer to the "similar thread"?

jflyper, 2019-08-27:
Can you show us results of `resource show` before and after the mod?

jflyper, 2019-08-29:
4.0.4 LUXV2_RACE target is also capable of remapping.

Exact same testing procedure, except the motor control is PWM.

Default mapping
[image]

Remapped
[image]

jflyper, 2019-08-29:
For the record, 3.5.7 LUXV2_RACE target firmware should be fully capable of motor remapping.

This test has been conducted with LUXV2_RACE target firmware loaded onto NUCLEO-F303RE, and probing motor output pins.

This is outputs from default motor mapping and (virtually) spin them by motor tab slider with controls at M1 = 1000, M2 = 1250, M3 = 1500 and M4 = 1750. Motor protocol is ONESHOT125.
Note that motor outputs PC6, PC7, PC8 and PC9 all show respective pulse widths.
[image]

Now, motor pins are remapped with
```
resource MOTOR 1 C09
resource MOTOR 2 C07
resource MOTOR 3 C06
resource MOTOR 4 C08
```
With the exact same controls, pins still show correct pulse widths.
[image]

I suspect the remapping procedure was not correct.

closed 2019-10-08 by stale[bot]