betaflight/betaflight #8773: Omnibus F4V6 GPS vs. SBUS collision

ondras, 2019-08-28:
The "GPS" JST connector (VCC, GND, SDA, SCL, RX, TX) on the board is mapped to UART6. However, a serial RX (SBUS/IBUS) is also mapped to UART6, creating a collision.

This happens with Betaflight 4.0.4, target OMNIBUSF4V6, with the board https://www.aliexpress.com/item/32950199878.html

According to the Resource table at https://blog.dronetrest.com/omnibus-f4-v6-flight-controller-guide/, the GPS JST connector shall be mapped to RX3/TX3, but this does not seem to be the case.

Resource mapping:
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED 1
A09: FREE
A10: FREE
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B03: BARO_CS
B04: BEEPER
B05: MOTOR 4
B06: FREE
B07: CAMERA_CONTROL
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: PREINIT 2
C15: FREE
D02: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: MOTOR 4
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

mikeller, 2019-08-28:
@ondras: What you describe is a hardware design defect of the board that you bought - there is nothing that Betaflight can do to address or fix this, you should get your board fixed or replaced under warranty.

closed 2019-08-28 by mikeller

ondras, 2019-08-29:
Hi @mikeller ,

thanks a lot for the prompt response and explanation! I have some short followup questions to clear things up:

  1. it should be possible to solder the GPS rx/tx pair to a different set of tx/rx pins (such as RX3), right?

  1. the GPS unit, BN-880, also has a compass. This works flawlessly (using SDA/SCL pins). How do these two pins fit into the whole image? Are they using their own set of RX/TX pins? I would like to evade the potential "compass vs. GPS on RXn" collision.

  1. is it possible to connect my receiver to a different pad (not the one currently labeled "SBUS", mapped to UART6)? So that I could continue using UART6 for GPS and control the quad via say RX1 or RX3 or similar. I am using IBUS.

mikeller, 2019-08-29:
@ondras:

1. yes;
2. I2S is a different interface to an UART - it does not use TX / RX pins, but uses SDA / SCL pins. Ask the manufacturer of your board for documentation about any SDA / SCL pins available on it;
3. yes.