betaflight/betaflight #8775: when I active integrated Yaw this feature，my quad freaking shake in yaw axis

RamFpv19, 2019-08-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

RamFpv19, 2019-08-29:
my quad use DALRC F4 and DALRC pro 40A ESC

RamFpv19, 2019-08-29:
my firmware version is 4.0.5

etracer65, 2019-08-29:
Please provide the requested information.

closed 2019-08-30 by no-response[bot]