betaflight/betaflight #8780: Cli 'tasks' lists Pid and Gyro with 0hz

TheIsotopes, 2019-08-30:
**Describe the bug**
since #8765 the cli tasks says 0hz on pid and gyro ... flying with it is not a problem

@hydra pls can you check this

```
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       3       1    0.5%    0.5%         0
01 - (         SYSTEM)    999     325       1   32.9%    0.5%       514
02 - (            PID)      0      49      20    0.5%    0.5%       965
   - (           GYRO)      0
03 - (            ACC)    999      12       9    1.6%    1.3%        66
04 - (       ATTITUDE)     99      12       9    0.6%    0.5%         6
05 - (             RX)     32      47      31    0.6%    0.5%         7
06 - (         SERIAL)     99     121       2    1.6%    0.5%         4
07 - (       DISPATCH)    999       3       1    0.7%    0.5%         4
08 - (BATTERY_VOLTAGE)     49       3       2    0.5%    0.5%         0
09 - (BATTERY_CURRENT)     49       1       0    0.5%    0.0%         0
10 - ( BATTERY_ALERTS)      5       3       1    0.5%    0.5%         0
11 - (         BEEPER)     99       6       1    0.5%    0.5%         0
17 - (      TELEMETRY)    249      36       0    1.3%    0.0%        36
19 - (            OSD)     59     476      12    3.3%    0.5%         6
21 - (            CMS)     59       2       1    0.5%    0.5%         0
22 - (        VTXCTRL)      5       1       0    0.5%    0.0%         0
23 - (        CAMCTRL)      5       1       0    0.5%    0.0%         0
25 - (    ADCINTERNAL)      3       4       0    0.5%    0.0%         0
26 - (       PINIOBOX)     19       3       3    0.5%    0.5%         0
RX Check Function                   3       1                         0
Total (excluding SERIAL)                        46.5%    7.3%

# 
```
**To Reproduce**
using build `#1834` an newer 

**Flight controller configuration**
defaults

**Setup / Versions**
MatekF405Mini, MatekF722Mini, SPRacing H7 Nano / Extreme

etracer65, 2019-08-30:
I see the problem - #8765 over-fixed it.

etracer65, 2019-08-30:
https://github.com/betaflight/betaflight/pull/8782

TheIsotopes, 2019-08-30:
working again ... @etracer65 thx 👍

PR #8782 (2019-08-31): Fix tasks period calculation
Fixes #8780 

Corrects a problem introduced in #8765
closed 2019-08-31 by mikeller