betaflight/betaflight #8781: VTX Frequency Hopping Proposal (w/ demo of working prototype)

iphong, 2019-08-30:
So the vtx channel conflict problem is quite obvious. When i am flying and some other dudes plugin their quad which on the same vtx frequency as my then ... everybody know what happens after that.

I have a little idea on how to deal with situation. What if the vtx can some how talk to the google vrx whenever it changes frequency,. And imagine when you lost video while flying, you simply flip a switch on your tx to hop to another frequency and continue flying uninterrupted.

To acchieve this, betaflight alone is not enough as this would require communication from both ends. And this is my proposal, and i think it should work with very little effort:
- First betaflight send infomation about the vtx via telemetry (eg. smartport, csfr etc...)
- TX broascast telemetry via cable or wirelessly over bluetooth or wifi
- VRX module pull vtx information from telemetry packets and tune to the new frequency automatically
- Betaflight needs a new mode, and when this mode is activated betaflight will command vtx to change to another backup frequency (predefined or random), and when this mode is deactivated, betaflight switches back to the original vtx frequency.

There are alternatives way such as vrx connect directly to betaflight via bluetooth and change vtx settings with msp. It kinda works but only within short distances. For this to really work both vrx and quad need to always be in sync. And my solution only require telemetry from quad to tx, which is pretty standard to quads nowadays.

So i've played around with betaflight code and here is what i have so far as a proof of concept:

https://www.youtube.com/watch?v=dkdR_Y_Onvs[https://www.youtube.com/watch?v=dkdR_Y_Onvs]

Hardwares i used:
- Taranis X9D+
- R9M and R9mm
- TBS Unify
- TrueD v3

I’m just getting to fimiliarize with betaflight source code, and still lots of things i dont know where to look for in the code.

Let me know what do you think.

Cheers!!!

etracer65, 2019-08-30:
I think you have a long road to get there with the hardware interfaces and communication completely outside of Betaflight. In particular I don't know how you're going to coordinate having a switch on your transmitter cause the receiver module in your goggles to switch channel. And even so how would it know what channel to switch to? And whatever interface you develop somehow has to be able to presumably talk to more than one kind of receiver module - nobody is going to want to implement this for only a single brand of module.

Implementing a new mode and support to "switch to an alternate" channel isn't really that big a deal. However Betaflight currently doesn't communicate with the VTX while armed. As this can cause interference with audio channels (because of smart audio) and also some VTX's don't handle channel or other changes cleanly (dropped video, delays, etc.).

If you have a telemetry link active (smartport, CRSF) you could already write a Lua script that runs on your OpenTX radio to cause a channel change and not even require a mode in Betaflight. But again the same armed restriction would need to be overcome.

mikeller, 2019-08-30:
I think the idea is good, but I agree that the way to getting there will probably be long.
But there is also one problem that I can see with this approach: This will only give you the desired result if the frequency you are hopping to is unused - if you use a pre-determined 'escape frequency', and somebody else started using it while you were flying then the end result of activating the escape frequency will be worse than not activating it: Now you and this other person are going to crash.
So, I think to make this work in a reliable way it will probably be necessary to make the communication flow somewhat more complex:
When activating the 'escape frequency', first ask the VRX to do a band scan (can be done continuosly in the background), and propose a frequency that was detected as unused. Then send this frequency to the VTX, an have it switch to it.

Dropsleutel, 2019-09-11:
Indeed very interesting, Video Channel Loss is seriously biggest risk. 
But this wont be easy and you'll need certainly new receiver, work on betalight, preferable also work on SPI VTX control, to be fast.

On the RX Goggles side you need two RX modules for your diversity reception, additionally you'll need a third module, which is continuously scanning/monitoring the band fast. Than it should detect the in use channels other pilots and the RSSI.  Depending on the info it selects automatically the best channel, far away from used channels and with a low noise floor. 

This "best" channel advise , can continuously vary. (indeed function could be limited to exclude some channels, or to limit the nr of monitoring channels) The alternative channel advise is send to Betaflight shown in osd.  Pressing push button on TX, send info to betaflight to switch to the next best channel, same time the RX switch to the new channel.

Ideally the VTX to transmit ID information, f.i. digital stream on an audio channel, than the RX system can also determine what is your signal and what not, f.i. if someone blasts your channel, or that its just multipath video loss because you fly behind a wall. 

Theoretically if all pilots use this system, you can safely switch on your Tx, without any channel configuration, it will be completely dynamic channel selection, it does not anylonger matter which channel you are,you will not clash.

For events t will be nasty to follow pilots etc.....

BTW, this Flightcontroller and Cleanflight do support control of RichWave VTX chip via SPI bus
https://shop.seriouslypro.com/sp-racing-h7-extreme
It would enable betaflight to have full control of VTX and very easy to flash 1 firmware only...

closed 2019-10-18 by stale[bot]