betaflight/betaflight #8786: Smart Audio does not work with Unify Pro 58G HV V1

atomgomba, 2019-08-31:
On my quad I have build #1839 (2019-08-31) with SmartAudio configured "as the book says" to my best knowledge. SA is soldered to TX5 on my Matek F722 as it was stated in the FC's manual, so I have enabled SA on Ports tab on UART5. I have also set vtxtable as a SA 2.0 device (as I read in TBS manual that Unify HV is a SA 2.0 device). I also had a battery connected to power the VTX and I tried reaching VTX config using OSD (stick commands), LUA scripts and the Configurator and none of them worked. Following is my diff.

<details>
    <summary>diff.txt</summary>
    ```
  # 

    # diff all

    # version
    # Betaflight / MATEKF722 (MKF7) 4.1.0 Aug 31 2019 / 13:58:40 (c33cc100e) MSP API: 1.42

    # start the command batch
    batch start

    # reset configuration to default settings
    defaults nosave

    board_name MATEKF722
    mcu_id 0033003d3137510434393337
    signature 

    # name: RS6

    # resources
    resource MOTOR 1 C07
    resource MOTOR 2 C09
    resource MOTOR 3 C06
    resource MOTOR 4 C08
    resource MOTOR 6 NONE
    resource CAMERA_CONTROL 1 A08

    # beeper
    beeper -GYRO_CALIBRATED
    beeper -RX_LOST
    beeper -RX_LOST_LANDING
    beeper -DISARMING
    beeper -ARMING
    beeper -ARMING_GPS_FIX
    beeper -ARMING_GPS_NO_FIX
    beeper -BAT_CRIT_LOW
    beeper -BAT_LOW
    beeper -GPS_STATUS
    beeper -RX_SET
    beeper -ACC_CALIBRATION
    beeper -ACC_CALIBRATION_FAIL
    beeper -READY_BEEP
    beeper -MULTI_BEEPS
    beeper -DISARM_REPEAT
    beeper -ARMED
    beeper -SYSTEM_INIT
    beeper -ON_USB
    beeper -BLACKBOX_ERASE
    beeper -CRASH_FLIP
    beeper -CAM_CONNECTION_OPEN
    beeper -CAM_CONNECTION_CLOSE
    beeper -RC_SMOOTHING_INIT_FAIL

    # map
    map TAER1234

    # serial
    serial 1 0 115200 57600 0 115200
    serial 3 64 115200 57600 0 115200
    serial 4 2048 115200 57600 0 115200

    # aux
    aux 0 0 0 1900 2100 0 0
    aux 1 2 1 900 1100 0 0
    aux 2 26 1 1900 2100 0 0
    aux 3 35 3 1900 2100 0 0
    aux 4 47 1 1400 1600 0 0
    aux 5 49 3 1400 1600 0 0

    # adjrange
    adjrange 0 0 2 900 2100 29 2 0 0

    # vtxtable
    vtxtable bands 5
    vtxtable channels 8
    vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
    vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
    vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
    vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
    vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
    vtxtable powerlevels 4
    vtxtable powervalues 0 1 2 3
    vtxtable powerlabels 25 200 500 800

    # master
    set gyro_sync_denom = 2
    set gyro_lowpass_type = BIQUAD
    set gyro_lowpass2_type = BIQUAD
    set dyn_notch_range = MEDIUM
    set dyn_notch_width_percent = 0
    set dyn_notch_q = 250
    set dyn_lpf_gyro_min_hz = 130
    set dyn_lpf_gyro_max_hz = 400
    set acc_calibration = 40,14,10
    set mag_hardware = NONE
    set baro_hardware = NONE
    set serialrx_provider = FPORT
    set serialrx_inverted = ON
    set serialrx_halfduplex = ON
    set blackbox_p_ratio = 128
    set min_throttle = 1070
    set dshot_idle_value = 350
    set dshot_burst = OFF
    set dshot_bidir = ON
    set use_unsynced_pwm = OFF
    set motor_pwm_protocol = DSHOT300
    set motor_pwm_rate = 480
    set bat_capacity = 1300
    set yaw_motors_reversed = ON
    set small_angle = 180
    set pid_process_denom = 1
    set osd_warn_batt_not_full = OFF
    set osd_warn_esc_fail = OFF
    set osd_warn_core_temp = OFF
    set osd_warn_fail_safe = OFF
    set osd_warn_rssi = ON
    set osd_warn_link_quality = ON
    set osd_vbat_pos = 450
    set osd_rssi_pos = 6445
    set osd_link_quality_pos = 6444
    set osd_tim_1_pos = 54
    set osd_tim_2_pos = 6614
    set osd_flymode_pos = 14370
    set osd_anti_gravity_pos = 43
    set osd_g_force_pos = 463
    set osd_throttle_pos = 2505
    set osd_vtx_channel_pos = 2091
    set osd_ah_pos = 200
    set osd_current_pos = 2473
    set osd_mah_drawn_pos = 2466
    set osd_craft_name_pos = 14393
    set osd_gps_speed_pos = 218
    set osd_gps_lon_pos = 82
    set osd_gps_lat_pos = 65
    set osd_gps_sats_pos = 51
    set osd_home_dir_pos = 302
    set osd_home_dist_pos = 303
    set osd_compass_bar_pos = 266
    set osd_altitude_pos = 247
    set osd_pid_roll_pos = 423
    set osd_pid_pitch_pos = 455
    set osd_pid_yaw_pos = 487
    set osd_debug_pos = 1
    set osd_power_pos = 321
    set osd_pidrate_profile_pos = 34
    set osd_warnings_pos = 14569
    set osd_avg_cell_voltage_pos = 14786
    set osd_pit_ang_pos = 257
    set osd_rol_ang_pos = 289
    set osd_battery_usage_pos = 433
    set osd_disarmed_pos = 138
    set osd_nheading_pos = 311
    set osd_nvario_pos = 279
    set osd_esc_tmp_pos = 82
    set osd_esc_rpm_pos = 83
    set osd_flip_arrow_pos = 14799
    set osd_log_status_pos = 14375
    set osd_stat_max_spd = OFF
    set osd_stat_endbatt = ON
    set osd_stat_battery = ON
    set osd_stat_bbox = OFF
    set osd_stat_min_link_quality = ON
    set osd_stat_total_time = ON
    set debug_mode = GYRO_SCALED
    set vtx_band = 5
    set vtx_channel = 8
    set vtx_power = 2
    set vtx_low_power_disarm = ON
    set vtx_freq = 5917
    set vcd_video_system = PAL
    set name = RS6

    profile 0

    # profile 0
    set dterm_lowpass_type = BIQUAD
    set dterm_lowpass_hz = 100
    set dterm_lowpass2_type = BIQUAD
    set vbat_pid_gain = ON
    set iterm_relax_type = GYRO
    set iterm_relax_cutoff = 15
    set throttle_boost = 0
    set d_min_boost_gain = 30
    set d_min_advance = 0
    set ff_interpolate_sp = ON
    set ff_spread = 18

    profile 1

    # profile 1
    set dterm_lowpass2_hz = 200
    set vbat_pid_gain = ON
    set iterm_relax_type = GYRO
    set d_min_boost_gain = 30
    set d_min_advance = 0

    profile 2

    # restore original profile selection
    profile 1

    rateprofile 0

    # rateprofile 0
    set roll_rc_rate = 95
    set pitch_rc_rate = 95
    set yaw_rc_rate = 95
    set roll_srate = 78
    set pitch_srate = 78
    set yaw_srate = 76

    rateprofile 1

    rateprofile 2

    rateprofile 3

    rateprofile 4

    rateprofile 5

    # restore original rateprofile selection
    rateprofile 0

    # save configuration
    save
    # 
    ```
</details>

etracer65, 2019-08-31:
Have you first confirmed that you have Smartaudio working with the actually released BF4.0.5? First confirm that to eliminate any wiring issues.

If that works, try different SA version `vtxtable` setups. Perhaps your device is not SA2.0.

atomgomba, 2019-09-01:
Okay, so after more investigation it turned out that the manual for Matek F722-STD was misleading. Sorry, for opening a useless issue. Nevertheless, I will describe how to set up Smart Audio with this board here for others (Google) to find it (there is a surprising lack of this info on the web). So the manual states explicitly that "Tramp & SmartAudio can be connected to any unused UART_TX". This turned out to be false since TX5 did not work. With Matek F722 Smart Audio will work if it is soldered to TX1 (UART1).

closed 2019-09-01 by atomgomba

mikeller, 2019-09-01:
@atomgomba: Glad to hear you managed to figure it out. Did you find any pointers as to why TX5 does not work for SmartAudio?

@MATEKSYS might be able to shed some light on this.

MATEKSYS, 2019-09-01:
@mikeller  TX5 is nothing special.  F722-STD and F405-STD has been in market for around 2 years. I don't remember someone reported any compatibility issue between TX5 and SA.  unless there is cold joint on components.

atomgomba, 2019-09-01:
Hm, I have this board for more than a year now. It has controlled several crafts before it ended up in its current place. It didn't occur to me that it might have been damaged, so sorry for the false accusation. Naked eye cannot notice anything and I don't have the expertise or the equipment or access to a brand new board to conduct a 100% fair test of the port.