betaflight/betaflight #8788: Can't configure Barometer with I2C. Commands missing

Aliasname, 2019-09-01:
Hello together,

i connected a Barometer BMP280 over I2C to a Mamba F405 on FuryF4OSD.
Enabling under "Configuration" with Betaflight Configurator 10.5.1 just causes that the switch is still disabled after restarting th FC.
Under the CLI no single command for Baro like set baro_hardware is available.
https://github.com/betaflight/betaflight/wiki/Barometer-Configuration-(3.2) seems to be outdated.
"status" shows "No I2C errors".
"resource" no doubled pins.
and "set" shows no single Barometer or I2C command.

Whats the matter of the missing commands? 
Thanks in advance for your help.

Aliasname, 2019-09-01:
Thanks for labeling.
Here are more informations out of the cli:

1. version
2. resource
3. tasks
4. set

[cli output.txt](https://github.com/betaflight/betaflight/files/3563188/cli.output.txt)

Please let me know if further informations are needed.

[image]

jflyper, 2019-09-01:
We need your CLI `diff` output.

etracer65, 2019-09-01:
The `FURYF4OSD` target did not include baro support. Already fixed in #8771 which is included in the Betaflight 4.0.6 release.

Aliasname, 2019-09-01:
@jflyper Thank you for that info. In future i will directly tell "diff".
@etracer65 Thank you very much for that information. Than it's absolutey no wonder that it won't work. :-)
Unfortunately I did not find this information myself.

closed 2019-09-01 by Aliasname