betaflight/betaflight #8789: Vector Tricopter

Saint-Drone, 2019-09-01:
First Time Posting On Here..... 
Not sure if in correct place etc.... 

Just built a Vector Tricopter, Manual was not clear on some bits so am abit lost.... 

Omnibus F4 V6 (Flight Controller)
Pitch Servo (goes to M1)
Yaw Servo (goes to M2)
Using a 4 in 1 esc
Xsr-M Receiver
Eachine TX1200 VTX (if that's the correct model)

I've made a baby tricopter from RC Explorer and it works fine.... But this one is not, motor 3 works but 1 & 2 aren't.... Can't get servos to work either.... Followed everything on the manual but to no avail....
Also says to assign pitch to Aux on controller and so forth on flight controller side but does not show how to do that....

Controller I am using is a Jumper T16 

Thanks

Saint-Drone, 2019-09-01:
Missing information????

IvoFPV, 2019-09-01:
@Saint-Drone Post your `diff` here.

jflyper, 2019-09-01:
@Saint-Drone Yes, you've landed at the correct place, and yes, we need a CLI `diff` output from the current configuration. Also tell us with how you wired motor signals.

closed 2019-09-02 by no-response[bot]