betaflight/betaflight #8793: REVO (REVO) config wrong mag alignment

guerradr, 2019-09-01:
I own a old openpilot revolution board and I just updated to the latest Betaflight 4.0.5 release.
I found that the magnetometer has a wrong configuration. It is 90° off.
I was able to fix the problem by typing set align_mag = CW180 in the CLI.

etracer65, 2019-09-02:
As requested when you opened the issue, please provide a `diff` of your settings.

guerradr, 2019-09-02:
Sure I will paste the dump here:
Board is an original Openpilot Revolution.
Thanks for your great work.

`# version
# Betaflight / REVO (REVO) 4.0.5 Aug  1 2019 / 12:19:50 (839661b20) MSP API: 1.41

# start the command batch
batch start

board_name REVO
manufacturer_id 

# name
name MiniSpiderHex

# resources
resource BEEPER 1 NONE
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A00
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
[... 886 lines cut ...]
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end`

etracer65, 2019-09-02:
The reason you think the alignment is wrong is because you haven't mounted the board in it's normal orientation.
```
set align_board_yaw = -90
```

guerradr, 2019-09-03:
hi etracer65
you made me take apart my copter and look at the board orientation. I flew my hexaspider with openpilot, cleanflight, and betaflight from the early versions on. I never had to change the board and mag orientation befor. As I was pretty sure and checked now again the board orientation is correct according the printed shevrons. I will post a picture to make it clear.

guerradr, 2019-09-03:
[image]

etracer65, 2019-09-03:
Well you have this in your config:
```
set align_board_yaw = -90
```
So then it would make sense that you would have to set the MAG to `CW180` since the default is `CW90` and you've applied another 90 degrees of rotation to it. Sounds like you should start with a clean install (`defaults` in CLI) and reconfigure just the minimum.

guerradr, 2019-09-07:
I found following description on the Betaflight 4.0 relesase notes which I overread.

the gyro configuration was unified from being done by gyro hardware model to being generic for all gyro models. This means that for some targets that are used for multiple different boards that come with different gyro models that are mounted with different orientations, it is necessary from Betaflight 4.0 on to manually set gyro_1_sensor_align (and / or gyro_2_sensor_align for boards with a second gyro) to match the gyro orientation on the board that is used. See note #6761 for an explanation, and instructions for individual boards. This is a temporary workaround, per-board configurations with correct gyro alignment will be released as Unified Target configurations (#5868);

So issue resolved.

closed 2019-10-14 by stale[bot]