betaflight/betaflight #8794: gyro rate variance in CLI - STATUS, but never in TASKS!

yoodog, 2019-09-01:
**Describe the bug**
Manual tells us to use TASKS to check for Gyro rate drops in order to judge for loop time issue. But surprisingly I see very drastic difference in Gyro Rate when checking rapidly multiple times TASKS and STATUS.

**To Reproduce**
1. Set f7 board to 4/4 loop time and dshot600
than go to CLI and enter TASKS many times and note Gyro Rate = .....
result for me is +/- 1 or 2 max at all times which is up to 0.05%
2.  now type STATUS many times and note same Gyro Rate = ......
result will be that occasionally you will see drops down to 3800 or even rises to 4200 on occasion. 5%!
3. Setting it to other variations 8/8 and 1200dshot doesn't change the situation.
- on TASKS we start to see variance of up to +/-30 points which is 0.3xx%
- on Status we start to see variance of 7750 (7570 even) (3.12%-5.3%!)and upper can not go even 1% above 8k due to mpu6000 limit. 

So at the end on f745 4/4k 600 works much more stable according to STATUS, with rare bursts, but 8/8 1200 usually is also in 1% recommender with COMMON bursts to 3 or 5%....

My question is, why do we see such difference in STATUS but not TASKS, which is the recommended for evaluation.

[Diff all defaults.txt](https://github.com/betaflight/betaflight/files/3564031/Diff.all.defaults.txt)

**Flight controller [configuration**]
Omnibus f7v2 
BF 4.0.5 stable - RPM filtering + standard filters
ESC Typhoon32 4in1 ESC 4x35A v2.1 BLHeli 32.6.4 Beta

etracer65, 2019-09-02:
It’s a side effect of “scheduler_optimize_rate” needed for RPM Filtering.

yoodog, 2019-09-02:
It, you mean the fact that STATUS Gyro rate jumps, or that TASKS remains much closer to set rate?

etracer65, 2019-09-02:
Both.

yoodog, 2019-09-02:
> Both.

thanks, good info - I have not seen it mention anywhere. In that case it is interesting how much more stable out put 4/4 produced compared to 8/8. BTW, 8/4 did produce max drops down to 5%, but as rare as 4/4. So I think 4/4 or 8/4 are better rates form performance point of view. 

I stick with 4/4 as 8/8 requires extra PT filter to get same result as 4/4 and I think final delay win is almost 0ms and added complexity. Any thoughts?

closed 2019-10-09 by stale[bot]