betaflight/betaflight #8795: Is there a way to add (or fake) custom variables to the OSD?

mactac, 2019-09-02:
Here's what I am trying to do, and I'm assuming that I will need some sort of workaround:

I want to take 2 telemetry variables, let's say speed and distance to home.  I want perform some simple math (like division or multiplication)on them, and display it in the OSD.

I presume that I'm going to have to use the radio to do the math (?), and send it on an aux channel back to the radio.

How do I get that onto the OSD?  One way I thought of is to set the RSSI channel on the receiver tab to the channel I'm sending the variable to... but then I lose the RSSI display.

Is there any way to add variables to the OSD, or is there a creative way to get it to display, maybe as one of the existing variables?

mikeller, 2019-09-02:
@mactac: There currently isn't - abusing RSSI for it as you suggest is the only way, short of modifying the firmware. Display of the value of a pre-selected AUX channel could be added as an OSD element, but I am not sure that there are many pilots that would want to make use of this.

mactac, 2019-09-02:
> @mactac: There currently isn't - abusing RSSI for it as you suggest is the only way, short of modifying the firmware. Display of the value of a pre-selected AUX channel could be added as an OSD element, but I am not sure that there are many pilots that would want to make use of this.

I think that a lot more people would want to to this if it were possible.  here's an example of a killer application of it: long range.   As a long range pilot, efficiency is everything.  One piece of information that I would love to have is exactly what attitude/speed/etc will get me the most efficient flight.  If I could display amps divided by speed in my OSD, I could maximize my efficiency to be able to cover the longest distance by choosing the most efficient speed based on my quad/wind/air conditions, etc.  It would be a bit of a game changer for people.

Being able to display an Aux channel would be fantastic for this, and a number of other things.

crteensy, 2019-09-13:
As a beginner I'd be curious about my heart rate. Should be interesting to see it go through the roof right from the get-go. Betaflight can't do that just by itself of course, but having some more freedom on the OSD would be a step towards such a feature.

Mateyhv, 2019-09-24:
For simple efficiency I don't see the need of sending the info back to the radio and again to the drone for OSD display. At some point you may loose return telemetry in wich case any relevant info on te OSD will be lost. Why dont just calculate them onboard and display them via the OSD menu? Inav already has watt/km and Amp/km indicators all calculated onboard. 

Of course there might be other useful parameters to be displayed on the OSD via an AUX channel but if it was just for the efficency meter its easier to be implemented directly on the flight controller as a new entry on the OSD page.

mikeller, 2019-09-25:
@Mateyhv: The appeal for implementing this by displaying an RC channel value is the versatility of the solution - as you say iNav already has 2 different metrics only to cater for long range users - if you'd want to cover all different aspects then you'd end up having to add / maintain 10 or so similar custom metrics. Thanks to the flexibility of the OpenTX mixer (including support for lua), these could all be done on the TX.

Mateyhv, 2019-09-25:
@mikeller I can clearly see the advantage of the system and the open door for a lot of possibilities. Actually the idea is great! My only concern was to rely on a down/up link and specially on rx down link on long range flights but that is related just to the example not to the feature itself. ☺

mactac, 2019-09-25:
> @mikeller I can clearly see the advantage of the system and the open door for a lot of possibilities. Actually the idea is great! My only concern was to rely on a down/up link and specially on rx down link on long range flights but that is related just to the example not to the feature itself. ☺

What about implementing both?   The code for the calculation is over at inav, should be easy to port over.

pierreh, 2019-10-31:
Not everyone has a transmitter that they can script (me) so being able to do this in Betaflight would be great. Especially watt/km.

crteensy, 2019-10-31:
Micropython! Maybe a bit over the top though...

pierreh, 2019-11-04:
> Micropython! Maybe a bit over the top though...

Will take too many resources of the flight controller. A simple expression evaluator however won't be too complicated to build/include.

nightflasher, 2021-05-18:
Sounds like the problem the dji Digital system users have, getting betaflight warnings in the osd. The inav devs have a nice way to inject "custom values" into the osd by "abusing" the 'pilot name' oder element.

fy-cat, 2023-03-01:
Wondering if anything more was accomplished towards this? id like to add a custom icon that indicates when the drone is at a correct angle, ex: white square square at bottom changing to a red square when drone at a set angle ? or even just changing shape ?

closed 2026-09-26 by blckmn