betaflight/betaflight #8798: Unstable cycle time on F3 board with telemetry enabled

Juzzle1, 2019-09-02:
This has been reported previously (#7805) but has been closed due to inactivity.

On my F3 board (Singularity), cycle time varies dramatically when telemetry is enabled.
CPU load doesn't change much with it enabled / disabled (~24%).
I connected a motor output to an oscilloscope and the changing cycle time is clearly visible with some loops taking up to 800us. With telemetry disabled, cycle time is relatively stable at around 380us (1.3kHz PID, 2.6kHz gyro).

Telemetry enabled:
[image]

Telemetry disabled:
[image]

**To Reproduce**
Steps to reproduce the behavior:
Enable telemetry - notice cycle time is erratic
Disable telemetry - cycle time is steady.

**Expected behavior**
Cycle time should not be greatly affected by enabling telemetry.

**Flight controller configuration**
```
# Betaflight / SINGULARITY (SING) 4.0.6 Sep  1 2019 / 01:03:42 (2a64051a2) MSP API: 1.41 / FEATURE CUT LEVEL 0
batch start

board_name SINGULARITY
manufacturer_id 
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 B09

feature MOTOR_STOP
feature TELEMETRY
feature LED_STRIP
serial 0 1 115200 57600 0 115200
serial 2 32 115200 57600 9600 115200
aux 0 0 1 1200 2075 0 0
aux 1 2 0 900 1225 0 0
aux 2 13 2 1750 2100 0 0
aux 3 15 3 900 1700 0 0
set acc_calibration = -112,-59,96
set serialrx_provider = SBUS
set blackbox_p_ratio = 16
set blackbox_device = NONE
set min_throttle = 1050
set max_throttle = 1950
set use_unsynced_pwm = OFF
set motor_pwm_protocol = ONESHOT125
set motor_pwm_rate = 2000
set current_meter = NONE
set battery_meter = ADC
set ledstrip_race_color = BLUE
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 2
set camera_control_ref_voltage = 325
set camera_control_key_delay = 100
profile 2

rateprofile 0

set thr_mid = 22
set thr_expo = 52
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
batch end
```
```
# resource show all
IO
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: VTX
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: SERIAL_RX 2
B00: MOTOR 3
B01: MOTOR 4
B02: ADC_BATT
B03: LED 1
B04: MOTOR 1
B05: MOTOR 2
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: GYRO_EXTI
C14: FREE
C15: BEEPER
F00: FREE
F01: FREE
F03: FREE
F04: FREE

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: LED_STRIP
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```
**Setup / Versions**
 - Codelayer Singularity
 - FRSKY X4R with smartport telemetry

etracer65, 2019-09-02:
Just out of processing power. Try changing the gyro/pid to 2/1K. Also disabling LEDSTRIP might help.

Juzzle1, 2019-09-02:
Disabling LED strip didn't make any noticeable difference, however even 4K/2K seems pretty consistent with telemetry disabled.
Cycle time is stable at 2K/1K with telemetry enabled.

Some quick CPU stats with telemetry enabled / disabled:
4K/2K,       50% / 40% 
2.6K/1.3K  23%/18% 
2K/1K        16% / 12% 

Doesn't this seem excessive just for sending telemetry, even if the F3 is limited in performance?

Is there any way of reducing the update rate for FRSky telemetry? Looking at the telemetry page on my Taranis, the telemetry seems to be updating at at least 10Hz, which is certainly overkill for getting battery voltage!

etracer65, 2019-09-02:
Changing the update rate wouldn't have much if any impact. It has to do with how long a single cycle of telemetry processing takes. 10hz is quite slow when compared to 2KHz. The cycle time jumps are because the MCU is busy running other code when it's "time" to do the next gyro/pid loop. So it's the duration of the "other code" that causes the variances. Running a slower gyro/pid gives more processing time between loops to accommodate the other tasks.

If you have a barometer on that board try disabling it as it can add significant loop delay.

Also the `tasks` command is most useful for looking at what's impacting loop time. Run the command a couple of times and look at the "max" time for the tasks to see what might be using to much processing time (don't worry about serial).

etracer65, 2019-09-02:
You could also try:
```
set telemetry_disabled_sensors = 1048572
```
This will disable all the telemetry sensors other than voltage and current. This is actually a bitmap value with a disable flag for each sensor but it's difficult to use on F3. On F4+ each sensor is broken out as a separate setting but not on F3 because of flash space constraints. The individual bits are:
```
    SENSOR_VOLTAGE         = 1 << 0,
    SENSOR_CURRENT         = 1 << 1,
    SENSOR_FUEL            = 1 << 2,
    SENSOR_MODE            = 1 << 3,
    SENSOR_ACC_X           = 1 << 4,
    SENSOR_ACC_Y           = 1 << 5,
    SENSOR_ACC_Z           = 1 << 6,
    SENSOR_PITCH           = 1 << 7,
    SENSOR_ROLL            = 1 << 8,
    SENSOR_HEADING         = 1 << 9,
    SENSOR_ALTITUDE        = 1 << 10,
    SENSOR_VARIO           = 1 << 11,
    SENSOR_LAT_LONG        = 1 << 12,
    SENSOR_GROUND_SPEED    = 1 << 13,
    SENSOR_DISTANCE        = 1 << 14,
    ESC_SENSOR_CURRENT     = 1 << 15,
    ESC_SENSOR_VOLTAGE     = 1 << 16,
    ESC_SENSOR_RPM         = 1 << 17,
    ESC_SENSOR_TEMPERATURE = 1 << 18,
    SENSOR_TEMPERATURE     = 1 << 19,
```

Juzzle1, 2019-09-03:
I had high hopes for the telemetry_disabled_sensors bitmask, however it seems exactly the same even when I disable the current sensor, leaving just one sensor running (voltage).

It's probably time to go shopping for an F4/F7 board especially as support is being dropped in 4.1... unless there are any more bright ideas out there!

Cheers.

mikeller, 2019-09-03:
@Juzzle1: Bright idea: Treat yourself! Go buy a nice F4 or F7 - you won't regret it once you realise how many enhancements you get that have had to be left out on F3 due to lack of sufficient flash storage.

closed 2019-09-03 by Juzzle1