betaflight/betaflight #8800: About the last Betaflight 4.0.6 and IRC Tramp do not work together

FrieKi, 2019-09-02:
Hello,
After flashing the new BF firmware, i have no control of the tramp anymore.  No control possible via: FC , lua script or OSD.
Are there already the same errors?

Betaflight 4.0.6
Betaflight F4
IRC Tramp HV
FrSKY R9M 2019
Frsky R9 MM
Taranis X9D plus lua script

best regards

DUMP:
Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# Betaflight / BETAFLIGHTF4 (BFF4) 4.0.6 Sep  1 2019 / 00:22:22 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

board_name BETAFLIGHTF4
manufacturer_id 

# name
name -

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 B06
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
[... 906 lines cut ...]
set thr_mid = 33
set thr_expo = 44
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 70
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

#

etracer65, 2019-09-02:
Please edit your comment and change the `dump` to a `diff` output.

etracer65, 2019-09-02:
Why have you changed this?
```
set vtx_halfduplex = OFF
```

FrieKi, 2019-09-02:
Oh, that was a mistake. I changed it to On. Now it works fine. Thanks a lot for your help!

closed 2019-09-02 by FrieKi