betaflight/betaflight #8801: dshot hexacopter setup on openpilot revolution board

guerradr, 2019-09-02:
Hi there
I'm trying to get my old revo board running with dshot. I did use multishot so far with no problem. I know that there must be a dma issue since motor 2 is not responding when dshot enabled. I found that in the resource mapping the pin of motor 2/5 was set to LED and SERIAL 4 Port so I disabled those entries. 
On the dma side there seems to be no conflict as I can see with my limited knowledge.  Here are the corresponding dumps:

`# resource
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A00
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 B05
resource LED 2 B04
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B14
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource PINIO 1 C08
resource FLASH_CS 1 B03
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05`

`# dma
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin C07 0
# pin C07: DMA2 Stream 2 Channel 0
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 6 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3
dma pin A01 0
# pin A01: DMA1 Stream 4 Channel 6
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6`

Any help to get the last motor working is highly appreciated.

Thank you

jflyper, 2019-09-03:
@guerradr CLI `diff` please. Also, please enclose output text with a pair of three back quotes to protect the text from formatting by md.

guerradr, 2019-09-03:
here:

```# diff

# version
# Betaflight / REVO (REVO) 4.0.5 Aug  1 2019 / 12:19:50 (839661b20) MSP API: 1.41

# start the command batch
batch start

board_name REVO
manufacturer_id 

# name
name MiniSpiderHex

# resources
resource LED_STRIP 1 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_RX 4 NONE

# timer

# dma

# mixer
mixer CUSTOM
mmix 0  1.000 -0.580 -0.890  0.850
mmix 1  1.000 -1.000  0.000 -1.000
mmix 2  1.000 -0.420  1.000  0.870
mmix 3  1.000  0.420  1.000 -0.870
mmix 4  1.000  1.000  0.000  1.000
mmix 5  1.000  0.580 -0.890 -0.850

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# led

# color

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3

# aux
aux 0 0 0 1925 2100 0 0
aux 1 1 1 1400 1600 0 0
aux 2 2 1 1925 2050 0 0
aux 3 46 4 1900 2100 0 0
aux 4 19 2 1900 2100 0 0
aux 5 26 2 1350 1650 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 7 h
rxfail 8 s 2000

# display_name

# master
set acc_trim_roll = -8
set align_mag = CW180
set mag_calibration = 242,-389,-3
set min_check = 1100
set rssi_channel = 8
set rc_interp_ch = RP
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = OFF
set serialrx_provider = FPORT
set airmode_start_throttle_percent = 35
set blackbox_device = SPIFLASH
set min_throttle = 1025
set min_command = 950
set dshot_idle_value = 400
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 16000
set failsafe_delay = 10
set failsafe_throttle = 1200
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = -90
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 178
set vbat_multiplier = 2
set ibata_scale = 900
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_rescue_descent_dist = 50
set gps_rescue_throttle_max = 1800
set gps_rescue_throttle_hover = 1500
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 50
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set gyro_1_sensor_align = DEFAULT

# profile
profile 0

set feedforward_transition = 30
set iterm_relax_type = GYRO
set throttle_boost = 0
set p_pitch = 66
set i_pitch = 50
set d_pitch = 35
set p_roll = 52
set i_roll = 50
set d_roll = 30
set p_yaw = 70
set angle_level_strength = 20
set horizon_level_strength = 60
set d_min_advance = 50

# rateprofile
rateprofile 0

set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82

# end the command batch
batch end```

jflyper, 2019-09-04:
There is a DMA stream conflict between motor 2 and 6.

Using burst mode DShot will resolve this issue (`set dshot_burst = ON`) in this particular configuration.

If you are going for RPM filtering (aside from whether or not the RPM filtering works for HEXA), the burst mode DShot solution would not work, but bit banging DShot (planned to come with 4.1.0) will probably provide you a way to do it.

Another solution at this moment is to use PWM/PPM input signals as motor outputs. You can route these input lines available at FlexIO as motor outputs.

guerradr, 2019-09-07:
Thanks alot. It works with Dshot_burst=ON
Hopefully I can use bit banging for the RPM filtering in the future.
Great stuff to come.

closed 2019-10-14 by stale[bot]