betaflight/betaflight #8802: Total distance inaccurate

Jetrell, 2019-09-03:
These flights were done with development firmware from a few days ago. I had not done a firmware upgrade for over a month, previous to this. But the Total Distance did work correctly then. (that firmware did have the Km/Mi total distance changes).

From the moment the quad is armed, it starts counting up be many 1km's at a time. 
I flew over 800kms on an 850 pack :-) That's long range duration! Lol

The result was the same on two flights that I did.  The image shows how the TD was at 13.1kms while still at the home location after takeoff.

[image]

[Diff all.txt](https://github.com/betaflight/betaflight/files/3567703/Diff.all.txt)

[btfl_002.zip](https://github.com/betaflight/betaflight/files/3567704/btfl_002.zip)

etracer65, 2019-09-03:
@TonyBlit Any thoughts?

Jetrell, 2019-09-03:
@etracer65  I thought that the 3D speed #8759 around command line 1286, might have been the problem (with my limited code understanding), before I opened this issue.
Today after doing another test without  `gps_user_3d_speed` Disabled.  The Total Distance figure worked correctly again.

IvoFPV, 2019-09-03:
@Jetrell I may be wrong, but I think it should be:
 `dist = sqrtf(powf(dist, 2.0f) - powf(gpsSol.llh.altCm - lastAlt, 2.0f));`

TonyBlit, 2019-09-04:
Oh, I think I see the problem. LastAlt should be int32_t, what a mistake on my side and I'm surprised this bug has sneaked for almost a year. I'll test the change tomorrow to be sure. My apologies.

PR #8827 (2019-09-06): Bugfix: Vario GPS and 3D Total Distance overflow
Previous code was only working below 327m altitude (where I do most of my tests).

Fixes #8802.
closed 2019-09-06 by mikeller