betaflight/betaflight #8805: BF 4.1 and smart audio freezes

HrrrPippi, 2019-09-03:
**Describe the bug**
When i updated to BF 4.1 testiing, my FC froze, once i enabled Smart audio, and pluged in,so the VTX turned on.
I searched around github, stumbled upon the new VTX tables, and once i updated the FC with the new VTX table/info, the problem went away. Now smart audio is working.
I dont know if this has to do with fact that it is a F7 board, but i suspect it is a bug between BF 4.1 and my VTX.

**To Reproduce**
Enable Smart Audio on a BF 4.1 Build, Plug lipo in, FC Freezes.

**Expected behavior**
Can you fix this, so when upgrading from BF 4.0.x to BF 4.1, prople dont have to trouble shoot with finding and updating to the correct Smart audio version.

**Flight controller configuration**
```
# Betaflight / MATEKF722SE (MF7S) 4.1.0 Aug 29 2019 / 14:42:01 (fc97d888f) MSP API: 1.42

# vtxtable
board_name MATEKF722SE

# timer
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)

# dma
dma pin A03 0
# pin A03: DMA1 Stream 1 Channel 6

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 3 2048 115200 57600 0 115200

**This is with BF 4.1 Functioning. I remove this, FC will freeze
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

set vtx_band = 5
set vtx_channel = 2
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5695
```

```
A00: SERIAL_TX 4
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: ADC_EXT
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: USB_DETECT
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 4
DMA1 Stream 3: FREE
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 3
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
Matek F722mini (uses F722SE hex)
AKK FX2-Ultimate

**Additional context**
I dont know if this will be fixed with a new configurator or once you are ready to release BF 4.1. But i do know, that not enabling Smart audio or updating the VTX table makes me able to fly.

IvoFPV, 2019-09-03:
This has been happening a lot lately. Several users reported this issue where FC locks up if vtxtables are not set before UART is assigned to Smartaudio. This is not the expected behaviour and will have to be looked into.

HrrrPippi, 2019-09-03:
So i finally reported a correct firmware bug? Would sooo love to help with the developing

mikeller, 2019-09-03:
I think for now the least we can do is catch / cover this in configurator: When a VTX protocol is selected in the Ports tab but no VTX table is configured, show a popup telling the user they first need to configure a VTX table, and then deselect the VTX protocol. @IvoFPV: Do you have time to work on this?

But of course we'll also have to look into fixing the lockup in the firmware.

IvoFPV, 2019-09-03:
@mikeller I don't think that's a good approach because you have to first set up the smartaudio port before you can access the VTX tab in the configurator if I'm not wrong. So by having the warning in the ports tab and disabling it if VTX table is not set up forces the users to set the VTX table from CLI. This is a firmware bug and should be fixed there. If port is assigned but VTX table is not set it should just not send any commands to VTX if that is even possible.
This seems to only happen if the VTX is actually powered on, maybe @HrrrPippi can confirm this.

mikeller, 2019-09-03:
@IvoFPV:

> I don't think that's a good approach because you have to first set up the smartaudio port before you can access the VTX tab in the configurator if I'm not wrong.

Well, I guess changing this useful way would have to include making the VTX tab available even if no VTX protocol is selected for now then.
What we have now is a chicken and egg situation where users are forced to enable SmartAudio on a port in order to get the VTX table tab, and then their flight controller is locking up before they are able to configure the VTX table due to a not yet identified bug in the firmware.
We can either:
- watch and wait until such time that the bug in the firmware is found and fixed;
- implement a workaround in configurator that will enable users to configure VTX tables without having to use CLI.

In addition to this, I think the work to add a prominent warning to the ports tab if a user tries to set up a port for VTX control but no VTX table is defined is an important improvement of usability anyway, as it will keep users from trying to use VTX control without having configured a VTX table. So even once the firmware bug is fixed we will want to keep the 'VTX table not configured' warning in the ports tab around.

HrrrPippi, 2019-09-03:
The fc will only freeze, once vtx is on. So it will be possible to turn on the uart in ports tab, and then go to the vtx tab. I haven't tried it yet, though.

IvoFPV, 2019-09-03:
I will add a warning then and enable vtx table to be configured even without a port assigned to Smartaudio later this week if that is okay. @mikeller

mikeller, 2019-09-03:
Thanks @IvoFPV. Much appreciated.

etracer65, 2019-09-07:
I don't understand why you have this in your `diff`:
```
# timer
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)

# dma
dma pin A03 0
# pin A03: DMA1 Stream 1 Channel 6
```
Pin A03 is UART2 RX so I'm not sure why you're changing the dma/timer.

HrrrPippi, 2019-09-07:
Etracer65

Those changes were pasted in from the snippit for the matek f722se. I don't know what it does, but I guess it has to do with Rpm telemetry.

Am I wrong by using it?

etracer65, 2019-09-07:
@HrrrPippi I'm not sure it it's a problem or not but just to eliminate possibilities can you test without the timer and dma settings - but just use the rest of your `diff`? I set up a board with smart auto to try and debug this problem but I can't replicate the problem. Board has no problem with the device powered on, UART set, but with no `vtxtable` defined.

HrrrPippi, 2019-09-07:
@etracer 65 I just did a test, and my board still freezes once i power on the VTX. I tried dshot_bidir = off & dshot_burst = on and still freezes.

This is on build from 

Here is the diff from when not working

´´´
# version
# Betaflight / MATEKF722SE (MF7S) 4.1.0 Aug 29 2019 / 14:42:01 (fc97d888f) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722SE
mcu_id 003000343237510a34363730
signature 

# name: Flos2tyle

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 3 2048 115200 57600 0 115200

# aux
aux 0 0 0 1575 1825 0 0
aux 1 1 1 1950 2050 0 0
aux 2 27 4 1300 1400 0 0
aux 3 13 0 1375 1425 0 0
aux 4 28 1 950 1050 0 0
aux 5 35 0 1750 1850 0 0

# master
set gyro_lowpass_hz = 225
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set acc_calibration = 38,-11,455
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 11
set serialrx_provider = IBUS
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_idle_value = 100
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set vbat_scale = 109
set ibata_scale = 300
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set transient_throttle_limit = 0
set osd_warn_batt_not_full = OFF
set osd_rssi_alarm = 50
set osd_cap_alarm = 10000
set osd_tim1 = 0
set osd_tim2 = 17
set osd_vbat_pos = 8609
set osd_rssi_pos = 10697
set osd_tim_2_pos = 406
set osd_g_force_pos = 88
set osd_throttle_pos = 8225
set osd_current_pos = 8656
set osd_mah_drawn_pos = 14807
set osd_motor_diag_pos = 8248
set osd_craft_name_pos = 10282
set osd_altitude_pos = 366
set osd_debug_pos = 8545
set osd_warnings_pos = 14666
set osd_avg_cell_voltage_pos = 14785
set osd_core_temp_pos = 34
set osd_stick_overlay_left_pos = 4353
set osd_stick_overlay_right_pos = 4374
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set gyro_rpm_notch_harmonics = 2
set name = Flos2tyle

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 140
set dyn_lpf_dterm_max_hz = 340
set dterm_lowpass2_hz = 300
set pid_at_min_throttle = OFF
set anti_gravity_gain = 10000
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set i_pitch = 90
set d_pitch = 22
set f_pitch = 95
set i_roll = 85
set d_roll = 20
set f_roll = 90
set i_yaw = 90
set f_yaw = 90
set d_min_roll = 0
set d_min_pitch = 0
set idle_min_rpm = 14

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 150
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass2_hz = 200
set pid_at_min_throttle = OFF
set anti_gravity_gain = 10000
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set d_min_roll = 0
set d_min_pitch = 0
set idle_min_rpm = 14

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
set tpa_breakpoint = 1750
set throttle_limit_type = SCALE

rateprofile 1

# rateprofile 1
set roll_rc_rate = 103
set pitch_rc_rate = 103
set yaw_rc_rate = 103
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 66
set tpa_breakpoint = 1750
set throttle_limit_type = SCALE

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1
´´´

etracer65, 2019-09-07:
Not sure how to debug this as I can't replicate. Connecting a smart audio vtx, having the UART configured, but with no `vtxtable` configuration I have no problems (other than smart audio not working). Configuring `vtxtable` and smart audio control works as expected.

monrapps, 2019-09-10:
Same here. FC freezes when VTX is turned ON. (KAKUTEF7)

HrrrPippi, 2019-09-10:
@monrapps what exact kakute f7 are you using. Wondering if it has to do with only f7 processors, or maybe all f722 processor fc's.

monrapps, 2019-09-10:
I don't know how to identify that... its NOT a KAKUTEF7V2 and its NOT AIO version....  Just KAKUTEF7.... Configuring the vtxtables seems to solve the problem... Working fine now

etracer65, 2019-09-10:
@HrrrPippi I think I found the problem. Can you try this version? Please test to see if the lockups are gone if smartaudio is selected on a UART, vtx is connected and powered, and `vtxtable` is **NOT** configured.

[betaflight_4.1.0_MATEKF722SE.hex.zip](https://github.com/betaflight/betaflight/files/3597465/betaflight_4.1.0_MATEKF722SE.hex.zip)

@monrapps If you want to try as well, here's your target.

[betaflight_4.1.0_KAKUTEF7.hex.zip](https://github.com/betaflight/betaflight/files/3597467/betaflight_4.1.0_KAKUTEF7.hex.zip)

monrapps, 2019-09-10:
@etracer65 it worked... FC didn't froze without configuring vtxtables (smartaudio selected on a UART, vtx is connected and powered). Still not possible to configure the smartaudio device with some "default table" (I think that is the expected behavior, configuring vtxtables still necessary to work properly ).

etracer65, 2019-09-10:
> Still not possible to configure the smartaudio device with some "default table" (I think that is the expected behavior, configuring vtxtables still necessary to work properly ).

No, this is the expected behavior. There is no "default" table - the user must always configure it based on what bands, channels, and power levels are legal in their country.

monrapps, 2019-09-10:
> No, this is the expected behavior. There is no "default" table...
@etracer65 Sorry. I was not clear. That's exactly what I meant.