betaflight/betaflight #8808: Feature request: delayed GPS Rescue Mode Enable

VoicOfReason, 2019-09-03:
I'd like to suggest an option for GPS Rescue Mode in which the mode enables after a delay. Specifically, my problem, and I'm sure many other people would agree it's a problem, is that you can't just take-off until you get at least 5 sats. Sometimes this can take quite awhile, especially if there is tree cover. At the same time, allowing you to temporarily turn off rescue mode could be dangerous since you may forget you don't have that option for a particular flight and your failsafe may be set up for GPS rescue. So, I'm wondering if we could take off with 0 sats, fly around to get a solid GPS lock (it's better in air), and then a warning message will pop up that says something like "GPS Rescue Mode available" (until that point, failsafe defaults to "drop") Then, you simply land nearby, quickly disarm and arm, and you have the home position set. 

So, the routine would be that I'd fly around nearby where I know I won't lose my quad, wait for GPS to get sats, then land quick, and go flying again with GPS rescue enabled. 

The GPS warn message could keep blinking and bothering you until you land and save home position. 

Also, you could have a time limit, say 1 minute or 30 seconds or some number you could set, so that if GPS doesn't come in by that time on initial startup, then it will go right into "delayed GPS Rescue Mode" and allow you to take off early. This way, if you have already got a lock and just powered cycled, and you have a battery in your GPS, delay would not be necessarily.

etracer65, 2019-09-03:
I think you can already do what you're asking (BF 4.0+). Just:
```
set gps_rescue_allow_arming_without_fix = ON
```
If after flying you get a fix, then landing and re-arming should establish the fix and GPS Rescue will be active.

VoicOfReason, 2019-09-04:
Oh! Sorry I asked, in this case. Looks like that should do the trick. I did a GET command in the CLI and did not see this option.

VoicOfReason, 2019-09-04:
Before this thread get closed, perhaps I could suggest an improvement: only when RSSI value is above a certain value (meaning quad is close by) auto set GPS home position mid-air automatically? That would be nice.

etracer65, 2019-09-04:
> Before this thread get closed, perhaps I could suggest an improvement: only when RSSI value is above a certain value (meaning quad is close by) auto set GPS home position mid-air automatically? That would be nice.

Not viable. How close is close by? You could still have a strong radio link with high RSSI with Crossfire over a kilometer away. That's probably not "close".

>Also, is there currently a way to set home position via a switch like iNav can? That would be the only other thing that would helpful to have.

No, this is not possible.

VoicOfReason, 2019-09-04:
> > Also, is there currently a way to set home position via a switch like iNav can? That would be the only other thing that would helpful to have.
> 
> No, this is not possible.

Why is this latter option not possible?

etracer65, 2019-09-04:
Because Betaflight doesn't have this feature.

VoicOfReason, 2019-09-04:
I am fully aware of this fact. That is why I labelled this thread a "FEATURE SUGGESTION". Is Betaflight closed to new features?

etracer65, 2019-09-04:
Perhaps you should be a little clearer with your questions. You asked:

> Also, is there **_currently_** a way to set home position via a switch like iNav can?

And I answered, no. Could it be added as a feature? Maybe. Would be best to close this and open a separate issue with just that as a feature request rather than expecting someone to find it embedded in this issue with other things.

VoicOfReason, 2019-09-04:
> Perhaps you should be a little clearer with your questions. You asked:
>
>
> And I answered, no. Could it be added as a feature? Maybe. Would be best to close this and open a separate issue with just [...]

Got it. Yes, I was unclear in my wording. I'll re-open a new feature request.

closed 2019-09-04 by etracer65