betaflight/betaflight #8810: Feature Request:  Aux channel in OSD

mactac, 2019-09-03:
Long range is getting more and more popular, and in long range, efficiency is king.

To be able to best determine the greatest distance that can be covered on the current flight, it would be sensible to be able to see the ratio of Amps to speed.  In other words, what is the most efficient speed for the current conditions (wind, air pressure, etc).

Being able to have the radio send something like amps divided by speed via an aux channel and then displaying that on the OSD would be absolutely fantastic - similar to the "mpg" display on a car.  You could also estimate rage (via volts left on the battery) if you use a bit of math as well, and even expand that to "will I make it home on the current battery?".

The solution is to add some Aux channels to the OSD.  It should be fairly simple.

There are a lot of other uses for it as well other than long range.

etracer65, 2019-09-03:
Seems like if the "amps / speed" is a valid statistic, then it would make more sense to simply add that as an OSD element. Leaving it generic to pass through an aux channel value is kind of the long way to get there as it forces people to know how to do the programming in their radio and route the result back to a channel. And leaves out people who don't have programmable radios.

Whether the statistic is of value or not is a separate question for discussion. On the surface I'm not sure "Amps per meter per second" is that meaningful a value.

mactac, 2019-09-03:
> Seems like if the "amps / speed" is a valid statistic, then it would make more sense to simply add that as an OSD element. Leaving it generic to pass through an aux channel value is kind of the long [...]

You don't need to do any programming, it's just a custom mix.

For the value part, it has tremendous value to people doing long range (I'm guessing you don't do long range?)  as getting the most efficiency out of your motors/flight is extremely important.  Imagine that you are are trying to push the boundaries of how far you can go on a tank of gas in your car - knowing your MPG is critical so you can drive at the most efficient speed.   If you don't know your MPG, you're essentially guessing.

etracer65, 2019-09-03:
> You don't need to do any programming, it's just a custom mix.

But this still doesn't solve the problem for people that don't have programmable radios, don't have telemetry, etc.

> Imagine that you are are trying to push the boundaries of how far you can go on a tank of gas in your car - knowing your MPG is critical so you can drive at the most efficient speed. If you don't know your MPG, you're essentially guessing.

Well now you've flipped your units. To use the same analogy you originally asked for gallons per mile per hour. Still not convinced that the requested units are worthwhile. If you want to compare to the car analogy you'd want something like distance per amp.

mactac, 2019-09-03:
> > You don't need to do any programming, it's just a custom mix.
> 
> But this still doesn't solve the problem for people that don't have programmable radios, don't have telemetry, etc.

True, though most people doing long distance have telemetry.  Also, I don't think the guiding principle of BF is "everyone has to be able to use it" :)
 
> Well now you've flipped your units. To use the same analogy you originally asked for gallons per mile per hour. Still not convinced that the requested units are worthwhile. If you want to compare to [...]

The units are unimportant - it's the concept.   I know you are not convinced, but you are not the person who would use it.  Just because you do not understand how important it is, doesn't make it unimportant.

LD flyers need to know what the most efficient speed is for the conditions.  A readout would give them that.  Same reason the MPG readout on a car is useful.  Same reason an electric car reads out efficiency.   People find those useful.

etracer65, 2019-09-03:
> Also, I don't think the guiding principle of BF is "everyone has to be able to use it"

Actually yes, wherever possible it is. And in this case it is possible directly in the OSD. So doing it via calculations on telemetry data in the radio and then rerouting that to an aux channel for presentation is a bit a Rube Goldberg approach. Also there are channel range limitations depending on the radio and protocol. For example you can't just send arbitrary values through a channel. Some accept 1000-2000, some 0-1000, and so on. SBUS for example actual encodes the range as 192-1792 to correspond to the full 1000-2000us range. So the point is that even putting the calculated result in the channel in such a way as to present a usable number with the raw channel value would be problematic.

> I know you are not convinced, but you are not the person who would use it. Just because you do not understand how important it is, doesn't make it unimportant.

Whether I would use it or not is irrelevant. The point is that if you want something like this or somebody to work on developing it then yes, we need to be convinced of the value.

mactac, 2019-09-03:
I'd also like to remind you that my proposal was for the Aux channel, no specific units or anything.  People can do with it what they wish.

The first time you are 3km away and your battery is at 3.5v per cell, you'd give anything to figure out what the most efficient speed was :)  This type of situation happens all the time in long distance world.   When we are flying out to a mountain, we want to know the most efficient speed so we can conserve battery as much as possible.

It may not be as useful for the typical park flyer, but like I indicated above a couple of time, efficiency is everything with LD.

mactac, 2019-09-03:
> > Also, I don't think the guiding principle of BF is "everyone has to be able to use it"
> 
> Actually yes, wherever possible it is. And in this case it is possible directly in the OSD. So doing it

Not everyone uses custom motor mixes, camera angle compensations, etc etc.  The list is endless.

> via calculations on telemetry data in the radio and then rerouting that to an aux channel for presentation is a bit a Rube Goldberg approach. Also there are channel range limitations depending on [...]

Only the relative number is important, not the absolute number.  The radio can make it whatever it needs.

Yes, I agree it would be better to have it all done on the FC, but that would make it all that more specific in application.  Simply adding the ability for an Aux channel would leave it open to many uses.  

> Whether I would use it or not is irrelevant. The point is that if you want something like this or somebody to work on developing it then yes, we need to be convinced of the value.

The value is huge, I explained in my previous comment (we posted at the same time).   The value tyo you is zero, I get that.  The value to LD flyers is high.   Like the efficiency readout on your car, it'useful - that's why they put it there.   If you were trying to "hypermile" with your car, it would be indispensable.  LD flyers are trying to do exactly that.  I'm sort of confused as to how you do not see that., actually.   perhaps a miscommunication?

mactac, 2019-09-03:
>he point is that if you want something like this or somebody to work on developing it then yes, we need to be convinced of the value.

It is so important that I'm willing to pay for it.

etracer65, 2019-09-03:
> I'm sort of confused as to how you do not see that., actually. perhaps a miscommunication?

I don't have any problem with the concept of some kind of "efficiency" indicator. The point was that the metrics you proposed are not necessarily convincing and there are probably better ways to represent efficiency.

Also remember that for any change we need to look at the big picture and how this can be used generally by a wider audience. Naturally you're focused on your use-case so it's difficult for you to see this.

> It is so important that I'm willing to pay for it.

Well it doesn't really work this way. The way for you to "pay for it" is to develop the code and do the testing and submit it to the project. Or make a well thought out and convincing argument in the hope the one of the development team wants to take it on.

mactac, 2019-09-03:
> The point was that the metrics you proposed are not necessarily convincing and there are probably >better ways to represent efficiency.

This is what I don't understand.  If you goal was to squeeze out as long a distance as possible from your battery (That's what us long distance flyers constantly obsess over), would an efficiency measurement not be one of the first things you think of that is needed?    Do you not agree with my car analogy?  If your goal is to be able to drive farther and farther on a tank of gas, would you not want to see your fuel consumption on a display so you can be constantly driving as the most efficient rate?

Or is your argument that there aren't many people who fly long distance?  Sure, there are more park flyers, etc, but long distance is likely the fastest growing segment of drone flying.   The amount of 7 inch frames coming out is mind boggling.

 > Also remember that for any change we need to look at the big picture and how this can be used >generally by a wider audience. Naturally you're focused on your use-case so it's difficult for you to >see this.

I stand by my my statement about the growth of long distance.  The alternative is iNav, but I believe that most people would prefer to stay with BF.

>Or make a well thought out and convincing argument in the hope the one of the development >team wants to take it on.

I thought that I did.  I'm not the only one who needs this.

etracer65, 2019-09-03:
> This is what I don't understand. If you goal was to squeeze out as long a distance as possible from your battery (That's what us long distance flyers constantly obsess over), would an efficiency measurement not be one of the first things you think of that is needed?

You really should read my comments. Because I said:

> I don't have any problem with the concept of some kind of "efficiency" indicator. The point was that the metrics you proposed are not necessarily convincing and there are probably better ways to represent efficiency.

mactac, 2019-09-03:
> I don't have any problem with the concept of some kind of "efficiency" indicator. The point was that the metrics you proposed are not necessarily convincing and there are probably better ways to represent efficiency.

Again, I did not propose metrics - I was using this as an example of what you could do if you have a pseudo-customizable OSD element. .  I proposed making *one* customizable OSD element (there are zero currently) - and the best choice would likely be an Aux channel (or more).

I thought that it would be useful to more people (see your argument about usefulness to a wide variety of people) That an efficiency indicator.

If you are saying that telemetry and custom mixes are too hard for people to understand/set up (not sure I agree with that... there are WAY more complicated things in BF right now), then an efficiency indicator would work as a second best option.

Now, it would be best to make this productive, so if you are saying "an Aux channel in the OSD simply will never happen", then....

Since LD flyers care most about maximizing flight distance , then the best metric would involve amps and distance. ... for example mAh per km.  That can be determined from amps and speed.

IvoFPV, 2019-09-03:
@mactac A lot of things could be added to the Betaflight like this but its focus is on flight performance, that's why you probably prefer it over iNav which I assume has these metrics in its OSD. I see a use scenario where you would want to have these but that is specific to a small amount of Betaflight users like yourself. Implementing that is not high up on the list of things devs are focused on.

mactac, 2019-09-03:
> @mactac A lot of things could be added to the Betaflight like this but its focus is on flight performance, 

This is most definitely a flight performance issue.  It is one of the most important parts of flying long distance.

>that's why you probably prefer it over iNav which I assume has these metrics in its OSD. 

it does not

>I see a use scenario where you would want to have these but that is specific to a small amount of >Betaflight users like yourself. Implementing that is not high up on the list of things devs are focused >on.

What are you basing that assumption on?  How do you know that it's a small number?  Like I said, it's likely the largest growing segment of flyers right now.   

Plus, I imagine that it would be ridiculously easy to set up.

jflyper, 2019-09-03:
Aside from the discussion of whether the gas mileage is the flight performance metric or long range is the fastest growing segment in the hobby, it would not be so hard to add an OSD element that displays several RC channel values. I think this is what the OP initially proposed, right?

IvoFPV, 2019-09-03:
@mactac When I say flight performance I think about setpoint tracking, filtering, etc. Doesn't iNav have something like power/km? I thought I have seen that, never mind.
>What are you basing that assumption on? How do you know that it's a small number? Like I said, it's likely the largest growing segment of flyers right now.

It's just my guess but it is not that far fetched to say it is small in comparison to 5" segment. For sure it is on the rise and as I've said I think metrics like what you said are useful and could be added.

mactac, 2019-09-03:
> Aside from the discussion of whether the gas mileage is the flight performance metric or long range is the fastest growing segment in the hobby, it would not be so hard to add an OSD element that displays several RC channel values. I think this is what the OP initially proposed, right?

That's exactly what I proposed.   Seems like it would be super easy, and really helpful for a number of possible applications.

etracer65, 2019-09-03:
So if that's the only point I'm not sure why you opened a duplicate issue. Closing as a duplicate of #8795

closed 2019-09-03 by etracer65

mactac, 2019-09-03:
> So if that's the only point I'm not sure why you opened a duplicate issue. Closing as a duplicate of #8795

The other thread was a question that resulted in this feature request.  Someone else subsequently changed it to a feature request.

Unfortunately all of the useful discussion is in this thread.   I would much rather have preferred the other thread to be closed, since I've linked to this thread from a long range group.

mikeller, 2019-09-03:
@mactac:

> The other thread was a question that resulted in this feature request. Someone else subsequently changed it to a feature request.

That is incorrect. I labelled the #8795 as a feature request hours before you created this new issue - please enable notifications from GitHub, and make sure they get sent to an email address you monitor, to make sure you get notifications for these changes.

Anyhow, this issue is linked from the feature request, so people wanting to participate will find it.