betaflight/betaflight #8811: omnibusf4sd did not support dji hdl protocol.

chowonpapa, 2019-09-03:
I cannot find dji hdl protocol in omnibusf4sd.

McGiverGim, 2019-09-03:
It was removed because it is only a sbus protocol, more info https://github.com/betaflight/betaflight/pull/8716

chowonpapa, 2019-09-04:
Thank you for posting!

Docteh, 2019-09-07:
Reading that PR It sounds like the CLI command `set sbus_baud_fast = on` should use the faster rate, is that correct?

mikeller, 2019-09-07:
@Docteh: Yes.

spatzengr, 2019-09-18:
@chowonpapa , did you ever get this working?  I'm on BF 4.1, RC3 and when i "set sbus_baud_fast=on" the FC loses connection with the RX.

spatzengr, 2019-09-18:
@jerryxin123 , I'm wondering if you could test this.  I have connectivity of the DJI Air Unit as SBUS, but when I switch "set sbus_baud_fast=on" it stops working.  Are their other setting I need to implement?  Do I need to power everything down and back on again?  Or is it just not working?

mikeller, 2019-09-19:
@spatzengr:

> I have connectivity of the DJI Air Unit as SBUS, but when I switch "set sbus_baud_fast=on" it stops working.

Does this mean that the DJI unit you are using is working when `sbus_baud_fast` is `off`? This would suggest that this unit is using SBus at the standard speed (100 kbps). Since SBus is unidirectional auto-detection of speed on the sender side is not possible, so unless you can change the configuration of this device to use a different SBus speed you will have to use it with `sbus_baud_fast = off`.

Unfortunately nobody in the Betaflight team has got an engineering sample of DJI's hardware, so we will not be able to support this or fix anything to do with it.

EggsBenedict, 2019-09-20:
@spatzengr in the goggles you have to tell the Air Unit to switch to "DJI HDL".  I tested this on the NBD **Infinity** FC, which I can't find a Unified target or target for, but just used the STM405 and a previous full dump and nothing's exploded yet :/
_**Setting > Device > Protocol > DJI HDL**_

spatzengr, 2019-09-21:
@EggsBenedict , yep.  Suttumbled upon the same.  Works great.  Standard is around 10ms frames.  Fast is 5ms frames.

closed 2019-10-28 by stale[bot]