betaflight/betaflight #8812: bf 4.1 filtering problem

m013, 2019-09-03:
**Describe the bug**
in the recent builds of BF 4.1  + blheli 32 32.6.6 filtering is broken. i hear clear ringin or grinding noise at some throttle values, usually below 50%

**To Reproduce**
install any september build and try it with rpm filters and without it - the problem appears in both cases

**Expected behavior**
with the builds around 24 of july + blheil32 32.6.5 it was ok

**Flight controller configuration**
check these 2 logs in the archive
normal - july build
blheli - september build

i am experiencing  this problem not only with dysF4 pro, but with mamba 722 mini as well

[BTFL_cli_Leopard_20190902_111633.txt](https://github.com/betaflight/betaflight/files/3570826/BTFL_cli_Leopard_20190902_111633.txt)

[Archive.zip](https://github.com/betaflight/betaflight/files/3570834/Archive.zip)

IvoFPV, 2019-09-03:
@m013 As I can see you have disabled all gyro lowpass filters, and D term lowpass 2 filter and the second dynamic notch. Is that correct? Did you run these settings before on the previous builds and had no issues?

m013, 2019-09-03:
@IvoFPV  ,  strange. i was sure i copy pasted the same config

IvoFPV, 2019-09-03:
@m013 That's because defaults recently changed to use PT1 instead of BIQUAD for D term lowpass. That means that setting wasn't in your `diff` so it was not changed back to your previous settings of BIQUAD when you pasted it.

m013, 2019-09-04:
@IvoFPV  , damn you are right
how could i not notice that twice in the configurator
after i manually set it to biquad it started to fly 
works great with the latest blheli \
thanks!

closed 2019-09-04 by m013