betaflight/betaflight #8814: Video Transmitter tab shows "Not configured" when SA enabled on SS port

atomgomba, 2019-09-04:
On the Video transmitter tab I get the "Device not configured" notice when SmartAudio is enabled on soft serial. When I use a hardware UART it shows up okay.

atomgomba, 2019-09-06:
So as a short follow up, I just got and installed a Nano Pro32 and Smart Audio works and Configurator works as well. Smart Audio is still configured on SOFTSERIAL2, but it is on PWM5 hardware pin. When battery plugged in Configurator shows "Device ready: yes".

mikeller, 2019-09-08:
Ok, I figured this out:

On OMNIBUSF4SD, the B10 pin does not have a timer associated. This means that opening softserial 2 will fail because it is assigned to it and it needs a timer to work. As a consequence the VTX system is not initialised for lack of a serial port.

Try

```
timer B10 AF1
```

and VTX control on softserial 2 should be fine.

closed 2019-10-15 by stale[bot]