betaflight/betaflight #8815: No magnetometer support in FURYF4OSD

MordorGuy, 2019-09-04:
It seems to me that there is no support for I2C communication with magnetometer on BN-880 GPS
(no set mag_hardware = auto command in CLI)
Im using MambaF405 FC and I managed to get it working with INav but no luck with latest stable release of Betaflight.
Any ideas?

etracer65, 2019-09-04:
When the manufacturer created the target they didn't include magnetometer support (not compiled in or configured).

etracer65, 2019-09-05:
@MordorGuy I added magnetometer support to the FURYF4OSD target. Can you please test the firmware attached below?

[betaflight_4.0.6_FURYF4OSD.hex.zip](https://github.com/betaflight/betaflight/files/3581217/betaflight_4.0.6_FURYF4OSD.hex.zip)

MordorGuy, 2019-09-06:
Works good, thanks

closed 2019-09-06 by MordorGuy

Yeetakus, 2020-09-04:
How do we upload this firmware?

PR #8834 (2019-09-07): Add missing MAG support for FURYF4 and FURYF4OSD targets
Fixes #8815 

Should be cherry-picked into 4.0-maint if there are any more releases.