betaflight/betaflight #8819: TalonF7v2 and TalonF7fusion targets not in firmware dropdown

Gozzz, 2019-09-04:
Both TalonF7 unified targets are not showing up in the firmware flasher dropdown.

Check firmware flasher tab on configurator 10.5 and 10.6, both do not show the TalonF7v2 or Talonf7fusion targets but they do show other targets, such as tmotorf7 which were recently merged. Enabled "expert mode", and "show unstable release"

TalonF7v2 https://github.com/betaflight/betaflight/pull/8650

Talonf7fusion- https://github.com/betaflight/betaflight/pull/8649

closed 2019-09-04 by Gozzz

mikeller, 2019-09-05:
@Gozzz: Try now with a configurator build from current `master`. :grinning:

Gozzz, 2019-09-05:
@mikeller 10.6 is sooo pretty on dark mode macOS. ;-) I see them in there now! Thanks!

flyTib, 2020-06-04:
hi !
i juste flash my talon f7v2 mpu6000 with the talonf7v2 target.
i can not see the "user1" mode ...so my board do not give me 5V output for cam and vtx .

thks !