betaflight/betaflight #8820: FEATURE REQUEST: Reset home GPS position mid-air with a switch

VoicOfReason, 2019-09-04:
Just a quick feature suggestion: add reset GPS home position via a switch for GPS Rescue Mode. iNav has this, I believe. The purpose: combined with the CLI command to launch without GPS fix, you could fly around and get a quicker GPS fix above the trees and then set home position when your sats are sufficient.

Other uses would be if you changed locations while flying, such as flying your quad from a moving target, e.g. boat, car, etc.

Mateyhv, 2019-09-23:
Agree, would like to see such a feature. I don't think it would be too hard to implement.

AmiSMB, 2019-09-24:
This functionality is in [iNav](https://github.com/iNavFlight/inav). Maybe this can be back ported

closed 2019-10-31 by stale[bot]