betaflight/betaflight #8821: Bad altitude value with Baro AND GPS

Aliasname, 2019-09-04:
FuryF4OSD 4.0.6

Hello together,

when i use a barometer (BMP280) with GPS (BN-220) the altitude value is not right in the OSD.
It seems to be divided by 2 (round about and it jumps very often about 50 Meter up or down).

When i use the baro as single sensor it shows me a zero at startup. Even here not the real height is shown. Maybe thats normal. When i arm and moved the quad up or down, it levels again to 0. Like with gps.

When i use GPS as single Sensor it shows me the real height in OSD.

When i check the values under "Sensors" in Betaflight i receive the very same values as in the OSD.

I already asked myself which height the fc would take. It seems it use both values an build an average.
I expected it would take the baro value (what else should a barometer be for) or i expected i could choose between the sensor for the altitude value.

Here are further infos about:
version, tasks, diff. etc.

[diff.txt](https://github.com/betaflight/betaflight/files/3576310/diff.txt)

1. Is it possible to show the real height with barometer?
2. Is it possible to use both sensors and get a correct height value and/or
3. is it possible to choose which sensor for altitude should be use?

If you need further informations please let me know.
Thank you in advance.

IvoFPV, 2019-09-04:
1. Yes and no. It would be possible to use air pressure and determine the actual height above sea level like it is done in aviation, this requires 'calibration' before flight where you set the current air pressure at sea level at your current location. No, this is not how it works in Betaflight. Betaflight only calculates how much you deviated from your reference pressure when you armed. You get height above ground level.
2. Betaflight used both BARO and GPS if available to get the most precise height possible. Again, this is above ground level, i.e above your take off location height.
3. I think it is possible in the latest version 4.1 which hasn't been released yet.

Barometers on all current FCs seem to be really bad in terms of getting a reliable readout therefore you often see jumps in readings or consistent rising as the sensor gets hotter.

etracer65, 2019-09-04:
Baro shows altitude relative to position when the quad was armed. GPS shows absolute altitude (above sea level) when disarmed, but when armed it switches to relative altitude above ground level. GPS altitude can jump around quite dramatically if you don't have a lot of satellites and a strong 3D fix.

Aliasname, 2019-09-05:
@IvoFPV Thank you very much to let me understand the bahavior with Baro and GPS.
At first view it looks like wrong behavior. But now it makes absolute sense.
Nevertheless both values together leading to missunderstanding when you are not armed.
Especially when you know the real heightat this place.
@etracer65 Thank you. Barometer values are very stable. GPS jumps more often around.
I wish i could choose only one sensor or give priority.
But like @IvoFPV said, it probably finds place in version 4.1.

Thank you very much.

closed 2019-09-05 by Aliasname