betaflight/betaflight #8822: ESC Telemetry not working with Bit-banging Dshot active

netzwergracer, 2019-09-04:
**Describe the bug**
When configuring DSHOT telemetry with the new Bit-banging Dshot, the ESC telemetry (to get current and voltage from the ESCs) stops working.
When configuring DSHOT telemetry the "old" way via timer and dma adjustments, the ESC telemetry works just fine.

**To Reproduce**
Steps to reproduce the behavior:
1. install latest Betaflight 4.1 build on CLRACINGF4
2. try activating ESC telemetry on RX4 like usual

**Expected behavior**
ESC telemetry should work like normal

**Flight controller configuration**
(ESC telemetry disabled here)
'''

# version
# Betaflight / CLRACINGF4 (CLR4) 4.1.0 Sep  4 2019 / 11:50:43 (0d62e71a8) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF4
mcu_id 001a00243436471837333638
signature 

# feature
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 1 1300 1700 0 0
aux 2 26 2 900 1300 0 0
aux 3 31 2 1700 2100 0 0
aux 4 35 3 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 500

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 225
set gyro_lowpass2_hz = 350
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 900
set rssi_scale = 200
set serialrx_provider = FPORT
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set current_meter = NONE
set vbat_scale = 220
set vbat_divider = 20
set yaw_motors_reversed = ON
set small_angle = 180
set transient_throttle_limit = 0
set osd_cap_alarm = 1400
set osd_rssi_pos = 6209
set osd_tim_2_pos = 2485
set osd_vtx_channel_pos = 2517
set osd_ah_sbar_pos = 4302
set osd_ah_pos = 4174
set osd_current_pos = 449
set osd_mah_drawn_pos = 4515
set osd_avg_cell_voltage_pos = 6562
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 2
set vtx_freq = 5732
set vcd_video_system = PAL
set camera_control_internal_resistance = 270
set camera_control_button_resistance = 450,150,68,18,0

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 100
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass2_hz = 200
set vbat_pid_gain = ON
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set d_pitch = 50
set d_roll = 43
set d_min_roll = 22
set d_min_pitch = 24
set idle_min_rpm = 16

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 27
set pitch_expo = 27

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
'''

**Setup / Versions**
CLRACINGF4 FC
Bardwell 4in1 ESC
R9MM receiver
Matek HV VTX

mikeller, 2019-09-04:
@netzwergracer:
- what version of firmware are you running on your ESCs?
- did you recently update the ESC firmware?

netzwergracer, 2019-09-04:
I am running 32.6.6 with GCR on my ESC.

DSHOT telemetry is working fine both ways, set up the new bit-banging way or the "old" way via timer and dma. 

I updated the firmware to 32.6.6 when it came out. Have been running RPM filter for a long time already. All good.

Just since today when I tried the new bit-banging DSHOT, ESC telemetry does not work.

joelucid, 2019-09-05:
@netzwergracer, could you paste the outputs of the cli commands ``resource show all`` and ``resource`` with esc telemetry enabled both with ``dshot_bitbang=off`` and ``dshot_bitbang=auto``, please?

joelucid, 2019-09-05:
@netzwergracer I have a fix here: https://github.com/betaflight/betaflight/pull/8828. Untested since I don't own a serial esc telemetry setup. Could you check if this binary fixes it?

[betaflight_4.1.0_CLRACINGF4.hex.zip](https://github.com/betaflight/betaflight/files/3578449/betaflight_4.1.0_CLRACINGF4.hex.zip)

netzwergracer, 2019-09-05:
Yes, this fixes the problem. 
Many thanks for the fast reply, I will send a Paypal donation, 

Voltage reading with ESC telemetry working:
[image]

netzwergracer, 2019-09-05:
Current reading from ESC telemetry is working as well:

[image]

joelucid, 2019-09-05:
Excellent - thank you!

Gozzz, 2019-09-06:
@joelucid can you make a test hex for TalonF7v2? I had the same issue with mine and would like to test the fix (unless #8828 is going to be merged soon?)

Thanks,
Mike

ecob81, 2019-09-06:
Hi guys, the dshot_bitbang also caused my rpm filter to stop working at dshot 600 on a kakute F7 and Tmotor F55a V1. I started to get packet errors that weren't occurring previously and had to reduce dshot to 300 or turn off bitbang. Would you like me to open a separate issue or will it get fixed with this one?

joelucid, 2019-09-06:
Separate issue please for the dshot600 issue.

closed 2019-09-07 by mikeller