betaflight/betaflight #8823: Unified Targets - MSP_MOTORS doesn't like missing motors.

Docteh, 2019-09-05:
**Describe the bug**
MSP_MOTOR Hangs the board when no motors are defined. Or less than 4? Not enough motors? I'm still looking into this part.

**To Reproduce**
Steps to reproduce the behavior:
Flash a unified target, connect the configurator and look at either the `Motors` or `Tethered Logging` tabs, this will cause a MSP_MOTOR command to be sent to the FC

**Expected behavior**
Please no hang/crash

**Flight controller configuration**
This is a minimal configuration that _doesn't_ hang when I try and look at the Motors tab.
```
# resources
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02

# timer
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer A02 AF1
# pin A02: TIM2 CH3 (AF1)

# dma
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 6 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3
```
If I free one of the motor pins and restart, I now have trouble using the motors tab.

**Setup / Versions**
Bare KAKUTEF4V2

**Additional context**
Something like `MSP.send_message(104,undefined,undefined,undefined,undefined);` in the Configurators Javascript Console can be used to minimalistically crash a board

jflyper, 2019-09-05:
What version is this happening?

jflyper, 2019-09-05:
Ah, I found it. `motorIsEnabled` function in `vTable` for `motorDevice` is not initialized for non-existent motor case. I'll fix it.

```
(gdb) p *motorDevice
$2 = {vTable = {convertExternalToMotor = 0x801f305 <motorConvertFromExternalNull>, 
    convertMotorToExternal = 0x801f311 <motorConvertToExternalNull>, 
    enable = 0x801f2f1 <motorEnableNull>, disable = 0x801f2f5 <motorDisableNull>, 
    isMotorEnabled = 0x0, updateStart = 0x801f2f7 <motorUpdateStartNull>, 
    write = 0x801f2fb <motorWriteNull>, writeInt = 0x801f2fd <motorWriteIntNull>, 
    updateComplete = 0x801f2ff <motorUpdateCompleteNull>, 
    shutdown = 0x801f301 <motorShutdownNull>}, count = 0 '\000', initialized = false, 
  enabled = false}
(gdb) p motorDevice->vTable.isMotorEnabled
$3 = (_Bool (*)(uint8_t)) 0x0
```

Docteh, 2019-09-05:
Yeah, version 4.1 Sorry.

Thanks for tracking it down.

PR #8825 (2019-09-06): Provide isMotorEnabled function for motorNullDevice
Provides pragmatic fix for #8823 

#8823 was caused by two reasons

- `isMotorEnabled` in `vTable` for `motorNullDevice` was not initialized.
- `MSP_MOTOR` is not checking for `motorIsEnabled()` (motor subsystem status; whereas `motorIsMotorEnabled()` is individual motor status) first.

This PR fixes both of these.

Fixes #8823
closed 2019-09-06 by mikeller