betaflight/betaflight #8831: Clicking on "Reset Settings" in BF configurator disconnects FC from USB

0crap, 2019-09-05:
Using BF configurator 10.5.1 I flashed the latest nightly on my FC. (MATEKF411RX)
After flash, with Full chip erase, connect again.
In Setup click on Reset Settings, the FC will now disconnect from USB.

This might be intentional behaviour but I'm not sure.
Nightly build 1844 is still OK, build 1845 and higher gives above behaviour.

mikeller, 2019-09-05:
@0crap: Yes, this change is intentional - for Unified Targets a reboot is required  after resetting the configuration in order to get changed pin definitions to be applied.

But well spotted. :eyes:

0crap, 2019-09-05:
Ahhh, OK. Just wanna make sure it was not a bug,
Thx!
I'll close this one.

closed 2019-09-05 by 0crap

mikeller, 2019-09-05:
It's good to have QA keep an eye on this project. :smiley:

0crap, 2019-09-05:
No problem, the least I can do. 👍