betaflight/betaflight #8836: Bidirectional DSHOT not working on SPRACINGH7EXTREME

TheIsotopes, 2019-09-06:
**Describe the bug**
Tested 4.1.0 RC1 on a new build with SPRACINGH7EXTREME and found that the bidirectional dshot not working ... Motor 4 does not spinning, shows 100% errors in the motor tab and in blhelisuite32 he is not listed ... if i'm `set dshot_bidir = OFF` all is working without issue

**Flight controller configuration**
```
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Sep  5 2019 / 20:27:09 (62f0eb7ed) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# name: Mark1+

# feature
feature -TELEMETRY
feature -RSSI_ADC
feature -OSD
feature -TRANSPONDER

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 9,2::AW:0
led 1 9,3::AW:0
led 2 9,4::AW:0
led 3 9,5::AW:0
led 4 6,5::AW:0
led 5 6,4::AW:0
led 6 6,3::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 0 3
mode_color 6 1 6

# aux
aux 0 0 0 1600 2000 0 0
aux 1 13 1 1000 1200 0 0
aux 2 26 0 1800 2000 0 0
aux 3 31 0 1200 1400 0 0
aux 4 36 1 1800 2000 0 0

# master
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SUMD
set blackbox_device = SPIFLASH
set dshot_idle_value = 150
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1550
set use_cbat_alerts = ON
set cbat_alert_percent = 20
set vbat_duration_for_warning = 50
set vbat_duration_for_critical = 20
set pid_process_denom = 1
set name = Mark1+

profile 0

# profile 0
set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set idle_min_rpm = 28

rateprofile 0

# rateprofile 0
set roll_rc_rate = 135
set pitch_rc_rate = 135
set yaw_rc_rate = 120
set roll_expo = 15
set pitch_expo = 15
set roll_srate = 75
set pitch_srate = 75

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: FREE
A05: FREE
A06: VTX_DATA 10
A07: VTX_CLK 10
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: GYRO_CS 1
B00: VTX_CS 10
B01: VTX_POWER
B02: QSPI_CLK 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: QSPI_BK1CS 1
B11: FREE
B12: GYRO_CS 2
B13: FREE
B14: FREE
B15: SERIAL_RX 1
C00: ADC_CURR
C01: ADC_BATT
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: FREE
C05: ADC_EXT
C06: FREE
C07: FREE
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SDIO_CMD
D03: SPI_SCK 2
D04: GYRO_EXTI
D05: FREE
D06: SPI_MOSI 3
D07: BEEPER
D08: FREE
D09: FREE
D10: FREE
D11: QSPI_BK1IO0 1
D12: QSPI_BK1IO1 1
D13: QSPI_BK1IO3 1
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: QSPI_BK1IO2 1
E03: LED 1
E04: SYSTEM
E05: CAMERA_CONTROL
E06: FREE
E07: QSPI_BK2IO0 1
E08: QSPI_BK2IO1 1
E09: QSPI_BK2IO2 1
E10: QSPI_BK2IO3 1
E11: PREINIT 3
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: GYRO_EXTI
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15:
    CH1: CAMERA_CONTROL
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 1
DMA1 Stream 1: MOTOR 2
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - SP Racing H7 Extreme
 - Aikon AK32 35A
 - EMAX Eco 2306 2400KV

McGiverGim, 2019-09-06:
Have you tested with DSHOT300? It seems there are errors when used with DSHOT600.

jflyper, 2019-09-06:
Bit-banging DShot is not available for H7 yet.

If you have `dshot_bitbang` turned on, turn it off.

TheIsotopes, 2019-09-06:
@McGiverGim tested with DShot300 but same issue

@jflyper yep, H7 have no bitbang and yes motor 4 is not spinning

emski0, 2019-09-06:
@TheIsotopes I have the dshot_bidir working on SPRACINGH7EXTREME on BF 4.1 # 1846 (sept 3rd) with Dshot1200. Maybe try with this build

emski0, 2019-09-06:
@TheIsotopes Tried RC1 with dshot600, all motors spins with dshot_bidir.
N.B: i have motor 4 resourced to  motor 8 pin.
Also having above 55% error on motor 3

emski0, 2019-09-09:
flight tested with dshot_bidir = on, few seconds into the flight, the quad did continuous roll and crashed. Two of the motors kept moving after disarm. (happened twice on  BF4.1 Build #1857 and Build #1864. 
Also large error on all motors: 15%(M1) 15%(M2) 50%(motor 3) 15%(M4).
disabled the dshot_bidir, and the quad flies normal.

TheIsotopes, 2019-09-14:
@hydra do you have an idea why this does not work

hydra, 2019-09-15:
@TheIsotopes no.  It's especially puzzling because the H7NANO and H7EXTREME use the same pins on the CPU for the motor outputs...

Im not seeing any details.9f exactly which firmware you guys are using.

@emski0 your issue might be signal related. can you make sure you twist the ground and motor signal wires together.  photos of your build as-is and after twisting would help here.

do either if you have a scope?  Looking at the signals on a scope would likely be revealing.

emski0, 2019-09-15:
@hydra I don't have additional ground connected with the signal. It is hard for me now to try adding ground. Probably i will try adding the ground and twisting when i can find time.
But do you have the  RPM filter working normally?
I am using blheli ver 32.6.6.
I don't have scope.

TheIsotopes, 2019-09-15:
@hydra 
> It's especially puzzling because the H7NANO and H7EXTREME use the same pins on the CPU for the motor outputs...

therefore my wondering ... on my H7Nano it works without problems

> Im not seeing any details of exactly which ESC firmware you guys are using.

using BLheli 32.6.6

Do you have the BiDir DShot / RPM filter working normally on H7Extreme?

hydra, 2019-09-16:
> Do you have the BiDir DShot / RPM filter working normally on H7Extreme?

@TheIsotopes I've not tested it recently but yes it has been reported working fine by many people.  There's no reason why it wouldn't work either as theres nothing between the MCU pins and the motor signal pads nor are there any hardware conflicts.

This needs some investigation.  Can you perhaps try some newer and older firmware to see if you can find when it worked and when it stopped working?

maGfpv, 2019-09-16:
In my case i never get it working, but i will try the rc3 and let you know if its working.

haslinghuis, 2019-09-16:
@TheIsotopes, Can you please compare the output off CLI command "resource show all" in bf configurator 10.6.0 on the Nano H7 and Extreme H7 if there is DSHOT_BITBANG on TIM8 and DMA Stream? (with dshot_bidir = on).
Also when drone is connected to LIPO without USB connection the ESC init sounds abnormal (doing 2 times three notes and after that the final two notes),

TheIsotopes, 2019-09-18:
Tested now the first `build #1813` that supported bidir dshot on h7 ... this is the build there #8712 was merged ... also here with activated bidir dshot, i get motor 4 not to spin ... bidir dshot off and all motors are spinning

maGfpv, 2019-09-19:
Ich checked the rc3 and have the Same issues as mentioned above. 

How could it be that motor one starts to Spin (automatic arm) with USB and lipo connected. I thought this should mit Happen?

hydra, 2019-09-19:
yeah, that sounds really really messed up. Hopefully @joelucid or @jflyper have some time soon to investigate this issue.

 it is critical that you are testing with blheli firmware that definitely supports GCR.

maGfpv, 2019-09-19:
@Hydra "it is critical that you are testing with blheli firmware that definitely supports GCR."

Im using this blheli32 firmware: 

"https://github.com/bitdump/BLHeli/tree/master/BLHeli_32 ARM/BLHeli_32 Test code Rev32.6.4 hex files/Rev32.6.6 for inverted bidirectional dshot and GCR return data only"

And this hex for my esc tekko32_f3_metall

"https://github.com/bitdump/BLHeli/blob/master/BLHeli_32 ARM/BLHeli_32 Test code Rev32.6.4 hex files/Rev32.6.6 for inverted bidirectional dshot and GCR return data only/TEKKO32_F3_Metal_Multi_32_66.Hex"

maGfpv, 2019-09-19:
another try:

Flashed ESC again. -->Everything fine OK!
Flashed betaflight rc3 with full chip erase. -->OK!
activated the rpm filter using the following snippet: "# set looptime to 4k to have enough computational time
set gyro_sync_denom = 2
set pid_process_denom = 1
# Start with dshot 300 which is most robust
set motor_pwm_protocol=DSHOT300
# basic settings required to enable rpm filtering:
set dshot_bidir=on
# enable debugging for noise assessment
set debug_mode=gyro_scaled
# reconfigure dynamic notch filter to focus on narrow frame resonances
# since motor noise will already be eliminated by the rpm filter.
set dyn_notch_range = medium
set dyn_notch_width_percent = 0
set dyn_notch_q = 250"

while im pasting in the cli. motors start to spin before i saved the cli commands.
-->ERROR

than i restarted and pluged in the lipo and usb.
On tab motors i get 100% Errro for Motor4 (which is going to be motor 1 because i switched the rotation of the board.)

Motor 1-3 are working

here is a diff: 

diff
# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Sep 15 2019 / 09:12:03 (644ab5d90) MSP API: 1.42
# start the command batch
batch start
board_name SPRACINGH7EXTREME
# serial
serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
# master
set gyro_sync_denom = 2
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
set debug_mode = GYRO_SCALED
profile 0
rateprofile 0
# end the command batch
batch end

i forgott to mention. i tired first dshot300 and afterwards 600

maGfpv, 2019-09-19:
@hydra you reported that some users run the rpm succesfull. can you share wich build they use?

maGfpv, 2019-09-19:
So, after hours of flashing.
For every try i set no boot sequenz to disabled and erase chip to enabled.

Here is a summary...
After successfully flashing and connecting via usb i pasted the following diff:
 
# set looptime to 4k to have enough computational time
 
set gyro_sync_denom = 2
set pid_process_denom = 1
 
# Start with dshot 300 which is most robust
 
set motor_pwm_protocol=DSHOT300
 
# basic settings required to enable rpm filtering:
 
set dshot_bidir=on
 
# enable debugging for noise assessment
 
set debug_mode=gyro_scaled
 
# reconfigure dynamic notch filter to focus on narrow frame resonances
# since motor noise will already be eliminated by the rpm filter.
 
set dyn_notch_range = medium
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
 
 
Saved, rebooted and unplugged usb and connected the lipo.

Always the same behavior.
After pasting the snippet, motors will arm not synchron.
Motor 4 always starts to spin.
 
Firmware downloaded from hydra at:
https://github.com/spracing/betaflight/issues/1
 
betaflight_4.1.0_SPRACINGH7EXTREME-20190805.zip
-->Programming was unsuccessful
 
betaflight_4.1.0_SPRACINGH7EXTREME-20190813.zip
-->flashing ok. BUT Not possible to connect via usb

betaflight_4.1.0_SPRACINGH7EXTREME-20190816.zip
-->flashing ok. RPM -->ERROR
 
 
4.1.0#1793-->Programming was unsuccessful
4.1.0#1798-->Programming was unsuccessful
4.1.0#1799-->flashing ok. BUT Not possible to connect via usb
4.1.0#1805-->flashing ok. BUT Not possible to connect via usb
4.1.0#1812-->flashing ok. BUT Not possible to connect via usb
4.1.0#1815-->flashing ok. BUT Not possible to connect via usb
4.1.0#1818-->flashing ok. RPM -->ERROR
4.1.0#1821-->flashing ok. RPM -->ERROR
4.1.0 #1822-->flashing ok. RPM -->ERROR
4.1.0 #1824-->flashing ok. RPM -->ERROR
4.1.0 #1825-->flashing ok. RPM -->ERROR
4.1.0 #1827-->flashing ok. RPM -->ERROR
4.1.0 #1829-->flashing ok. RPM -->ERROR
4.1.0 #1830-->flashing ok. RPM -->ERROR
4.1.0 #1832-->flashing ok. RPM -->ERROR
4.1.0 #1835-->flashing ok. RPM -->ERROR
4.1.0 #1838-->flashing ok. RPM -->ERROR
4.1.0 #1839-->flashing ok. RPM -->ERROR
 
 
4.1.0 RC1-->flashing ok. RPM -->ERROR
4.1.0 RC2-->flashing ok. RPM -->ERROR
4.1.0 RC3-->flashing ok. RPM -->ERROR
 
4.1.0 #1885 -->flashing ok. RPM -->ERROR

I assume, it never worked :D

@hydra please share the build which has worked for others?
Did you ever tested the RPM filter by your own ???

hydra, 2019-09-19:
@maGfpv 'erase chip' doesn't do anything when flashing H7 EXST target.  that's only for FC's that have config stored on the CPU itself.  To erase the config you must either use the cli 'defaults' command or use the bootloader to erase the config partition on the flash as per the flashing instructions.

you didn't say which firmware you used on your ESCs, older FC firmware needs older non-GCR blheli firmware.

It definitely HAS worked because I've had reports from users saying how great it is.  Robert Leclercq has a channel with videos of it working and has posted on facebook about it too - https://www.youtube.com/user/BobbyBIII/videos 

I've not had time to work on H7 specific DSHOT myself and have been relying on the authors of the DSHOT code to work on it and test it - i.e. @joelucid and @jflyper - I have done some cleanup and performance optimization work which benefited all MCUs a few weeks ago.

I'm pretty sure @joelucid has tested H7 bi-directonal DSHOT on his JESC firmware too.

@joelucid can you confirm that here? did you test it on any BlHeli firmware? if so specific ESC hardware and firmware revision and specific FC hardware and git revision would be useful here.

@jflyper can you also confirm if you have tested your H7 bi-directional dshot code on any BlHeli firmware?  again if so specific ESC hardware and firmware revision and specific FC hardware and git revision would be useful here.

Due to the vast amount of code, FCs and peripherals it's not possible for one developer to test everything. The nature of BF is that work is done, authors test it on hardware they have access to, people review it and it is merged.  Sometimes, because the authors don't have access to specific hardware, or because the specific combination was not tested, things get broken.  This is why BF has a release candidate phase, devs think 'it should work' but not everything does all the time.

Without the ability to test every line of code on every piece of hardware with every possible combination of user setting this will always be the case, being 'Mr Cleanflight' I'd prefer it if more focus was placed on running firmware built by the continuous integration system on real hardware, automatically, as part of the build process, so that failures of key features can be identified before the build is released to users.  Alas the BF development team doesn't see this as being as valuable as I do and no-one else other than me has attempted to do anything about this situation.  I've personally spent upwards of 1000EUR on designing and building a flight-control test system which can be interfaced to a raspberry pi which allows this, but there's a SIGNIFICANT development effort required to have the system developed further and integrated into BF build system and I can't currently justify spending time on it because I need to keep the bills paid.

[image]

hydra, 2019-09-19:
I just looked through my conversations regarding H7 + BiDirectional DSHOT, here's the comments from Robert about his experience:

```
RL> So i got all FW up to the latest on FC 4.1/8.13 build and ESC 32.66.
RL> woohoo!
DC> so bi-directional working now?
DC> what ESC firmware and ESCs are you using?
RL> Tekko32 f3 metal 32.66
RL> all working great hovers awesome, need to add my tune and see how she rips
RL> Smoothest my 7" has ever flown!!!!
RL> Bidir Dshot1200 8k/8k at 4% load!! Suck on that F7!
```

emski0, 2019-09-19:
@maGfpv  try erasing the the flash using bootloader function : " 8 Erase entire external flash" , from https://github.com/spracing/betaflight/issues/1 
Before this, i had all motor spinning with dshot_bidir, but with large error. i noticed the while trying flashing several firmwares, at some point this behaviour started : the motors keep beeping at some interval when enabling dshot_bidir, and then only motor 3 spins. 
After Erasing the external flash , all motor spin again with dshot_bidir, but still having large error.

haslinghuis, 2019-09-19:
@hydra, Where to find build 8.13??? I like to try this fw. Also when using defaults after flashing a new firmware should have the same effect as erasing the config partition? Or does emski0 have a point using bl function 8?

hydra, 2019-09-19:
@haslinghuis it's on this issue - https://github.com/spracing/betaflight/issues/1

maGfpv, 2019-09-20:
@hydra and @emski0 this is exactly my hardware 
(Tekko32 f3 metal 32.66), so it should work. i will try " 8 Erase entire external flash" and check if its working. I will report. Thx in advance

maGfpv, 2019-09-20:
So i tried: 
-8 Erase entire external flash
- Flashed FC 4.1/8.13 build and ESC 32.66
-Betaflight coinfigurator is the one listed here: https://github.com/spracing/betaflight/issues/1
-Dshot1200 and 8k8k
-Snippet: 
set dshot_bidir=on
set debug_mode=gyro_scaled
set dyn_notch_range = medium
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set motor_poles = 14

-save

=> Behavior has changed but still not working. Now, motor4 doesnt start to spin automatically but wont start via motor tab. Blheli32 suite shows me no input frames / no bad frames. 

:(

haslinghuis, 2019-09-20:
## After flashing this firmware
https://github.com/spracing/betaflight/files/3497050/betaflight_4.1.0_SPRACINGH7EXTREME-20190813.zip

## Paste in CLI (MAKE SURE TO HAVE PROPS OFF - I DID NOT AND HAD FLY AWAY):

defaults

resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 A00
resource MOTOR 4 A01

# feature
feature -LED_STRIP
feature -OSD
feature -TRANSPONDER
feature -AIRMODE
feature GPS

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 2 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 13 2 1700 2100 0 0
aux 3 28 1 1700 2100 0 0

# master
set serialrx_provider = CRSF
set motor_poles = 16
set align_board_roll = 180
set gps_provider = UBLOX

set gyro_sync_denom = 2
set pid_process_denom = 1
set motor_poles = 16
set motor_pwm_protocol=DSHOT300
set dshot_bidir=on
set debug_mode=gyro_scaled
set dyn_notch_range = medium
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
save

## BE AWARE - WHEN SAVING - MOTORS WILL SPIN FAST !!!

### all motors are spinning !!!

dshot_telemetry_info
---> is not Enabled (has to reboot first)

### reconnect lipo+usb

dshot_telemetry_info

Dshot reads: 5766344
Dshot invalid pkts: 1
Dshot irq micros: 2

Motor     RPM   Invalid
=====   =====   =======
    0       0     0.00%
    1       0   NO DATA
    2       0     0.00%
    3       0     0.00%

# Then spin all motors with master slider in motor tab

Dshot reads: 578849
Dshot invalid pkts: 0
Dshot irq micros: 2

Motor     RPM   Invalid
=====   =====   =======
    0      22     0.00%
    1       0   NO DATA
    2      23     0.00%
    3      23     0.00%

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

haslinghuis, 2019-09-20:
It has nothing to do with Camera Control sharing TIMER 5 with Motors?

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 3
A01: MOTOR 4
A02: MOTOR 1
A03: MOTOR 2
A04: FREE
A05: FREE
A06: VTX_DATA 10
A07: VTX_CLK 10
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: GYRO_CS 1
B00: VTX_CS 10
B01: VTX_POWER
B02: QSPI_CLK 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: QSPI_BK1CS 1
B11: FREE
B12: GYRO_CS 2
B13: FREE
B14: SERIAL_TX 1
B15: SERIAL_RX 1
C00: ADC_CURR
C01: ADC_BATT
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: ADC_RSSI
C05: ADC_EXT
C06: FREE
C07: FREE
C08: SDCARD
C09: SDCARD
C10: SDCARD
C11: SDCARD
C12: SDCARD
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SDCARD
D03: SPI_SCK 2
D04: GYRO_EXTI
D05: FREE
D06: SPI_MOSI 3
D07: BEEPER
D08: SERIAL_TX 3
D09: SERIAL_RX 3
D10: SDCARD_DETECT
D11: QSPI_BK1IO0 1
D12: QSPI_BK1IO1 1
D13: QSPI_BK1IO3 1
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: QSPI_BK1IO2 1
E03: LED 1
E04: SYSTEM
E05: CAMERA_CONTROL
E06: FREE
E07: QSPI_BK2IO0 1
E08: QSPI_BK2IO1 1
E09: QSPI_BK2IO2 1
E10: QSPI_BK2IO3 1
E11: PREINIT 3
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: MOTOR 3
    CH2: MOTOR 4
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15:
    CH1: CAMERA_CONTROL
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 1
DMA1 Stream 3: MOTOR 2
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

haslinghuis, 2019-09-20:
@maGfpv tried flashing in different order:

Scenario 1: reset fc; flash esc 32.6; flash fc to 3.18; enable dshot_bidir
Scenario 2: reset fc; flash esc 32.4; flash fc to 3.18; flash esc to 32.6; enable dshot_bidir

(Each step I disconnected lipo and usb).
No difference. Motor 4 not working. Arm disable flag: RPM Filter

But still - when !dshot_bdir  then esc init sound 3-3-2 instead of 3-2 beeps. When dshot_bdir is enabled than we have out of sync esc init. Do you have the same experience?

emski0, 2019-09-20:
@hydra  i added ground wires and twisted them around the signal wires.
I noticed when the camera is unplugged i get 0% error, and all the motor spins.
[image]

When i plug the camera the error appears (signal ground 5v), (plugging & unplugging the camera control control has no effect) :
[image]

So the error is coming from the camera. 

Btw : BF 4.1 RC3 and tekko32 F3 ver. 32.6.6( furling32 target)

haslinghuis, 2019-09-20:
My camera is not connected to FC. I am not using OSD. In my case the cam is wired only to the Evo VTX, which is only connected with VBAT + GND pad on H7 ? But I do not have connected telemetry wires from the ESC's (Siskin ST 32 Plus 32.6.6).
.

emski0, 2019-09-20:
@haslinghuis  you don't need the ESC telemetry wire, with GCR it send the the rpm via the signal wire.
Just so you know, with BF 4.1 you need blheli32 ver 32.6.6. So no point of trying to flash lower version of blheli. Since the rpm filter was added for H7 in BF 4.1, so  rpm filter does not work on H7 with Blheli version < 32.6.6.
Give this a try:
-make sure the esc are on 32.6.6 from https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/BLHeli_32%20Test%20code%20Rev32.6.4%20hex%20files/Rev32.6.6%20for%20inverted%20bidirectional%20dshot%20and%20GCR%20return%20data%20only
-erase the the flash using bootloader function : " 8 Erase entire external flash" , like in https://github.com/spracing/betaflight/issues/1 .
-Flash BF 4.1 rc3
-enable dshot_bidir

haslinghuis, 2019-09-20:
@emski0 No luck - I'm already aware of ESC GCR 32.6.6 - I did dl flash and 8 erase entire external flash - you never know with - but to no avail. I also did remove the GPS and CAM to further isolate. I'd rather not desolder the VTX vbat+gnd (SmartAudio and Video is not connected anyway - SA goes to CH4 so I can use crossfire.luac on Taranis rather than OSD).

hydra, 2019-09-21:
>... so the error is coming when a camera is plugged in.
> BF 4.1 RC3 and tekko32 F3 ver. 32.6.6( furling32 target)

@emski0 so you confirm that bidir dshot is working for you until you connect the camera?

also do you have a `diff` for others to compare with.

If that's the case then it seems like you probably have some grounding issue or electrical noise issue.

Would be really nice to get a scope trace of what the ESC is outputting and what is being received at the pin on the CPU.  Do you have access to a scope or know anyone with one?

hydra, 2019-09-21:
@maGfpv can you use the `<>` button when pasting in CLI output and avoid using `##` for headers as it makes things hard to read.

emski0, 2019-09-21:
`# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Sep 15 2019 / 09:12:03 (644ab5d90) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# resources
resource MOTOR 4 C07
resource MOTOR 8 NONE
resource PINIO 1 E06

# feature
feature -RSSI_ADC
feature -LED_STRIP
feature -TRANSPONDER
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature ESC_SENSOR

# beeper
beeper -BAT_LOW
beeper -ON_USB

# beacon
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 1 8192 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 1 19200 57600 0 115200
serial 7 2 115200 57600 0 115200

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 46 3 1700 2100 0 0
aux 3 13 3 1250 1650 0 0
aux 4 28 1 1300 2100 0 0
aux 5 35 2 1700 2100 0 0
aux 6 44 1 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745    0
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM     0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860    0
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843    0    0
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# rxfail
rxfail 3 h
rxfail 5 s 1200
rxfail 6 s 1100
rxfail 7 s 1000
rxfail 11 s 1000

# master
set yaw_spin_threshold = 1000
set acc_calibration = 25,80,9
set mag_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 12
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set min_throttle = 1060
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_throttle = 1180
set failsafe_procedure = AUTO-LAND
set vbat_max_cell_voltage = 450
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330
set vbat_scale = 111
set ibata_scale = 305
set beeper_dshot_beacon_tone = 4
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 60
set gps_rescue_throttle_hover = 1229
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = FIXED_ALT
set pid_process_denom = 1
set osd_rssi_alarm = 30
set osd_cap_alarm = 1300
set osd_core_temp_alarm = 70
set osd_tim2 = 769
set osd_vbat_pos = 6584
set osd_rssi_pos = 6529
set osd_link_quality_pos = 353
set osd_tim_2_pos = 6616
set osd_anti_gravity_pos = 1
set osd_g_force_pos = 4289
set osd_throttle_pos = 6552
set osd_vtx_channel_pos = 6604
set osd_current_pos = 6561
set osd_mah_drawn_pos = 6593
set osd_motor_diag_pos = 4162
set osd_gps_speed_pos = 6368
set osd_gps_sats_pos = 2090
set osd_home_dir_pos = 6190
set osd_home_dist_pos = 6191
set osd_altitude_pos = 6391
set osd_pid_roll_pos = 46
set osd_pid_pitch_pos = 14
set osd_debug_pos = 482
set osd_warnings_pos = 14729
set osd_pit_ang_pos = 4312
set osd_disarmed_pos = 6474
set osd_esc_tmp_pos = 6177
set osd_esc_rpm_pos = 4161
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 4
set vtx_channel = 4
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5800
set vcd_video_system = PAL
set camera_control_ref_voltage = 319
set camera_control_key_delay = 110
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 80
set dyn_lpf_dterm_max_hz = 180
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set feedforward_transition = 23
set iterm_relax = OFF
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set p_pitch = 71
set i_pitch = 70
set d_pitch = 37
set f_pitch = 90
set p_roll = 66
set i_roll = 60
set d_roll = 33
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100
set level_limit = 75
set d_min_roll = 25
set d_min_pitch = 27
set d_min_boost_gain = 40
set ff_interpolate_sp = ON
set ff_boost = 20

rateprofile 0

# rateprofile 0
set roll_rc_rate = 49
set pitch_rc_rate = 49
set yaw_rc_rate = 33
set roll_srate = 91
set pitch_srate = 91
set yaw_srate = 91

# end the command batch
batch end
`

@hydra 
> @emski0 so you confirm that bidir dshot is working for you until you connect the camera?
Yeh bidir dshot works while the camera is unplugged. all the motor spins, 0% error on all 4 motors, and no strange behaviour. This is on the bench, no flight test.

> Would be really nice to get a scope trace of what the ESC is outputting and what is being received at the pin on the CPU. Do you have access to a scope or know anyone with one?

Unfortunately i don't have access to a scope.

> If that's the case then it seems like you probably have some grounding issue or electrical noise issue.

Anything i can do about it?

hydra, 2019-09-21:
@maGfpv @emski0 I've ordered one of the HolyBro Tekko32 Metal 4in1ESCs and a T-Motor F55 Pro II so I can check what the signals looks like on the scope with my H7 boards.  Might be a while before it gets here though.

Mateyhv, 2019-09-21:
Hi @hydra and @joelucid I have exactly the same issue with SPH7EXTREME and Mamba 506 4in1. With Bidirectional Dshot ON, ESCs 1-2-3 work fine with telemetry error 0% but ESC4 does not, It beeps on startup but does not send telemetry and does not spin Checked in Blheli suite and Betaflight. Startup beeps are unsynchronized among ESCs, sometimes one or two ESCs keeps doing several reset beeps after the initial one as if they were just coming online. That happens very often but seems very random issue. Turned Bidirectional Dshot OFF and everything is correct, I can spin all motors, all four ESCs beeps sound synchronized and all four motors spin correctly. No random initialization beeps.
Here is the config information sans VTX table and beepers as I don't have a buzzer. I don't have camara connected yet although ready for 5V but the VTX is connected on BAT

Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Sep 15 2019 / 09:12:03 (644ab5d) MSP API: 1.42
reset configuration to default settings

defaults nosave

board_name SPRACINGH7EXTREME
mcu_id 0042001a3437511931353431
signature
feature

feature -RSSI_ADC
feature -LED_STRIP
feature -TRANSPONDER
feature ESC_SENSOR
beacon

beacon RX_SET
serial

serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
aux

aux 0 0 0 1700 2100 0 0
aux 1 26 0 1800 1900 0 0
aux 2 35 0 1700 1800 0 0
aux 3 39 0 1200 1300 0 0
master

set gyro_sync_denom = 2
set mag_hardware = NONE
set fpv_mix_degrees = 25
set serialrx_provider = SBUS
set rx_min_usec = 925
set rx_max_usec = 2075
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set current_meter = ESC
set battery_meter = ESC
set vbat_scale = 255
set vbat_divider = 22
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set debug_mode = GYRO_SCALED
set name = Mark 4

profile 0
profile 0

set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set throttle_boost = 0
resource show all
Currently active IO resource assignments:
(reboot to update)

A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: FREE
A05: FREE
A06: VTX_DATA 10
A07: VTX_CLK 10
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: GYRO_CS 1
B00: VTX_CS 10
B01: VTX_POWER
B02: QSPI_CLK 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: QSPI_BK1CS 1
B11: FREE
B12: GYRO_CS 2
B13: FREE
B14: FREE
B15: SERIAL_RX 1
C00: FREE
C01: FREE
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: FREE
C05: ADC_EXT
C06: FREE
C07: FREE
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: FREE
D00: SERIAL_RX 4
D01: FREE
D02: SDIO_CMD
D03: SPI_SCK 2
D04: GYRO_EXTI
D05: SERIAL_TX 2
D06: SPI_MOSI 3
D07: BEEPER
D08: SERIAL_TX 3
D09: FREE
D10: FREE
D11: QSPI_BK1IO0 1
D12: QSPI_BK1IO1 1
D13: QSPI_BK1IO3 1
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: QSPI_BK1IO2 1
E03: LED 1
E04: SYSTEM
E05: CAMERA_CONTROL
E06: FREE
E07: QSPI_BK2IO0 1
E08: QSPI_BK2IO1 1
E09: QSPI_BK2IO2 1
E10: QSPI_BK2IO3 1
E11: OSD_CS
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: GYRO_EXTI
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE
Currently active Timers:

TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
CH1: MOTOR 1
CH2: MOTOR 2
CH3: MOTOR 3
CH4: MOTOR 4
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15:
CH1: CAMERA_CONTROL
TIM16: FREE
TIM17: FREE
Currently active DMA:

DMA1 Stream 0: MOTOR 1
DMA1 Stream 1: MOTOR 2
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

[image]

[image]

maGfpv, 2019-09-21:
@Hydra i have tmotor f40proiii and tekko f3 metal single escs. Dont know if its important to distinguish between single and 4in1 escs ...

Mateyhv, 2019-09-21:
@hydra I might be able to send you a Mamba 506 4in1. PM me if interested.

monko760, 2019-09-21:
Yea there is something to this camera issue.... I do have everything working but my startup tones TAKE FOREVER and when I unplug camera, the startup tones are like normal.

haslinghuis, 2019-09-21:
With bidirectional dshot enabled one of the ESC's (resource A03) is not recognized in Blheli32.6 configurator. But after connect - read - disconnect the FC will reboot - when you connect for the second time the ESC is displayed with one difference: Input Protocol: Unknown, The 3 other ESC shows Dshot as protocol.
I did resolder the connections, rechecked for solder bridges, VTX and CAM are not connected (only crossfire nano is soldered on uart1) but still same issue. Also flashed RC1, RC2 and RC3 using BIND and 8 blinks. The H7 Extreme replaced a Matek F7 board which had a broken USB connector running iNav.

kill-a-watt, 2019-09-21:
I am also having this exact same problem with the SPH7EXTREME and I'm using Gemfan Maverick 30 amp esc's running Blheli version < 32.6.6. On BF 4.1.0 tried RC 2 & 3. 
Motor 4 stops working when dshot_bidir is enabled. I did try " 8 Erase entire external flash" between flashing new firmware.
I also tried unplugging the vtx and camera, with on change. Motor 4 shows 100% E and dosen't spin 1,2,3 show 0% E and all spin normally.

Mateyhv, 2019-09-21:
At this point I doubt the problem is brand ESC related, seems to be some problem for the H750 understanding the bidir dshot. 

Dominic, my guess is any ESC you have on hand might help you spot the issue.

hydra, 2019-09-21:
@Mateyhv I emailed you, yes the ESC would be helpful.  I don't have *any* BlHeli32 ESCs here to test with, agree that it's probably not brand related.  All my ESCs are BlHeli, BlHeli_S or JESC.  I used to rely on Cleanflight donations to provide me with hardware, but I only get $2 a month from that now - https://www.patreon.com/cleanflight

Jryan819, 2019-09-23:
Hello everyone.

I was relieved to see this thread knowing im not the only one experiencing this issue. I was pulling my hair out tonight. 
Exact same problems as listed above. Ive tried everything also resourcing Motor4 to LED.
Results vary everytime between which motors want to work. RPM errors 20-100%. Along with passthrough to BLheli with error codes "corrupted firmware" , "missing firmware", inability to read ESC's, and continuous start up tones.

4in1 Hobbywing Xrotor 60amp
Im on mobile so no dump. I can post later if requested.

joelucid, 2019-09-23:
Just installed rc3 on an h7 extreme, switched on bidir and scoped the motor traces. Two motors show an artifact after the line is set to input. It looks like the line is driven low just before it's switched to input:

[image]

joelucid, 2019-09-23:
Very weird. Looks like this happens in ``pwmDshotSetDirectionInput`` when ``LL_TIM_IC_Init`` is called. Returning one line above it prevents the artifact. @hydra does this give you something to work with?

hydra, 2019-09-23:
can you try updating the GPIO register to manually set the bit high before the call to `LL_TIM_IC_Init` and retest?

jflyper, 2019-09-23:
Hmm... it looks as if a large capacitance suddenly appeared when switching to input ...

Mateyhv, 2019-09-23:
@hydra, Mamba 4in1 blheli32 esc on the way. Should show up in 2-4 days.

joelucid, 2019-09-23:
> Hmm... it looks as if a large capacitance suddenly appeared when switching to input ...

I think this is just the normal capacitance of the dshot line. The issue is that the line was actively driven low when switching to input, but is then pulled up again by the pull-ups. Same capacitance generates very different rise/fall times when driven actively vs pulled up.

joelucid, 2019-09-24:
I'm not entirely sure why - but this branch fixes the artifacts: https://github.com/joelucid/betaflight/tree/fix_bidir_h7. I just switch the line to output_pp while initializing the timer input channel, suppressing the transient artifact. Could someone test if this makes bidir actually work?

I've tried a long time to make this faster using the LL driver instead of a complete reconfig of the pins, but couldn't get that to work.

TheIsotopes, 2019-09-24:
@joelucid tested your branch on latest master 064ca9b with a H7 Extreme
all motors now spinning ... but still to many errors

[image]

[image]

TheIsotopes, 2019-09-24:
@joelucid Tested also on a H7 NANO with JESC ... everything works normally as before

[image]

maGfpv, 2019-09-26:
Tested rc5 with my h7extreme and tekko32 f3 metall esc (32.66). Still having the same issue as mentioned above. 

After activating bidirectional dshot motor 1 starts to arm and spin when lipo is plugged in.

TheIsotopes, 2019-09-26:
@hydra @joelucid I tested your branch again with the current master of rc5
This time I reconfigured from scratch and had to find that the led feature has a disturbing effect.
I have attached again two screenshots. first screenshot is without led feature and on the second screenshot is the led feature enabled. you can see that the led feature causes an increased error rate

[image]

[image]

haslinghuis, 2019-09-26:
I like to test this on my board for comparison. After following the wiki I have setup a fedora machine, cloned JoeLucid's branche and I am able to build a hex file for the target SPRACINGF7DUAL. But when I try:  <make SPRACINGH7EXTREME> it complains that there is no rule to create target <> ? So, how to create this target?

hydra, 2019-09-26:
@haslinghuis 1. make sure you're on the right branch.  When you clone a repo you will be looking at the default branch which may not be the one you want, use `git branch` to show the current branch, use `git checkout ...` to checkout the branch you need. for more help git and building BF please refer to the git manual and tutorials and the BF developer docs.  Let's try and keep this thread on-topic rather than tech-support for building a target. 2. try `make TARGET=SPRACINGH7EXTREME`. 

@TheIsotopes that LED strip issues is kinda weird.  Can you show us a photo of your quad please?

[55 more replies]

PR #8961 (2019-09-28): H7 - Fix bi-directional DSHOT
Fix for DSHOT signal artifacts on H7 when switching between output and input mode.

For artifact scope trace see https://github.com/betaflight/betaflight/issues/8836#issuecomment-534027871

Confirmed fixed by at least one user - https://github.com/betaflight/betaflight/issues/8836#issuecomment-535538177

Fixes #8836
closed 2019-09-28 by mikeller