betaflight/betaflight #8838: Large amount of packet errors in bi-directional telemetry when using Bit-banging Dshot600..

ecob81, 2019-09-06:
With dshot_bitbang=auto and motor_pwm_protocol = DSHOT600 i'm getting a large amount of packet errors in rpm data and esc telemetry stops working (no vbat reading in gui or on OSD).

Expecting to get little or no packet errors during dshot_telemetry_info test and to see vbat telemetry readings in gui/on OSD.

Using Tmotor F55A V1 and Kakute F7 with V32.6.6 and 4.1.0 firmwares.

Works ok if I set dshot to 300 or turn off bitbang.

```# version
# Betaflight / KAKUTEF7 (KTF7) 4.1.0 Sep  5 2019 / 13:14:54 (1a56a51ef) MSP API: 1.42

# start the command batch
batch start

board_name KAKUTEF7

# name: XL8

# feature
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -ARMED
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 16384 115200 57600 0 115200
serial 2 2 115200 57600 0 115200

# aux
aux 0 0 0 1400 2100 0 0
aux 1 1 0 1400 1600 0 0
aux 2 27 1 1900 2100 0 0
aux 3 46 1 1650 1850 0 0
aux 4 13 0 1150 1350 0 0
aux 5 13 2 1925 2100 1 0
aux 6 13 1 1925 2100 1 0
aux 7 26 1 1400 2100 0 0
aux 8 35 0 1650 1850 0 0

# adjrange
adjrange 0 0 2 900 2100 29 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 1W

# rxfail
rxfail 4 s 1500

# master
set mag_hardware = NONE
set rssi_channel = 8
set rc_smoothing_auto_smoothness = 50
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_idle_value = 500
set dshot_bidir = ON
set dshot_bitbang = AUTO
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set battery_meter = ESC
set ibata_scale = 90
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 38
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1800
set gps_rescue_throttle_min = 1150
set gps_rescue_throttle_hover = 1275
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warn_batt_warning = OFF
set osd_cap_alarm = 4000
set osd_alt_alarm = 9999
set osd_vbat_pos = 14761
set osd_rssi_pos = 14778
set osd_link_quality_pos = 440
set osd_rssi_dbm_pos = 216
set osd_tim_1_pos = 471
set osd_tim_2_pos = 14785
set osd_flymode_pos = 14370
set osd_throttle_pos = 14753
set osd_vtx_channel_pos = 8461
set osd_ah_sbar_pos = 4334
set osd_ah_pos = 4206
set osd_current_pos = 14799
set osd_mah_drawn_pos = 14769
set osd_gps_speed_pos = 14424
set osd_gps_lon_pos = 14352
set osd_gps_lat_pos = 14337
set osd_gps_sats_pos = 14401
set osd_home_dir_pos = 14382
set osd_home_dist_pos = 14383
set osd_flight_dist_pos = 14375
set osd_compass_bar_pos = 42
set osd_altitude_pos = 14390
set osd_pid_roll_pos = 8353
set osd_pid_pitch_pos = 8321
set osd_pid_yaw_pos = 8385
set osd_debug_pos = 4481
set osd_power_pos = 416
set osd_pidrate_profile_pos = 8481
set osd_avg_cell_voltage_pos = 14793
set osd_disarmed_pos = 298
set osd_nheading_pos = 45
set osd_esc_rpm_pos = 8344
set osd_rtc_date_time_pos = 33
set osd_flip_arrow_pos = 14369
set osd_core_temp_pos = 8454
set osd_log_status_pos = 8449
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_flight_dist = ON
set osd_stat_max_fft = ON
set osd_profile = 3
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5732
set name = XL8

profile 0

# profile 0
set vbat_pid_gain = ON
set anti_gravity_gain = 8000
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set p_pitch = 55
set d_pitch = 50
set f_pitch = 150
set p_roll = 50
set d_roll = 45
set f_roll = 125
set p_yaw = 45
set f_yaw = 125
set d_min_roll = 30
set d_min_pitch = 35
set d_min_boost_gain = 30
set d_min_advance = 0

rateprofile 0

# rateprofile 0
set thr_mid = 0
set thr_expo = 30
set yaw_rc_rate = 80
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set tpa_rate = 80
set tpa_breakpoint = 1450
set throttle_limit_type = SCALE
set throttle_limit_percent = 80```

```# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: USB_DETECT
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: FREE
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: SERIAL_RX 7
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: DSHOT_BITBANG 1
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE```

Kakute F7
Tmotor F55A V1
TBS CRSF Nano Diversity
TBS Unify pro HV SE
BN-220 GPS
Runcam swift with Uart Cam control hooked up

evilmerlin, 2019-09-06:
Having the exact same issue with both AikonF4 (last known working Nightly #1837 and EXF722Dual (Last known working 4.0.6)

joelucid, 2019-09-06:
This is interesting - just tested. I'm seeing two motors report perfect rpm (motors 1 & 2) and motors 3 & 4 have issues:

```
# dshot_telemetry_info
Dshot reads: 1509635
Dshot invalid pkts: 2880
Dshot directionChange cycles: 0, micros: 0

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0     0.00%
    1         0        0       0     0.00%
    2         0        0       0     0.46%
    3         0        0       0     0.23%

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

Motors 1 & 2 are on port B, motors 3 & 4 on port A. I suspect this issue is due to dma latency jitter: telemetry bits will arrive at about the same time, so the first dma transfer is likely in progress when the second one hits, leading to varying times between samples.

@jflyper any ideas?

ecob81, 2019-09-06:
Here's mine if it's of any help

```
# dshot_telemetry_info
Dshot reads: 2131060
Dshot invalid pkts: 123339
Dshot directionChange cycles: 0, micros: 0

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0     0.00%
    1         0        0       0    20.10%
    2         0        0       0     2.00%
    3         0        0       0     0.97%

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
```

joelucid, 2019-09-06:
Just a decode problem. I have it fixed and verified on f4. @ecob81 here is a kakutef7 hex - could you try that out?

[betaflight_4.1.0_KAKUTEF7.hex.zip](https://github.com/betaflight/betaflight/files/3584773/betaflight_4.1.0_KAKUTEF7.hex.zip)

joelucid, 2019-09-06:
@evilmerlin here is AikonF4:
[betaflight_4.1.0_AIKONF4.hex.zip](https://github.com/betaflight/betaflight/files/3584779/betaflight_4.1.0_AIKONF4.hex.zip)

joelucid, 2019-09-06:
BTW these have debug in so you will likely have to run at 4k

joelucid, 2019-09-06:
These are even better:

[betaflight_4.1.0_KAKUTEF7.hex 2.zip](https://github.com/betaflight/betaflight/files/3584818/betaflight_4.1.0_KAKUTEF7.hex.2.zip)
[betaflight_4.1.0_AIKONF4.hex 2.zip](https://github.com/betaflight/betaflight/files/3584819/betaflight_4.1.0_AIKONF4.hex.2.zip)

joelucid, 2019-09-06:
Please try these at 4k4k and if you still have problems paste the output of dshot_telemetry_info.

ecob81, 2019-09-06:
Yep dshot telemetry now ok, zero packet errors, still haven't got any vbat telemetry from the esc though if that matters.

joelucid, 2019-09-06:
Thanks! Yeah there’s a separate issue for telemetry. We already have a fix waiting to be merged.

evilmerlin, 2019-09-07:
Tested the hex 2 on the bench. zero telemetry errors but also no ESC telemetry data. Will be doing a test flight later but I don't think it will have any issues. 
Is the telemetry fix in one of the nightlies already?

mikeller, 2019-09-07:
@joelucid: What is the fix for this?

joelucid, 2019-09-07:
@mikeller - https://github.com/betaflight/betaflight/pull/8843

closed 2019-09-08 by mikeller

evilmerlin, 2019-09-09:
Just a quick update.
FC: AikonF4 on Nightly #1867
With Bitbang_Dshot = auto
In the configurator, while spooling up the motors, the error amount would increase to about 0.3% at 50% to 60% throttle.
Set bitbang_dshot = off and applied the DMA and Timer configs in the snippet and there are 0% errors.

ecob81, 2019-09-09:
Interesting, was this using the hex above posted above by joelucid or one from the configurator? I didn't try it at high throttle only tested at about 10%.

evilmerlin, 2019-09-10:
This was with the Nightly build 1867.

ecob81, 2019-09-10:
Sorry i'm a bit new to all this. Do you know if that build had the fix merged into it or not? Also was it at dshot600 or 300? Just trying to figure out if I need to go back to 300 or not. I ran 600 in the Release candidate and it tested ok at low throttle so I flew it and it seemed ok apart from a few vibes that i'm currently trying to tune out.