betaflight/betaflight #8840: Strange behaviour introduced by activating RPM filtering on latest versions

SupaflyFPV, 2019-09-06:
Hi, I will attempt to contribute this issue I hope it is useful/relevant...I am not currently sure how related to my hardware it might be.

I have been currently using earlier 4.1 builds with GCR RPM filtering no probs.

I tried to flash the most recent versions (development builds and RC1) and after flashing things seemed ok-I was able to connect back into BF configurator ok and behaviour seemed ok.

Then when I pasted in my setting to CLI and saved, I no longer was able to connect back into BF configurator. Also - my mouse activity became patchy and jumpy. I tried another of the same FC and another computer and it was the same. 

I went thru my diff to see what might be causing it and whittled it down to the set bidirectional dshot and dshot burst command - prior to putting that command in the behaviour was normal. I ran out of time to see which of those were the problem. I tend to change them at the same time as per rpm setup guide...so Im not totally clear on what dshot burst is.

My FC is Aikon F4, configurator was 1.50.1 on macs. I failed to track down the 10.6 configurator.

I went back to a mid august version of 4.1 and it worked fine with my previous diff and settings.

SupaflyFPV, 2019-09-06:
...ok I update to 10.6 config and same thing...everything normal until I set bidirect dshot = on - then I cannot connect to the FC and have to reflash via boot button.

joelucid, 2019-09-06:
With 4.1 rc you get dshot bitbang by default which is more cpu intensive than the old timer based implementation. If you’ve been running 8k on overclocked f4 that might no longer work. If you set dshot_bitbang to off you will get the timer based implementation which should match your previous experience.

SupaflyFPV, 2019-09-06:
Hi Joe....I did try it at 4/4 dshot300 and the same thing happened - but it may be related to my settings...I cant work it out atm...but in some circumstances when I switch on the RPM filtering with the bidir ON command then it goes wrong...

I can replicate by flashing the RC1 then putting this into CLI (basically my tune/filter settings...and I amended with dshot300 to take less CPU):

feature -DYNAMIC_FILTER
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 3 2 115200 57600 0 115200
serial 30 8192 115200 57600 0 115200

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_lowpass_hz = 150
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 0
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 300
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set vbat_warning_cell_voltage = 330
set ibata_scale = 210
set small_angle = 180
set deadband = 4
set yaw_deadband = 4
set osd_vbat_pos = 2545
set osd_rssi_pos = 6648
set osd_tim_1_pos = 6625
set osd_mah_drawn_pos = 2487
set osd_debug_pos = 321
set osd_pidrate_profile_pos = 2465
set osd_warnings_pos = 10666
set osd_avg_cell_voltage_pos = 2537
set debug_mode = GYRO_SCALED
set vcd_video_system = PAL
set gyro_1_sensor_align = CW0FLIP
set gyro_1_align_pitch = 1800

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 120
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 90
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 10000
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set p_pitch = 70
set d_pitch = 40
set f_pitch = 100
set p_roll = 60
set i_roll = 80
set d_roll = 40
set f_roll = 100
set p_yaw = 50
set i_yaw = 100
set f_yaw = 100
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 30
set d_min_advance = 40
set auto_profile_cell_count = 4
set ff_interpolate_sp = ON
set ff_max_rate_limit = 100
set ff_boost = 20

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 120
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 80
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 10000
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_cutoff = 11
set p_pitch = 70
set d_pitch = 50
set f_pitch = 100
set p_roll = 60
set i_roll = 80
set d_roll = 50
set f_roll = 100
set p_yaw = 50
set f_yaw = 100
set d_min_roll = 36
set d_min_pitch = 36
set d_min_boost_gain = 30
set motor_output_limit = 80
set auto_profile_cell_count = 5
set ff_interpolate_sp = ON
set ff_max_rate_limit = 100
set ff_boost = 20

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 120
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 80
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 10000
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_cutoff = 11
set p_pitch = 70
set d_pitch = 45
set f_pitch = 100
set p_roll = 60
set i_roll = 80
set d_roll = 45
set f_roll = 100
set p_yaw = 50
set f_yaw = 100
set d_min_roll = 34
set d_min_pitch = 36
set d_min_boost_gain = 40
set motor_output_limit = 66
set auto_profile_cell_count = 6
set ff_interpolate_sp = ON
set ff_max_rate_limit = 100
set ff_boost = 20

SupaflyFPV, 2019-09-06:
....really not sure what has been going on but I went through it bit by bit and this following does not cause the issue:

# feature
feature -DYNAMIC_FILTER
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 3 2 115200 57600 0 115200
serial 30 8192 115200 57600 0 115200

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_sync_denom = 2
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 0
set baro_hardware = NONE
set dshot_idle_value = 520
set motor_pwm_protocol = DSHOT300
set vbat_warning_cell_voltage = 330
set ibata_scale = 210
set small_angle = 180
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 1
set osd_vbat_pos = 2545
set osd_rssi_pos = 6648
set osd_tim_1_pos = 6625
set osd_mah_drawn_pos = 2487
set osd_debug_pos = 321
set osd_pidrate_profile_pos = 2465
set osd_warnings_pos = 10666
set osd_avg_cell_voltage_pos = 2537
set debug_mode = GYRO_SCALED
set vcd_video_system = PAL
set gyro_1_sensor_align = CW0FLIP
set gyro_1_align_pitch = 1800

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 120
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 90
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 10000
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set p_pitch = 70
set d_pitch = 45
set f_pitch = 100
set p_roll = 60
set i_roll = 80
set d_roll = 45
set f_roll = 100
set p_yaw = 50
set i_yaw = 100
set f_yaw = 100
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 30
set d_min_advance = 40
set auto_profile_cell_count = 4
set ff_interpolate_sp = ON
set ff_boost = 20

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 120
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 80
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 10000
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_cutoff = 11
set p_pitch = 70
set d_pitch = 50
set f_pitch = 100
set p_roll = 60
set i_roll = 80
set d_roll = 50
set f_roll = 100
set p_yaw = 50
set f_yaw = 100
set d_min_roll = 36
set d_min_pitch = 36
set d_min_boost_gain = 30
set motor_output_limit = 80
set auto_profile_cell_count = 5
set ff_interpolate_sp = ON
set ff_boost = 20

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 120
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 80
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 10000
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_cutoff = 11
set p_pitch = 70
set d_pitch = 45
set f_pitch = 100
set p_roll = 60
set i_roll = 80
set d_roll = 45
set f_roll = 100
set p_yaw = 50
set f_yaw = 100
set d_min_roll = 34
set d_min_pitch = 36
set d_min_boost_gain = 40
set motor_output_limit = 66
set auto_profile_cell_count = 6
set ff_interpolate_sp = ON
set ff_boost = 20

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 220
set pitch_rc_rate = 220
set yaw_rc_rate = 212
set roll_expo = 71
set pitch_expo = 71
set yaw_expo = 71
set roll_srate = 1
set pitch_srate = 1
set yaw_srate = 1
set tpa_rate = 40
set tpa_mode = PD

SupaflyFPV, 2019-09-07:
Ive been trying to isolate the command which causes the problem and it would appear its 

set dshot_bidir = ON

because the second diff I posted works fine - until you set dshot_bidir = ON...

However - if I do another re-flash and simply set dshot_bidir = ON - it wont stay on...I go to CLI type diff and there is no record of dshot_bidir = ON...so it doesnt switch on...

...so tbh Im at a loss whats causing the problem here...

mikeller, 2019-09-07:
@SupaflyFPV: Can you please post complete diffs, including the header line at the top? Without knowing the exact version of the firmware that you are using, this is all just guesswork.

SupaflyFPV, 2019-09-07:
I can do tomorrow but this has all been with 4.1 RC1....it was the same on development builds I dont know up to what date, but I flash back to 4.1 development on august 15th and Im back to normal...

mikeller, 2019-09-07:
Ok, it's softserial 1 that is causing the problem:

Resource allocation wit `dshot_bidir = off`:

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A09: SERIAL_TX 11
A10: SERIAL_RX 1
A11: USB
A12: USB
A15: OSD_CS
B00: FREE
B01: FREE
B03: CAMERA_CONTROL
B04: LED 1
B05: BEEPER
B06: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: USB_DETECT

Currently active Timers:
-----------------------
TIM1:
    CH2: SERIAL_TX 11
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

Resource allocation with `dshot_bidir = on`, but `serial 31 0 115200 57600 0 115200`:

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A15: OSD_CS
B00: FREE
B01: FREE
B03: CAMERA_CONTROL
B04: LED 1
B05: BEEPER
B06: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: USB_DETECT

Currently active Timers:
-----------------------
TIM1:
    CH1: DSHOT_BITBANG 1
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: DSHOT_BITBANG 1
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

@joelucid: Looks like the dynamic timer allocation does not quite work in this case.

SupaflyFPV, 2019-09-07:
thanks for looking into this

mikeller, 2019-09-08:
@ripperdrone: I can reproduce the problem (no Dshot output), and the resource allocation (`resource show all`, like asked for when opening an issue) looks as it should:

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: CAMERA_CONTROL
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

@jflyper: I disagree, this is a situation that should work just fine with bitbanged Dshot, as timer 8 is chosen since timer 1 is detected as already allocated. Not sure what is going wrong here.

@joelucid: Do you have an idea why the above resource allocation would result in no Dshot output?

```
# diff

# version
# Betaflight / STM32F411 (S411) 4.1.0 Sep  8 2019 / 22:03:58 (f331cb3dd) MSP API: 1.42

board_name MATEKF405MINI
manufacturer_id MTKS

# resources
resource LED_STRIP 1 NONE
resource CAMERA_CONTROL 1 A08

# master
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
```

jflyper, 2019-09-08:
> @jflyper: I disagree, this is a situation that should work just fine with bitbanged Dshot, as timer 8 is chosen since timer 1 is detected as already allocated. Not sure what is going wrong here.

@mikeller Are we looking at F411 here, too? It doesn't have TIM8.

mikeller, 2019-09-08:
@jflyper: Eh dammit, you nailed it. According to Betaflight F411 has TIM8 (see above resource listing). :confounded:

joelucid, 2019-09-08:
See https://github.com/betaflight/betaflight/pull/8853 for a hopefully simple solution to these issues.

joelucid, 2019-09-08:
@SupaflyFPV, could you try this hex? 
[betaflight_4.1.0_AIKONF4.hex 3.zip](https://github.com/betaflight/betaflight/files/3587885/betaflight_4.1.0_AIKONF4.hex.3.zip)

SupaflyFPV, 2019-09-08:
@joelucid ok flashed it - seems like it works, no crash or weird behaviour. 

Only thing, I checked and bidir is ON but when I type 'status' - I dont get the column of RPM reports...this is my usual way to check RPM filtering is active...

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       2       1    0.5%    0.5%         0
01 - (         SYSTEM)    997       5       0    0.9%    0.0%         6
02 - (            PID)   3998      77      41   31.2%   16.8%      5422
   - (           GYRO)   7996
03 - (            ACC)    997      14       8    1.8%    1.2%       132
04 - (       ATTITUDE)    499      12       7    1.0%    0.8%        63
05 - (             RX)     31      57      36    0.6%    0.6%        20
06 - (         SERIAL)     99  470764       3 4661.0%    0.5%       892
07 - (       DISPATCH)    997       2       0    0.6%    0.0%         9
08 - (BATTERY_VOLTAGE)     49       6       1    0.5%    0.5%         1
09 - (BATTERY_CURRENT)     49       2       1    0.5%    0.5%         0
10 - ( BATTERY_ALERTS)      4       3       1    0.5%    0.5%         0
11 - (         BEEPER)     99       5       2    0.5%    0.5%         4
16 - (      TELEMETRY)    249      18       0    0.9%    0.0%        35
18 - (            OSD)     59     623      35    4.1%    0.7%        33
20 - (            CMS)     59       3       1    0.5%    0.5%         0
21 - (        VTXCTRL)      4      19       6    0.5%    0.5%         0
22 - (        CAMCTRL)      4       1       1    0.5%    0.5%         0
24 - (    ADCINTERNAL)      2       3       1    0.5%    0.5%         0
25 - (       PINIOBOX)     18       2       1    0.5%    0.5%         0
RX Check Function                   8       2                         0
Total (excluding SERIAL)                        46.1%   25.1%

# status
MCU F40X Clock=192MHz (PLLP-HSE), Vref=3.29V, Core temp=34degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3595, Max available config: 16384
GYRO=ICM20602, ACC=ICM20602
System Uptime: 23 seconds, Current Time: 2019-09-08T19:33:15.133+00:00
CPU:9%, cycle time: 123, GYRO rate: 8130, RX rate: 33, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP

jflyper, 2019-09-09:
@SupaflyFPV How about `dshot_telemetry_info`?

SupaflyFPV, 2019-09-09:
# dshot_telemetry_info
Dshot reads: 0
Dshot invalid pkts: 0
Dshot directionChange cycles: 0, micros: 0

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0   NO DATA
    1         0        0       0   NO DATA
    2         0        0       0   NO DATA
    3         0        0       0   NO DATA

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

SupaflyFPV, 2019-09-09:
diff all in case I did something stupid

# version
# Betaflight / AIKONF4 (AIK4) 4.1.0 Sep  8 2019 / 20:23:59 (81f80636f) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name AIKONF4
mcu_id 004f0038374d500320313048
signature 

# name: Juicemode

# resources
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 C07
resource MOTOR 4 C06

# feature
feature -DYNAMIC_FILTER
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 3 2 115200 57600 0 115200
serial 30 8192 115200 57600 0 115200

# aux
aux 0 0 1 1650 2100 0 0
aux 1 1 0 1300 2100 0 0
aux 2 13 2 1300 1700 0 0
aux 3 26 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_lowpass_type = BIQUAD
set gyro_lowpass_hz = 0
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 0
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 300
set baro_hardware = NONE
set rssi_channel = 16
set dshot_idle_value = 500
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set vbat_max_cell_voltage = 440
set vbat_warning_cell_voltage = 330
set ibata_scale = 210
set small_angle = 180
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 20
set gps_rescue_throttle_min = 1220
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set pid_process_denom = 2
set osd_vbat_pos = 2545
set osd_rssi_pos = 6648
set osd_tim_1_pos = 6625
set osd_mah_drawn_pos = 2487
set osd_debug_pos = 321
set osd_pidrate_profile_pos = 2465
set osd_warnings_pos = 10666
set osd_avg_cell_voltage_pos = 2537
set debug_mode = GYRO_SCALED
set cpu_overclock = 192MHZ
set vtx_band = 3
set vtx_channel = 8
set vtx_power = 4
set vtx_freq = 5945
set vcd_video_system = PAL
set gyro_1_sensor_align = CW0FLIP
set gyro_1_align_pitch = 1800
set name = Juicemode

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 120
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 100
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 10000
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set p_pitch = 70
set d_pitch = 40
set f_pitch = 50
set p_roll = 60
set i_roll = 80
set d_roll = 40
set f_roll = 50
set p_yaw = 50
set i_yaw = 100
set f_yaw = 50
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 30
set d_min_advance = 40
set motor_output_limit = 98
set auto_profile_cell_count = 4
set ff_interpolate_sp = ON
set ff_boost = 20

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 120
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 80
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 10000
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_cutoff = 11
set p_pitch = 70
set i_pitch = 80
set d_pitch = 40
set f_pitch = 50
set p_roll = 60
set i_roll = 70
set d_roll = 40
set f_roll = 50
set p_yaw = 55
set i_yaw = 100
set f_yaw = 50
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 30
set motor_output_limit = 80
set auto_profile_cell_count = 5
set ff_interpolate_sp = ON
set ff_boost = 20

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 120
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 80
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 10000
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_cutoff = 11
set p_pitch = 70
set d_pitch = 40
set f_pitch = 50
set p_roll = 60
set i_roll = 80
set d_roll = 40
set f_roll = 50
set p_yaw = 50
set i_yaw = 100
set f_yaw = 50
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 40
set motor_output_limit = 60
set auto_profile_cell_count = 6
set ff_interpolate_sp = ON
set ff_boost = 20

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 220
set pitch_rc_rate = 220
set yaw_rc_rate = 212
set roll_expo = 71
set pitch_expo = 71
set yaw_expo = 71
set roll_srate = 1
set pitch_srate = 1
set yaw_srate = 1
set tpa_rate = 40
set tpa_breakpoint = 1500
set tpa_mode = PD

rateprofile 1

# rateprofile 1
set roll_rc_rate = 220
set pitch_rc_rate = 220
set yaw_rc_rate = 212
set roll_expo = 71
set pitch_expo = 71
set yaw_expo = 71
set roll_srate = 1
set pitch_srate = 1
set yaw_srate = 1
set tpa_rate = 40
set tpa_mode = PD
set throttle_limit_type = CLIP
set throttle_limit_percent = 99

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

MN-237, 2019-09-15:
I am having the same issue on Kakute F7!
BF 4.1.0-RC2
Jenkins BetaFlight Configurator 10.6.0 #3​06 12.09.2019 23:28
After setting and saving dshot_bidir = on, the board isn't recognized properly anymore by windows.  STM drivers are working so it isn't a driver related problem. Board was working properly until dhsot_bidir was set on.

frescof, 2019-09-15:
Same problem here. Using Mamba 722 mini 10.6 and RC2. I'm pretty sure RC1 wasn't having this issue. If more testing is needed let me know.

IvoFPV, 2019-09-15:
@frescof Could you try the latest development build 1879 and report back if the issue still persists.

frescof, 2019-09-15:
@IvoFPV #1879 seems fine on MAMBA722.

```# diff all

# version
# Betaflight / MAMBAF722 (MBF7) 4.1.0 Sep 15 2019 / 08:57:36 (644ab5d90) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722
mcu_id 0031004d3237511734363730
signature 

# name: Lessie

# resources
resource MOTOR 1 A08
resource MOTOR 2 A09
resource MOTOR 3 C08
resource MOTOR 4 C09

# feature
feature GPS

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 4 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1300 1625 0 0
aux 2 2 1 1675 2075 0 0
aux 3 13 2 1700 2100 0 0
aux 4 35 3 1700 2100 0 0

# adjrange
adjrange 0 0 4 900 1250 12 4 0 0
adjrange 1 0 4 1250 1675 12 4 0 0
adjrange 2 0 4 1700 2100 12 4 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 30
vtxtable powerlabels 25 100 400 1W

# master
set gyro_sync_denom = 2
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set serialrx_provider = CRSF
set dshot_idle_value = 350
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 360
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_angle = 25
set gps_rescue_ground_speed = 1500
set gps_rescue_min_sats = 5
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5917
set vcd_video_system = PAL
set name = Lessie

profile 0

# profile 0
set f_pitch = 114
set f_roll = 108
set f_yaw = 96

# dshot_telemetry_info
Dshot reads: 820356
Dshot invalid pkts: 0
Dshot directionChange cycles: 0, micros: 0

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0     0.00%
    1         0        0       0     0.00%
    2         0        0       0     0.00%
    3         0        0       0     0.00%

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

KAJE1977, 2019-09-15:
Release candidate 2 solved the issue for me

RipperDrone, 2019-09-15:
looks good now

closed 2019-09-23 by joelucid