betaflight/betaflight #8844: Mamba F722 Motor Outputs 3&4 not working with DSHOT1200

SteveCEvans, 2019-09-07:
This works OK with 4.0.6 but is broken in release candidate.

Betaflight / MAMBAF722 (MBF7) 4.1.0 Sep  5 2019 / 20:13:10 (62f0eb7ed) MSP API: 1.42

Outputs 1&2 appear OK.

I've tried applying output of dump hardware from 4.0.6 to the STM32F7x2 target and that's the same.

SteveCEvans, 2019-09-07:
@mikeller What do you need to know? Default config but with DSHOT1200 enabled, go to the motors tab and only motors 1&2 work. Do the same with 4.0.6 and it works. Not sure who maintains the DSHOT code. I will take a look myself, but can’t this weekend and so thought it best to flag that there’s and issue.

mikeller, 2019-09-07:
@SteveCEvans: Please provide the information you were asked to provide when opening this pull request.

SteveCEvans, 2019-09-07:
@mikeller It looks like it could be a mismatch between configuarator and FW as the configurator is attempting to set DSHOT1200, but the CLI doesn't acknowledge its existence.

**Describe the bug**
Mamba F722 Motor Outputs 3&4 not working with DSHOT1200. It works with the default of OneShot125, and with DSHOT150-600.

**To Reproduce**
Enable DSHOT1200 and use the Motors tab in the configurator to start the motors. Motors 1 and 2 run, 3 and 4 do not.

**Expected behavior**
All motors should work.

**Flight controller configuration**

Configurator 10.6.0 shows:

[image]

However the CLI shows:

```# diff

# version
# Betaflight / MAMBAF722 (MBF7) 4.1.0 Sep  5 2019 / 20:13:10 (62f0eb7ed) MSP API: 1.42

# start the command batch
batch start

board_name MAMBAF722

# master
set motor_pwm_protocol = PROSHOT1000

profile 0

rateprofile 0

# end the command batch
batch end```

Resources shown below as requested.

```# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: CAMERA_CONTROL
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 1
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: MOTOR 4
DMA2 Stream 7: MOTOR 2```

**Setup / Versions**
Mamba F722 Super Stack with Mamba 506 Stack

I've have a Crossfire Nano RX and TBS Unify Pro32 HV VTX connected, but they don't need to be configured to illustrate this issue.

**Additional context**
Add any other context about the problem here.

emski0, 2019-09-07:
@SteveCEvans Dshot1200 was removed, i think you are using older configurator. In the latest version of Configurator 10.6.0 Dshot1200 is not present in ESC/Motor protocol choice

SteveCEvans, 2019-09-07:
Ah, that would explain it. What protocol should I be using with BLHeli32 capable ESCs if I want to use RPM based filtering?

emski0, 2019-09-07:
I guess Dshot600

closed 2019-09-07 by SteveCEvans

mikeller, 2019-09-07:
@SteveCEvans: Yes, since support for Dshot1200 was removed in the firmware and configurator, using the configurator UI with an older unstable version of the firmware will not work any longer, and result in unexpected behaviour.

Use Dshot600 or Dshot300 for bidirectional Dshot - as opposed to Dshot1200 which never worked reliably these actually work.

SteveCEvans, 2019-09-08:
@mikeller having the error counts on the configurator's motors tab shows that DSHOT300 is the only viable option, even with an F7. I'd completely missed the fact that DSHOT1200 was being dropped. Amusing that there was such a fuss over faster and faster DSHOT which now appears to have been pure specmanship!

mikeller, 2019-09-08:
@SteveCEvans: #8843 should fix the issues with Dshot600 and bitbanged Dshot.

Re why we ever had Dshot1200:

Dshot600 was specified to support up to 32 kHz motor updates on paper, because back when we specified it the 32 kHz hype was still on.
But what we didn't take into account is that with current ESC firmwares, the ESC needs time to process the Dshot command after it has been received before it can start receiving the next Dshot command. And with Dshot600 at 32 kHz the dead time between the end of a frame and the start of the next one is not long enough for the ESC firmware to process the update, causing missed updates.
To address this Dshot1200 was bolted onto the existing Dshot code.
But the reality is that Dshot1200 is pushing the limits of the current timer based implementation of Dshot in Betaflight, and pushing the limits of timing resolution on BLHeli_32, so it always has a significantly higher flight controller => ESC error rate than Dshot600.
Now with 32 kHz loop frequency support gone, there is no incentive to put effort into addressing these issues, and we'd rather spend our time on change that actually improves the flight performance, and doesn't just feed the 'bigger numbers are better' fallacy.