betaflight/betaflight #8845: 4.1 RC1 fried furious racepit after setting dshot_bidir = on

fpvdoc, 2019-09-07:
Flashed 4.1 RC1 with configurator 10.6 incl. Sliders. After setting and saving dshot_bidir = on, the board isn't recognized properly anymore by windows. A friend of mine with the same board has exactly the same issue . Other board and another racepit with nightly 4.1 with rpm enabled are being recognized properly. STM drivers are working so it isn't a driver related problem. Board was working properly until dhsot_bidir was set on.

IvoFPV, 2019-09-07:
#8840 This should be the same issue it seems. Can you provide a `diff`?

fpvdoc, 2019-09-07:
I can't provide a diff from the board because I can't connect anymore but I can tell you what I changed before enabling dshot_bidir.

These are the changes

 beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 1 16384 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 4 32 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 2 1700 2100 0 0
aux 2 36 1 1700 2100 0 0
aux 3 40 0 1300 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

set dshot_bidir = ON
set failsafe_switch_mode = STAGE2
set vbat_max_cell_voltage = 420
set vbat_warning_cell_voltage = 340
set ibata_scale = 177
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON

set pid_process_denom = 1
set transient_throttle_limit = 0
set osd_mah_drawn_pos = 2098
set osd_debug_pos = 449
set osd_warnings_pos = 14793
set osd_avg_cell_voltage_pos = 2086
set osd_core_temp_pos = 46
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
set vtx_freq = 5806
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 900
set name = Amax R5e

set anti_gravity_gain = 10000
set iterm_relax_cutoff = 35
set pidsum_limit = 1000
set d_min_roll = 23
set d_min_pitch = 23
set d_min_boost_gain = 45
set d_min_advance = 0
set idle_min_rpm = 14

[image]

fpvdoc, 2019-09-07:
#8840 is the same issue from my point of view. Just on a different board so it seems to be not related to the racepit only.

mikeller, 2019-09-07:
Duplicate of #8840.

closed 2019-09-07 by mikeller