betaflight/betaflight #8846: Undefined vtxtable with SmartAudio and PitMode prevents arming

theely, 2019-09-07:
After flashing BF 4.1 RC1 on Matek F22Mini the quad would "freze" if turned.
After selectively disabling features I found that SmarAudio was causing the issue.
My hypothesis is that if PitMode via SmartAudio is enabled but the vtxtable is undefined (only figured out later you have to manually set it up) by booting up the FC, BF crashes.

FC: Matek F22Mini
VTX: TBS Unify Pro v3

board_name MATEKF722SE

# name: Smoothy

# resources
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 B05
resource MOTOR 4 B04

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200
serial 3 16384 115200 57600 0 115200

# aux
aux 0 0 1 1500 2100 0 0
aux 1 2 0 1300 1700 0 0
aux 2 13 2 1500 2100 0 0
aux 3 35 0 1700 2100 0 0
aux 4 36 2 1500 2100 0 0
aux 5 39 1 900 1500 0 0

# adjrange
adjrange 0 0 3 900 1300 12 3 0 0
adjrange 1 0 3 1300 1700 12 3 0 0
adjrange 2 0 3 1700 2100 12 3 0 0

# rxfail
rxfail 4 s 1500
rxfail 6 s 1500

# master
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_calibration = 49,74,-216
set rssi_channel = 12
set rc_smoothing_debug_axis = YAW
set serialrx_provider = CRSF
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 6
set align_board_roll = 180
set bat_capacity = 1100
set ibata_scale = 206
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_visual_beeper = OFF
set osd_rssi_alarm = 15
set osd_alt_alarm = 101
set osd_rssi_pos = 2081
set osd_flymode_pos = 2105
set osd_current_pos = 2507
set osd_mah_drawn_pos = 2520
set osd_home_dir_pos = 2095
set osd_avg_cell_voltage_pos = 14785
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set name = Smoothy

profile 0

# profile 0
set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set pidsum_limit = 1000
set p_pitch = 76
set i_pitch = 80
set d_pitch = 44
set f_pitch = 200
set p_roll = 56
set i_roll = 70
set d_roll = 44
set f_roll = 200
set p_yaw = 45
set d_min_roll = 38
set d_min_pitch = 38

rateprofile 1

# rateprofile 1
set thr_expo = 40
set roll_rc_rate = 112
set pitch_rc_rate = 112
set yaw_rc_rate = 112
set roll_expo = 32
set pitch_expo = 32
set yaw_expo = 32
set roll_srate = 64
set pitch_srate = 64
set yaw_srate = 64

IvoFPV, 2019-09-07:
@theely You must have vtx table set up before go out to fly in 4.1. Although, there is a bug currently where if you have your VTX powered and you set the Smartaudio on a port and vtx table is not defined it will cause a lock up.

theely, 2019-09-07:
Yep I suppose this is a duplicate of that bug! I understand vtx table needs to be set-up but if the FC frezes there is no way to know you forgot to setup the vtx table.

IvoFPV, 2019-09-07:
There is a new warning when you set up your ports to Smartaudio and vtx table is not set.

McGiverGim, 2019-09-07:
If it freezes this is a bug that needs to be fixed. The SmartAudio will not work, but the FC must be totally functional.

etracer65, 2019-09-10:
@theely I think I found the problem. Can you try this version? Please test to see if the lockups are gone if smartaudio is selected on a UART, vtx is connected and powered, and vtxtable is NOT configured.

[betaflight_4.1.0_MATEKF722SE.hex.zip](https://github.com/betaflight/betaflight/files/3597510/betaflight_4.1.0_MATEKF722SE.hex.zip)