betaflight/betaflight #8847: Matek F411: BF 4.1 DShot BitBang not working when assigning Pin A08 (LED) to cam control

RipperDrone, 2019-09-07:
BetaFPV Whoop AIO 2-4S with Matek F411 FC, BF 4.1: 
BF 4.1 DShot BitBang not working when assigning Pin A08 (LED) to cam control

Steps to reproduce the behavior:
- flash BF 4.1 Nightly (today's first)
- flash JESC, rpm telemetry (v2.1, 24kHz version) to ESCs
- paste universal snippet to enable rpm telemetry and DSHOT300 etc. (from BF 4.1)
- CLI this:
# resource LED_STRIP 1 NONE
# resource CAMERA_CONTROL 1 A08

Everything worked ok until I remapped the LED_STRIP pin. Seems timers are used differently now, because it did work in non bit bang mode before.

RipperDrone, 2019-09-07:
Might possibly related to Issue #8840 (just guessing)

etracer65, 2019-09-07:
We really shouldn't have to ask every time. Include your `diff` as requested when you opened the issue.

jflyper, 2019-09-08:
@RipperDrone 
Yes, please always remember to provide sufficient information to reproduce your board on our side.

However, in this case, the answer is simple: "You can't do that." This is because F411 requires TIM1 for BB DShot and PA8 is TIM1.

If the board was working without an issue, I'd say you go back to `dshot_bitbang = OFF`.

RipperDrone, 2019-09-08:
@jflyper thank you! @etracer65 - I added the 2 single lines that make the change, therefore was suspecting the answer was straight about timer use. Have added the full diff still as per your request.

RipperDrone, 2019-09-08:
[Twig Betaflight Diff.txt](https://github.com/betaflight/betaflight/files/3587379/Twig.Betaflight.Diff.txt)

RipperDrone, 2019-09-08:
@jflyper @joelucid This situation might be a 'trap' for some users upgrading, isn't it? 'Abusing' the LED pin for softserial or - like in my case - cam control is probably a common way to work around the limited number of exposed UART pins on F411 Whoop boards. People who have reassigned the LED pin will hit this issue when upgrading since the default dshot mode is Bitbang now, and it will want to use TIMER 1. Don't see an elegant way out right now... 🤔

mikeller, 2019-09-08:
@RipperDrone: Definitely something we need to fix.

Marking this as a duplicate of #8840 so that we can keep the discussion in one place.

joelucid, 2019-09-08:
@mikeller can we keep this issue open to discuss the case of boards where the timer dshot implementation is unproblematic and does not require pin/dma remapping? I think in these targets we might want to default ``dshot_burst=off`` and ``dshot_bitbang=off``. What do you think?

joelucid, 2019-09-08:
Here are F411 targets which use timer 1:

MATEKF411RX:
```
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
    DEF_TIM(TIM9, CH2, PA3,  TIM_USE_PPM,   0, 0), // PPM/RX2

    DEF_TIM(TIM2, CH3, PB10, TIM_USE_MOTOR, 0, 0), // S1_OUT - DMA1_ST1
    DEF_TIM(TIM4, CH1, PB6,  TIM_USE_MOTOR, 0, 0), // S2_OUT - DMA1_ST0
    DEF_TIM(TIM4, CH2, PB7,  TIM_USE_MOTOR, 0, 0), // S3_OUT - DMA1_ST3
    DEF_TIM(TIM4, CH3, PB8,  TIM_USE_MOTOR, 0, 0), // S4_OUT - DMA1_ST7

    DEF_TIM(TIM5, CH1, PA0,  TIM_USE_LED,   0, 0), // 2812LED - DMA1_ST2

    DEF_TIM(TIM9, CH1, PA2,  TIM_USE_PWM,   0, 0 ), // TX2
    DEF_TIM(TIM1, CH2, PA9,  TIM_USE_PWM,   0, 0 ), // TX1
    DEF_TIM(TIM1, CH3, PA10, TIM_USE_PWM,   0, 0 ), // RX1
};
```

MATEKF411:
```
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
    DEF_TIM(TIM9, CH2, PA3,   TIM_USE_PPM,   0, 0), // PPM IN

    DEF_TIM(TIM3, CH1, PB4,  TIM_USE_MOTOR,  0, 0), // S1_OUT
    DEF_TIM(TIM3, CH2, PB5,  TIM_USE_MOTOR,  0, 0), // S2_OUT
    DEF_TIM(TIM4, CH1, PB6,  TIM_USE_MOTOR,  0, 0), // S3_OUT
    DEF_TIM(TIM4, CH2, PB7,  TIM_USE_MOTOR,  0, 0), // S4_OUT

    DEF_TIM(TIM2, CH2, PB3,  TIM_USE_MOTOR, 0, 0), // S5
    DEF_TIM(TIM2, CH3, PB10, TIM_USE_MOTOR, 0, 0), // S6
    DEF_TIM(TIM5, CH1, PA0,  TIM_USE_ANY,   0, 0), // RSSI pad
    DEF_TIM(TIM5, CH3, PA2,  TIM_USE_ANY,   0, 0), // TX2

    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_LED,   0, 0), // LED
};
```

MAMBAF411:

```
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {

    DEF_TIM(TIM11,  CH1,  PB9,  TIM_USE_PPM,   0, 0 ), // PPM IN
    DEF_TIM(TIM2,   CH2,  PB3,  TIM_USE_MOTOR, 0, 0 ), // S1_OUT - UP1-1
    DEF_TIM(TIM3,   CH1,  PB4,  TIM_USE_MOTOR, 0, 0 ), // S2_OUT - UP1-2
    DEF_TIM(TIM4,   CH1,  PB6,  TIM_USE_MOTOR, 0, 0 ), // S3_OUT - UP1-2
    DEF_TIM(TIM4,   CH2,  PB7,  TIM_USE_MOTOR, 0, 0 ), // S4_OUT - UP1-2
    DEF_TIM(TIM2,   CH1,  PA15, TIM_USE_ANY,   0, 0 ), // S5_OUT - UP1-1
    DEF_TIM(TIM10,  CH1,  PB8,  TIM_USE_ANY,   0, 0 ), // S6_OUT - DMA NONE

    DEF_TIM(TIM1,   CH1,  PA8,  TIM_USE_LED,   0, 0 ), // LED_STRIP - DMA2_ST3_CH6

};
```

joelucid, 2019-09-08:
At first look these seem fine for timer based dshot.

jflyper, 2019-09-08:
MATEKF411RX has much less chance for TIM1 being used, as it is configured as a backdoor timer on RX1 and TX1.

joelucid, 2019-09-08:
True given that this board has an integrated rx and no need for inverted telemetry.

etracer65, 2019-09-08:
> True given that this board has an integrated rx and no need for inverted telemetry.

Probably not a good assumption. I know of several people that used this board (or clones) and switched to using an external receiver because of instability with the onboard SPI receiver.

0crap, 2019-09-11:
> > True given that this board has an integrated rx and no need for inverted telemetry.
> 
> Probably not a good assumption. I know of several people that used this board (or clones) and switched to using an external receiver because of instability with the onboard SPI receiver.

Correct, I'm using a MATEKF11RX in the TinyhawkS, the internal SPI RX is utter cr@p!
Using a FrSky XM (not +) receiver on UART1. Problem solved.

joelucid, 2019-10-14:
Don’t want this to be auto-closed.  It’s a very common issue that’s been reported more than 15x over the last month or so. A fix has been available for some time (https://github.com/betaflight/betaflight/pull/8978) but so far apparently isn’t targeted for 4.1.