betaflight/betaflight #8855: Log GPS coordinates into blackbox

sanchosk, 2019-09-08:
**Is your feature request related to a problem? Please describe.**
I installed a GPS module on my quad and wanted to test the GPS rescue. I realized there is no way how to verify what was the perceived location during the GPS rescue - only way was to display the coordinates in the video feed.
It would be fantastic if one would be able to read the GPS coordinates (lat, lon, alt, spd) from the blackbox and, possibly, transform them into KML output.

**Describe the solution you'd like**
The GPS data is logged with the blackbox data onto the blackbox recorder or at least passed-through to some serial TX pin for logging.

**Describe alternatives you've considered**
Recording the TX output of the GPS module directly onto some OpenLog that I put onto the copter.

**Additional context**
I would be willing to contribute with code, but I only have experience with Arduino coding and the betaflight code seems to be much more mature than my current skill-level :(

mikeller, 2019-09-09:
Could probably be done by logging the GPS coordinates into the debug array.

McGiverGim, 2019-09-09:
The GPS info is at the Blackbox file. It is being logged. The problem is that the Blackbox Explorer does nothing with them.

sanchosk, 2019-09-09:
Perfect! I'll close this one and open a feature request for the Explorer then. Thanks!

closed 2019-09-09 by sanchosk