betaflight/betaflight #8859: Make paralyze also stop all servo movements

F2GD, 2019-09-09:
Paralyze should do just that, stop all movement. Is it possible to lock the servo outputs at 1500 when paralyze is enabled?

You can use a disarmed, non paralyzed quad to check servos, but I want the ability to make the quad do nothing at all

tigert, 2019-09-12:
Would it be better to stop sending PWM altogether, to keep sending the last pwm value, or send 1500?

I am thinking of a scenario where you crash and possibly jam a servo. If you then blindly send 1500, it can stress the servo and draw a lot of current, which could burn either the servo or a 5V regulator / bec.

I don't know what a servo will do if you connect power but send no PWM, it might depend on the model and whether it is a digital servo or not. 

Also there might be the odd case where you configured your servo endpoints so that it never goes to 1500, and even though it's a kind of a silly to design your servo linkage so that you never use the midpoint, such scenario might still be possible.

klystrom, 2019-12-21:
@F2GD Just curious, is there a use case for having the craft disarmed but servos are still active? Otherwise, wouldn't it be better for the servos to just go limp when disarmed?

Also during paralyze mode, craft would be in disarmed right?

mikeller, 2019-12-21:
@klystrom: A lot of RC pilots like to test the motion of their servos before arming, so disabling them while disarmed will force them to change their habits.

mikeller, 2019-12-21:
But disabling servo output when in paralyse mode should not be complicated and should be done.

F2GD, 2019-12-21:
Add in a line in cli for "disable servos on disarm"

mikeller, 2019-12-21:
@F2GD: 

> Add in a line in cli for "disable servos on disarm"\

What you are asking for here seems to be entirely unrelated to paralyse mode. Can you specify what it is that you would like to see done, and if it is both then open a separate issue for this so it can be discussed / tracked independently?

F2GD, 2019-12-21:
When I initially made this I forgot that paralize is to permanently
disabled a quad. What I would like is an option to force servo outputs to
1500 upon disarm.

For me this is actually to control a skid steer vehicle, but it's also
useful for calibrating my gimbals on a cinewhoop which currently you have
to do by using. Servo programmer to center them if you have servo tilt
enabled

F2GD, 2019-12-21:
Sorry @tigert I didn't see your earlier message a few months ago. Yeah I would want to things added to cli, the first is "set servo disarm = true/false" and "set servo disarm pulse = *1500 for example*" just so that we have to option to configure it ourselves, it would be great in the GUI but for now cli is good enough. 

If you set it to 0 then it holds the last position it sent

closed 2020-01-28 by stale[bot]