betaflight/betaflight #8863: Can I use a BME280 on I2C?

crteensy, 2019-09-10:
Most configuration examples for Bosch pressure sensors use the BMP280 or BMP085. However, it's currently simpler for me to get a BME280 instead (added humidity sensing ability). Both BMP280 and BME280 have the same I2C base address and the register descriptions for anything other than humidity sensing look very similar.

I've found the following differences (BMP280 vs BME 280):
- ID register: 0x58 vs. 0x60
- Config register: sleep times are different:
  - The BMP280 has 0.5 to 1000 ms and then 2000 and 4000 ms;
  - The BME280 has 0.5 to 1000 ms and then 10 and 20 ms
- obviously the humidity register is missing in the BMP280.

That doesn't mean that there aren't more differing details.

I don't need to read humidity values from the sensor. I'd just like to know if the pressure part would work with betaflight. Specifically I see a potential problem if betaflight checks of the ID register contains the correct value.

Short: Can I use the BME280 or not?

mikeller, 2019-09-10:
@crteensy: If the BMP280 and the BME280 are as similar as you describe then you should be more or less able to use the BMP280 driver. The only problem will be the chip id, as the driver uses it to detect the chip:

https://github.com/betaflight/betaflight/blob/aeaf1acf4dfbdba7ae4df92086fb80c9879a6fb9/src/main/drivers/barometer/barometer_bmp280.c#L44

But you should be able to work around this for testing by changing the relevant line to `0x60`. If your testing proves to be successful it will be easy to amend the driver to check for either of these two chip ids. Please let us know how you are going.

crteensy, 2019-09-10:
If it's just that I'll go for it. Two things:
- It'll take a while for the test PCB to arrive.
- It might make sense to add a "proper" driver for the BME since the humidity data *might* be/become useful.

sanchosk, 2019-10-13:
Not sure how would the flight controller use the humidity data for flight, but if you have an idea, I'd propose to open a suggestion for the devs.
I tested both BMP and BME, they got detected and work OK. Just be careful - make sure you don't connect the 3.3V version to 5V - it works for few seconds and then stops forever ;)

crteensy, 2019-10-13:
Can you share code? Is it really just a matter of including the additional device ID in the presence check?

sanchosk, 2019-10-13:
No code, I just soldered it and it got auto-detected when the quad battery was present. I just enabled the baro in betaflight.

crteensy, 2019-10-13:
That could either mean that the BME is already coded for in the BF version you used, or the chip ID check doesn't work as intended

mikeller, 2019-10-15:
@crteensy:

> That could either mean that the BME is already coded for in the BF version you used, or the chip ID check doesn't work as intended

That is interesting - according to the information you posted earlier (https://github.com/betaflight/betaflight/issues/8863#issue-491539385), the BME 280 uses a different id than the BMP 280:

> ID register: 0x58 vs. 0x60

I checked the driver in Betaflight, and it only checks for 0x58, so either you information is outdated, or there are different versions of the BME 280 that return different ids.

crteensy, 2019-10-15:
Mildly puzzled... Which piece of my information would be outdated?

mikeller, 2019-10-15:
@crteensy: The bit about the id register being 0x60 for the BME280.

closed 2019-11-21 by stale[bot]