betaflight/betaflight #8868: Pit mode gets randomly enabled/disabled

IvoFPV, 2019-09-11:
**Describe the bug**
Vtx pit mode gets randomly enabled disabled when saving settings in configurator. A few reported this behavior on Slack and it seems to be a firmware issue. As far as I checked configurator never sends `VTX_CONFIG.vtx_pit_mode = 1` when I try to reproduce it so there is no issue there. After saving firmware returns that pit mode is enabled even though it wasn't told to enable it.

**To Reproduce**
Set up a vtx table, choose band, channel, power and save. Save again and again without 
changing anything. Pit mode will sometimes enable/disable just by doing that.

**Expected behavior**
Pit mode only gets enabled/disabled if you switch it on/off manually.

**Flight controller configuration**
Defaults settings on latest master build, assigned port to Smartaudio, vtx table set, no vtx connected to flight controller.

etracer65, 2019-09-11:
> no vtx connected to flight controller

Does the problem happen when a real vtx is connected? When there's no vtx the status is somewhat questionable and pit mode is part of the information actually received from the vtx to determine state. Remember, pit mode can be enabled directly on some vtx with button presses. Not saying there isn't something to fix, but a pit mode indicator when there's no vtx connected isn't very meaningful or really a big deal.

IvoFPV, 2019-09-11:
@etracer65 Users who have first reported this on Slack had vtx connected. Maybe it has to do with vtx not being powered when on USB, or in my case when it is not even connected. So the pit mode shouldn't change if there is no valid state information received from vtx. I guess that is the problem.

IvoFPV, 2019-09-11:
Just confirmed that it happens with both unpowered and powered VTX.

etracer65, 2019-09-11:
I can't seem to replicate. Can you provide a `diff` and steps used to cause the problem?

IvoFPV, 2019-09-11:
@etracer65 Here it is but it is simple as this to reproduce. In configurator just load vtx tab multiple times by switching to it or use Save both of which will call `MSP_VTX_CONFIG` and refresh. It takes a couple of tries, like 5-6. To me it is completely random.
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.0 Sep 10 2019 / 13:58:55 (2a1820949) MSP API: 1.42

# start the command batch
batch start

board_name MATEKF405CTR
manufacturer_id MTKS

# serial
serial 2 2048 115200 57600 0 115200

# master
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end
```

etracer65, 2019-09-11:
@IvoFPV Looks like just a case of a variable not being initialized and a code path around the logic that sets it. Please try this:

[betaflight_4.1.0_STM32F405.hex.zip](https://github.com/betaflight/betaflight/files/3600794/betaflight_4.1.0_STM32F405.hex.zip)

IvoFPV, 2019-09-11:
@etracer65 It works! Tried with the same setups and I can't replicate it anymore.

etracer65, 2019-09-11:
Fixed in https://github.com/betaflight/betaflight/pull/8869